ioBroker/ioBroker.modbus #5: Memory Leak

Sineos, 2017-02-26:
The modbus adapter seems to be leaking memory. 
I'm reading from 10 holding registers and writing to one input register. The adapter is configured as slave.

Environment:
- Raspian Linux on RPi3
- Node v6.10.0
- Modbus Adapter 0.5.0

Right after iobroker start, the modbus adapter is using roughly 120MB RAM as per `ps `VSZ reading.

After about 1 day:
```
root@piheizung /home/pi # date && ps aux --sort=-vsz,-rss | head -10 
Sa 25. Feb 23:14:46 CET 2017
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root      1348  1.0  5.6 158608 56664 ?        Sl   Feb24  17:28 node-red
root      1468  1.4  6.6 156704 65660 ?        Sl   Feb24  22:43 io.modbus.0
root      1298  3.3  4.5 146440 45700 ?        Ssl  Feb24  54:01 iobroker.js-controller
root      1308  1.8  4.4 146060 44408 ?        Sl   Feb24  30:00 io.admin.0
root      1364  1.5  3.9 139568 38884 ?        Sl   Feb24  25:05 io.mqtt.0
root      1449  1.2  3.7 126564 36892 ?        Sl   Feb24  19:39 io.influxdb.0
root      8550  0.2  2.9 119724 29376 ?        Sl   11:53   1:27 io.viessmann.0
root      1338  0.0  2.8 119328 28008 ?        Sl   Feb24   0:48 io.node-red.0
root       748  0.1  0.8  85704  7980 ?        Ssl  Feb24   2:13 /usr/local/bin/vzlogger -c /etc/vzlogger.conf
```
Another 9 hours later:
```
root@piheizung /home/pi # date && ps aux --sort=-vsz,-rss | head -10 
So 26. Feb 08:44:55 CET 2017
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root      1468  1.7  7.9 168992 79508 ?        Sl   Feb24  39:07 io.modbus.0
root      1348  1.0  5.8 159632 58276 ?        Sl   Feb24  23:25 node-red
root      1298  3.3  4.5 146788 44892 ?        Ssl  Feb24  73:57 iobroker.js-controller
root      1308  1.8  4.2 142612 41880 ?        Sl   Feb24  41:14 io.admin.0
root      1364  1.5  4.0 139568 40492 ?        Sl   Feb24  33:25 io.mqtt.0
root      1449  1.2  4.0 131216 40436 ?        Sl   Feb24  27:33 io.influxdb.0
root      8550  0.2  2.9 119724 29036 ?        Sl   Feb25   2:35 io.viessmann.0
root      1338  0.0  2.8 119588 28704 ?        Sl   Feb24   1:03 io.node-red.0
root       748  0.1  0.8  85704  7980 ?        Ssl  Feb24   2:24 /usr/local/bin/vzlogger -c /etc/vzlogger.conf
```
In the 9 hours the mem usage of the has increased about 12 MB and compared to the start of iobroker an increase of about 45 MB.

Although wrong topic:
The same seems to be true for the influxdb adapter. Here the mem increase is roughly 10 MB since start.

Apollon77, 2017-02-26:
For influxdb: The adapter is caching the last value per datapoint in memory and also (when buffering is used) some values in the buffer before writeing it out to influxdb. this needs some memory ...

And for both: nodejs has it's own garbage collection internally, so the JS-code have no real control over when memory is free'd up by the nodejs process.

Can you please monitor both further for some days to see if the memory usage still increases ?

Sineos, 2017-02-26:
I will continue to monitor it.

While it is clear that some memory is needed to buffer values etc, I would think that an increase of 45 MB is   a bit much.
It is also interesting to see, that the memory values shown under "Instances" does nowhere match the values as displayed by `ps`. The values as shown there is something around 40 MB, while `ps` reports over 150 MB for VSZ and even RSS is way above. How is the value under Instances calculated?

Sineos, 2017-03-01:
I have monitored mem usage via this script:
```
for i in `ps -eaf | grep io.modbus.0 | grep -v grep | awk '{print $2}'`; do
        date  >> totaluse_modbus.txt;
        echo -n "PID $i actual memory usage is :" >> totaluse_modbus.txt;
        pmap -d $i | grep -i "writeable/private: " >> totaluse_modbus.txt;
done
```

**influxdb:**
```
So 26. Feb 18:42:19 CET 2017
PID 17937 actual memory usage is :mapped: 129420K    writeable/private: 93072K    shared: 0K
So 26. Feb 23:00:24 CET 2017
PID 17937 actual memory usage is :mapped: 128396K    writeable/private: 92048K    shared: 0K
Mo 27. Feb 09:31:24 CET 2017
PID 17937 actual memory usage is :mapped: 131900K    writeable/private: 95544K    shared: 0K
Mo 27. Feb 13:04:54 CET 2017
PID 17937 actual memory usage is :mapped: 130876K    writeable/private: 94528K    shared: 0K
Mo 27. Feb 20:17:49 CET 2017
PID 17937 actual memory usage is :mapped: 123708K    writeable/private: 87360K    shared: 0K
Di 28. Feb 07:15:50 CET 2017
PID 17937 actual memory usage is :mapped: 130848K    writeable/private: 94500K    shared: 0K
Di 28. Feb 12:29:25 CET 2017
PID 17937 actual memory usage is :mapped: 128856K    writeable/private: 92604K    shared: 0K
Mi 1. MÃ¤r 07:50:15 CET 2017
PID 17937 actual memory usage is :mapped: 123924K    writeable/private: 87704K    shared: 0K
```
The influxdb adapter apparently has no mem leak as the garbage collector seems to work nicely.

**modbus:**
```
So 26. Feb 09:30:59 CET 2017
PID 17927 actual memory usage is :mapped: 124664K    writeable/private: 86232K    shared: 0K
So 26. Feb 16:00:54 CET 2017
PID 17927 actual memory usage is :mapped: 131592K    writeable/private: 93632K    shared: 0K
So 26. Feb 23:00:24 CET 2017
PID 17937 actual memory usage is :mapped: 139372K    writeable/private: 101412K    shared: 0K
Mo 27. Feb 09:31:24 CET 2017
PID 17927 actual memory usage is :mapped: 154732K    writeable/private: 116772K    shared: 0K
Mo 27. Feb 13:04:54 CET 2017
PID 17927 actual memory usage is :mapped: 158828K    writeable/private: 120868K    shared: 0K
Mo 27. Feb 20:17:49 CET 2017
PID 17927 actual memory usage is :mapped: 164972K    writeable/private: 128036K    shared: 0K
Di 28. Feb 07:15:50 CET 2017
PID 17927 actual memory usage is :mapped: 180332K    writeable/private: 143396K    shared: 0K
Di 28. Feb 12:29:25 CET 2017
PID 17927 actual memory usage is :mapped: 184428K    writeable/private: 146468K    shared: 0K
Mi 1. MÃ¤r 07:50:15 CET 2017
PID 17927 actual memory usage is :mapped: 209004K    writeable/private: 171044K    shared: 0K
```
At least for me, this shows no signs of any garbage collection, but constantly growing mem usage.

As the modbus adapter is running as slave, I'm seeing the following in the log:
```
modbus.0	2017-03-01 22:25:54.176	info	modbus.0 - Client connected:
modbus.0	2017-03-01 22:25:54.053	info	modbus.0 + Clients connected: 192.168.178.20
modbus.0	2017-03-01 22:25:46.421	info	modbus.0 - Client connected:
modbus.0	2017-03-01 22:25:46.303	info	modbus.0 + Clients connected: 192.168.178.20
modbus.0	2017-03-01 22:25:46.173	info	modbus.0 - Client connected:
modbus.0	2017-03-01 22:25:46.054	info	modbus.0 + Clients connected: 192.168.178.20
```
For every value the master pushes to the slave, a new connection is established. As far as I understand the code a `modbusClient.close();` is only called upon stopping the adapter or on reconnect after failure.
Is it possible that all these client instances add up the memory?

My coding skills are limited, but I will happily assist tracking this one down.

Apollon77, 2017-03-01:
hm ... Also das Log:
- Client connected:
kommt bei einem close-Event ... damit werden die Connections geschlossen ...
:-(

Sineos, 2017-03-02:
I have added [memwatch-ng](https://github.com/srowatt/memwatch-ng) to the adapter: https://github.com/Sineos/ioBroker.modbus/commit/a15faff40674a6217be4c2122babe78cd6ce558c

This gives me the attached output:
[modbus_leak.txt](https://github.com/ioBroker/ioBroker.modbus/files/813088/modbus_leak.txt)

Obviously the place to add the watch was badly chosen, but it kind of worked :-P

Maybe memwatch-ng can be used to track down this leak but I would require your support because this is slightly over my head.

Apollon77, 2017-03-02:
Ok, then we try to go further and I need your support too because I do not own any modbus device, so you must do that testing :-(
But the memwatch idea is great!!

Do the following. For now let the "leak" think where it is but enhance with HeapDiff thing, so do:
- add an variable where you had thge "require" added, like "var heapDiff = null;" so that it is in global scope
- then add under the leak logging:

if (heapDiff) {
  var diff = heapDiff.end();
  adapter.log.error('Diff: ' + JSON.stringify(diff));
}
heapDiff = new memwatch.HeapDiff();

Then we will see ... maybe it brings more insight

Sineos, 2017-03-02:
First try: https://github.com/Sineos/ioBroker.modbus/commit/5c50c2e992b53e2f78b25def1decc0a96fd3c95e
This is quite noisy imho. Output is the following: [heapDiff_1st_try.txt](https://github.com/ioBroker/ioBroker.modbus/files/813645/heapDiff_1st_try.txt)

Switched to: https://github.com/Sineos/ioBroker.modbus/commit/8c0d8e0294d01f2f085b55ae6da03e0cc6c40407
This one is running right now. Will post output as soon as available.

I'm trying to understand the adapter's code to find a place where the "main loop" is running to add the memwatch point, but I don't really get it. The connection function seems really bad as it is called way too often.

Apollon77, 2017-03-02:
Bei nodejs/JavaScript gibts keine echte main-loop, da alles eventbasiert abläuft. :-(
Man sieht das "String size" immer steigt ... also würde ich tippen das da was ist ... ich schaue mal rein

Apollon77, 2017-03-02:
Ich hab Code geschaut und bei tcp-rtu-client und serial-client zwei STellen gefunden wo was faslsch resettet wurde. Galube zwar nicht das es hier relevant ist aber vllt aktualisiere und teste mal :-)

Sineos, 2017-03-03:
Ich habs mit Deinen Änderungen probiert. Das Leak ist immer noch da.
Mein momentaner Stand ist: https://github.com/Sineos/ioBroker.modbus/commit/0d655633fe93486cbb5715e2c7a2ff67873e03b9

[memstats.txt](https://github.com/ioBroker/ioBroker.modbus/files/816263/memstats.txt) hier sieht man jetzt schön das anwachsende Mem.

Mit [heapdump](https://github.com/bnoordhuis/node-heapdump) gibt er mir jetzt bei jedem gc-Lauf den kompletten Heap aus. In Chrome kann man sich den Heap dann anschauen, leider ist mein Plan da etwas gering.
Was mir aufgefallen ist:
- Es gibt einige uncaught exceptions in dem dump
- Einiges an Speicher scheint auf den Logger zu deuten
- Die Sockets von modbus scheinen auch immer mehr zu werden

[Hier](https://dl.dropboxusercontent.com/u/77328279/heapdumps.zip) sind die HeapDumps.

Sineos, 2017-03-08:
Ich hab den Funktionen in main.js und der lib ModBus Namen gegeben: 
https://github.com/Sineos/ioBroker.modbus/commit/5bb2d70d0c8b835704f8a504f534c92761f6487a
https://github.com/Sineos/ioBroker.modbus/commit/97cc569bd5dd24cc42897f9bfad27ed80d285590
Die heapdumps scheinen mir jetzt deutlich lesbarer, nur ich werd noch nicht richtig schlau draus.

Für mich sieht es so aus als würde die "clients" Liste immer weiter anwachsen und die Connections nicht sauber beendet / gelöscht werden.

[Hier](https://dl.dropboxusercontent.com/u/77328279/heapsnapshot2.rar) die dumps

Apollon77, 2017-03-09:
also unter der Annahme hast Du recht. Objekte werden nur zerstört wenn der Server gestoppt wird.

Ich hab jetzt mal (blind weil kann es nicht testen) eingebaut das beim closen von tcp sockets diese direkt zerstört werden ... das "clients" wird zwar aktuell immer noch größer, aber mal zu testen als ersten Schritt ... versuch mal von github...
https://github.com/ioBroker/ioBroker.modbus/commit/e85abc9a7abc1fd51d66670249cf8e5d06bc82db

Apollon77, 2017-03-09:
PS: Ich habs nochmal leicht geändert und wir versuchen mal ob es reicht nur den Eintrag im "clients"-Array zu dereferenzieren ... explizit destroy könnte mit dem gerade emitteten "close" kollidieren

Sineos, 2017-03-10:
Die Änderung mag er nicht:
```
modbus.0	2017-03-10 06:33:55.907	error	TypeError: Cannot read property 'address' of undefined
modbus.0	2017-03-10 06:33:55.905	error	uncaught exception: Cannot read property 'address' of undefined
```
ich hab zum Test mal das `clients[socketId-1]` geändert auf `clients[socketId]`.

Die Funktion hatte ich mir auch schon angeschaut. Mein Verständnis war:
```
        var onSocketEnd = function (socket, socketId) {

            return function () {
                this.emit('close');
                this.log.debug('connection closed, socket', socketId);
```
über das emit('close') sollte doch folgendes aufgerufen werden:
```
        this.close = function (cb) {

            for(var c in clients) {
                clients[c].destroy()
            }

            server.close(function() {
                server.unref();
                if(cb) { cb() }
            });

};
```
oder liege ich da falsch? Damit müsste ja eigentlich sauber gemacht werden. Das einizge was ich hier nicht erkennen kann ist, ob damit auch die client list wieder abgebaut wird

Apollon77, 2017-03-10:
Ne the "this.close" ist der Server an sich. Da werden dann alle clients gekillt, was ja auch korrekt ist.

Ich glaub ich hab die Stelle gefunden wo es bei Dir geknallt hat. Versuch nochmal.

Das "socketId-1" ist denke ich nötig weil er mit nem leeren Array startet und dann jede neuen client mittels "push" einfügt. Der erste hat dann also die Array-ID =0. Er gibt der Funktion als SocketId aber einen um 1 erhöhten Wert mit.

Sineos, 2017-03-11:
Das wars. Der Speicherverbrauch bleibt weitgehend stabil:
[image]

Das memwatch.on('leak', function(info) Event wurde in 15 Stunden nur 2x aufgerufen (vorher teils mehrmals pro Stunde). 
In den [HeapDumps ](https://dl.dropboxusercontent.com/u/77328279/heapsnapshot3.rar) sieht man schon noch ein Leak aber deutlich kleiner.

Klasse. Vielen Dank für Deine Unterstützung 👍

Apollon77, 2017-03-11:
Super. kannst Du bitte mal die 0.5.1 vom Github testen ob das passt?

closed 2017-03-11 by Apollon77

Sineos, 2017-03-12:
Funktioniert und sieht stabil aus:
```
Sa 11. MÃ¤r 16:15:48 CET 2017
PID 9702 actual memory usage is :mapped: 123324K    writeable/private: 89580K    shared: 0K
Sa 11. MÃ¤r 16:54:24 CET 2017
PID 9702 actual memory usage is :mapped: 126116K    writeable/private: 88648K    shared: 0K
Sa 11. MÃ¤r 20:02:31 CET 2017
PID 9702 actual memory usage is :mapped: 126512K    writeable/private: 89192K    shared: 0K
Sa 11. MÃ¤r 22:42:45 CET 2017
PID 9702 actual memory usage is :mapped: 124704K    writeable/private: 87384K    shared: 0K
So 12. MÃ¤r 10:32:30 CET 2017
PID 9702 actual memory usage is :mapped: 125776K    writeable/private: 88568K    shared: 0K
So 12. MÃ¤r 11:46:39 CET 2017
PID 9702 actual memory usage is :mapped: 125776K    writeable/private: 88568K    shared: 0K
```

Danke!

Apollon77, 2017-03-12:
Perfekt, ich danke dir. Am Ende waren Deine Tipps ausschlaggebend!