ioBroker/ioBroker.modbus #7: Coils

Predi-ger-many, 2017-08-16:
Hello,

I tried to write coils. 2 Problems:

1. I ´want to update ID 200. But ID 199 was updatet.
2. iobroker.modbus crashed

host.vm02 2017-08-16 02:29:50.562 error instance system.adapter.modbus.0 terminated with code 0 (OK)  
modbus.0 2017-08-16 02:29:50.545 error at Timer.listOnTimeout (timers.js:92:15)  
modbus.0 2017-08-16 02:29:50.545 error at Object. (C:\ioBroker\node_modules\iobroker.modbus\node_modules\jsmodbus\src\handler\server\ReadCoils.js:38:22) 
modbus.0 2017-08-16 02:29:50.545 error at Object.emit (C:\ioBroker\node_modules\iobroker.modbus\node_modules\stampit-event-bus\src\stampit-event-bus.js:20:38) 
modbus.0 2017-08-16 02:29:50.545 error at Object. (C:\ioBroker\node_modules\iobroker.modbus\main.js:1603:44) 
modbus.0 2017-08-16 02:29:50.545 error at Buffer.readUInt8 (buffer.js:718:5) 
modbus.0 2017-08-16 02:29:50.545 error at checkOffset (buffer.js:680:11) 
modbus.0 2017-08-16 02:29:50.545 error RangeError: index out of range 
modbus.0 2017-08-16 02:29:50.545 error uncaught exception: index out of range

Predi-ger-many, 2017-08-16:
The Crash ocurrs when:

1. I definied one Modbus-I/O with ID 2, type coils.
2. I connect with razio modbus sim an set it to coils.

this results in a crash

GermanBluefox, 2018-01-20:
Should work now. https://github.com/ioBroker/ioBroker.modbus/commit/900e21950abc8d76ba49b0f056f35003d994145c

GermanBluefox, 2018-03-04:
Retest?

Apollon77, 2018-03-04:
Please reopen if still a problem

closed 2018-03-04 by Apollon77