ioBroker/ioBroker.modbus #8: Can't connect over RTU

m-obi, 2017-12-13:
I get no connection by RTU. Is it already tested?

Apollon77, 2017-12-14:
Please provide Debug log (set Instance in Debug)

m-obi, 2017-12-14:
modbus.1 | 2017-12-14 07:33:16.233 | info | Disconnected from slave 127.0.0.1
modbus.1 | 2017-12-14 07:33:16.233 | warn | Poll error count: 10 code: {'err':'timeout'}
modbus.1 | 2017-12-14 07:33:16.232 | warn | Error: undefined
modbus.1 | 2017-12-14 07:33:11.219 | info | Connected to slave
modbus.1 | 2017-12-14 07:33:01.203 | info | Disconnected from slave 127.0.0.1
modbus.1 | 2017-12-14 07:33:01.191 | warn | Poll error count: 9 code: {'err':'timeout'}
modbus.1 | 2017-12-14 07:33:01.189 | warn | Error: undefined
modbus.1 | 2017-12-14 07:32:56.180 | info | Connected to slave
modbus.1 | 2017-12-14 07:32:46.172 | info | Disconnected from slave 127.0.0.1
modbus.1 | 2017-12-14 07:32:46.169 | warn | Poll error count: 8 code: {'err':'timeout'}
modbus.1 | 2017-12-14 07:32:46.167 | warn | Error: undefined
modbus.1 | 2017-12-14 07:32:41.161 | info | Connected to slave
modbus.1 | 2017-12-14 07:32:31.146 | info | Disconnected from slave 127.0.0.1
modbus.1 | 2017-12-14 07:32:31.143 | warn | Poll error count: 7 code: {'err':'timeout'}
modbus.1 | 2017-12-14 07:32:31.141 | warn | Error: undefined
modbus.1 | 2017-12-14 07:32:26.128 | info | Connected to slave
modbus.1 | 2017-12-14 07:32:21.322 | warn | Poll error count: 6 code: 'App Timeout'
modbus.1 | 2017-12-14 07:32:16.113 | info | Disconnected from slave 127.0.0.1
modbus.1 | 2017-12-14 07:32:15.294 | warn | Poll error count: 5 code: 'App Timeout'
modbus.1 | 2017-12-14 07:32:15.112 | warn | Poll error count: 4 code: {'err':'timeout'}
modbus.1 | 2017-12-14 07:32:15.109 | warn | Error: undefined
modbus.1 | 2017-12-14 07:32:10.091 | info | Connected to slave
modbus.1 | 2017-12-14 07:32:05.305 | warn | Poll error count: 3 code: 'App Timeout'
modbus.1 | 2017-12-14 07:32:00.079 | info | Disconnected from slave 127.0.0.1
modbus.1 | 2017-12-14 07:31:59.233 | warn | Poll error count: 2 code: 'App Timeout'
modbus.1 | 2017-12-14 07:31:59.084 | warn | Poll error count: 1 code: {'err':'timeout'}
modbus.1 | 2017-12-14 07:31:59.072 | warn | Error: undefined
modbus.1 | 2017-12-14 07:31:54.020 | info | Connected to slave
modbus.1 | 2017-12-14 07:31:53.377 | info | starting. Version 0.5.4 in /home/pi/node_modules/iobroker.modbus, node: v6.11.5
modbus.1 | 2017-12-14 07:31:53.276 | debug | statesDB connected
modbus.1 | 2017-12-14 07:31:53.109 | debug | objectDB connected
modbus.1 | 2017-12-14 07:31:46.925 | info | terminating
modbus.1 | 2017-12-14 07:31:41.894 | warn | On error: {}
modbus.1 | 2017-12-14 07:31:41.693 | debug | sendTo 'listUart' to system.adapter.admin.0 from system.adapter.modbus.1
modbus.1 | 2017-12-14 07:31:41.687 | info | List of port: [{'comName':'/dev/ttyAMA0'},{'comName':'/dev/ttyUSB0','manufacturer':'FTDI','serialNumber':'FTDI_FT232R_USB_UART_AH02WBOR','pnpId':'usb-FTDI_FT232R_USB_UART_AH02WBOR-if00-port0','vendorId':'0x0403','productId':'0x6001'}]
modbus.1 | 2017-12-14 07:31:40.888 | warn | On error: {}
modbus.1 | 2017-12-14 07:31:40.211 | info | starting. Version 0.5.4 in /home/pi/node_modules/iobroker.modbus, node: v6.11.5
modbus.1 | 2017-12-14 07:31:40.041 | debug | statesDB connected
modbus.1 | 2017-12-14 07:31:39.879 | debug | objectDB connected

Apollon77, 2017-12-14:
Konfiguration ist korrekt? Der RTU ist auf 127.0.0.1?

m-obi, 2017-12-14:
Das wundert mich ja auch
[image]

m-obi, 2017-12-14:
Reading Holding Register works perfectly. But Discrete Inputs not.
The Request is wrong.
It sends
`0a 02 00 00 00 10 78 bd`
instead of
`0a 02 00 00 00 01 b8 b1`

I'm reading at address 0 one input.

GermanBluefox, 2017-12-14:
Does it mean you can see some values?

m-obi, 2017-12-15:
At Holding Regs yes, but not Discrete Inputs, because of the wrong request.

And I have one advice. Put the setting for the Device/Unit ID to the register site. So we can use more than one device. Because you can't use the same connection in two or more instances.

GermanBluefox, 2018-05-03:
Duplicate to https://github.com/ioBroker/ioBroker.modbus/issues/11

closed 2018-05-03 by GermanBluefox