ioBroker/ioBroker.modbus #11: Mehrere Modbusteilnehmer  über USB to Serialadapter RS232 - RS485 einbinden

smartboart, 2018-05-02:
Hallo, aktuell ist mir beim Einrichten des Mosbusadapter aufgefallen, dass leider nur ein Modbusteilnehmer über RS485 Adapter konfiguriert werden kann. Da ich aktuell gerne weitere Geräte über diese Schnittstelle einbinden würde, hoffe ich natürlich, dass dieser wirklich sinvolle und gelungene Adapter weiter entwickelt wird. Eine Installation einer weiteren Instanz hilft hier auch nicht weiter, weil die Hardwareschnittstelle nur von einem Adapter angesprochen werden kann.

GermanBluefox, 2018-05-03:
Wird aktuell sehr aktiv entwinkelt. Dafür sollte ich leider alles umschreiben.
https://github.com/ioBroker/ioBroker.modbus/tree/refactoring

GermanBluefox, 2018-05-05:
Bist du bereit zu testen? :)
```
cd /opt/iobroker
npm i https://github.com/ioBroker/ioBroker.modbus/tarball/refactoring
```

smartboart, 2018-05-05:
Super natürlich gerne....ich bin nur erst am mittwoch wieder zu hause um den rs485 adapter  wieder anzuschliessen......den modbus adapter werde ich aber gleich mal via vpn installieren... Sobald ich alles wieder an das tinkerboard angeschlossen habe gehts los... Echt klasse was ihr leistet....

smartboart, 2018-05-05:
So installiert. Super mit der zusätzlichen Möglichkeit die Slave ID einzutragen...Weiter kann ich leider erst am Mittwoch testen, dann werde ich meine Modbusgeräte wieder auf das Tinkerboard umschwenken, nachdem ich mit dem Oszilloskop die Leitung eingemessen habe.. Vielen Dank schonmal ...und sorry das es erst nächste Woche weiter geht...

smartboart, 2018-05-13:
Hallo, sorry, bin erst gestern wieder gekommen...Also ich habe den Modbus eingemessen, siehe PDF unten...Leider bekomme ich  mit dem USB RS485 Converter von Digitus  welcher fürs Tinkerboard gedacht ist  schon generell keine  Verbindung meinen Teilnehmern...Auch nicht mit dem PC über Modbus Poll oder den CAS Modbus Scanner...Solange ich nicht weis woran das liegt, bin ich leider keine große Testhilfe...

[Ganzseitiges Foto.pdf](https://github.com/ioBroker/ioBroker.modbus/files/1998412/Ganzseitiges.Foto.pdf)

wendy2702, 2018-05-16:
Hi,

ich nutze diesen Adapter 2 mal. Hast du bei der Verkabelung auf die "Polung" geachtet? ich weiß, klingt doof aber ich habe einen Energiezähler bei dem das Auslesen nur funktioniert wenn wirklich richtig gepolt. Wichtig könnten auch Abschlußwiderstände und die Art des Verbindungskabels sein.

Gibt es im Forum einen Thread dazu?

wendy2702, 2018-05-16:
Hat mit deiner Verkabelung und Einstellung (auch im Alten Adapter) denn das Auslesen **eines** Busteilnehmers funktioniert? Weil wenn das schon nicht geklappt hat liegt der Fehler eventuell woanders.

smartboart, 2018-05-16:
Der Fehler liegt nicht am iobroker Modbus Adpter...Es liegt an der Hardware...Hatte  ja oben schon geschrieben, dass ich kein guter Testkandidat bin, solange ich das nicht gelöst bekomme...

wendy2702, 2018-05-16:
Meinst du die HW die du auslesen möchtest oder das Tinkerboard?

smartboart, 2018-05-16:
die ich auslesen möchte, da es ja auch mit dem notebook über modbuspoll und CAS Modbusscanner nicht funktioniert...Den RS485 Adapter habe ich auch 2x mal gekauft um einen defekt auszuschließen... Muss also an den Geräten liegen die ich auslesen möchte...

wendy2702, 2018-05-16:
Darf man fragen was das für Geräte sind?

TomSC1984, 2018-05-16:
Ich habe das Modul installiert so wie oben beschrieben.Nur wie bekomme ich es gestartet?

wendy2702, 2018-05-16:
Du musst dann auch eine Instanz hinzufügen.

wendy2702, 2018-05-16:
Denke Bluefox liest hier mit.

Vielleicht kannst du mal einen Screenshot der Einstellungsseite posten.

TomSC1984, 2018-05-16:
[image]
[image]
Hier mal die Screenshots der Adapter

GermanBluefox, 2018-05-16:
Wichtig ist, dass man aus branch "refactoring" installiert.

smartboart, 2018-05-16:
Klar darfst du Fragen, bin für jede Hilfe dankbar...
Sind 3 Stromzähler von ABB...Doku siehe pdf...
[Energiezaehler_B-Serie_Handbuch.pdf](https://github.com/ioBroker/ioBroker.modbus/files/2010910/Energiezaehler_B-Serie_Handbuch.pdf)

TomSC1984, 2018-05-17:
Ich kann leider trotzdem nichts sehen wo ich mehrere Slave ID's eintragen kann. Stell mich wohl zu blöd. Aus dem branch "refactoring" installiere ich ja über den npm mit "npm i https://github.com/ioBroker/ioBroker.modbus/tarball/refactoring"

Da kann ich doch garnichts falsch machen? 

Gruß Thomas

smartboart, 2018-05-17:
cd /opt/iobroker npm i https://github.com/ioBroker/ioBroker.modbus/tarball/refactoring 

Die slaves können bei den einstellungen eingetragen werden wo auch die datenpunkte eingestellt werden.

wendy2702, 2018-05-17:
@smartboart : du bist aber sicher das deine Zähler die Modbus Option haben (aus der Doku): Optionale Schnittstellen: M-Bus, RS-485 (Modbus oder EQ-Bus einstellbar)

Du hast also einen dieser Zähler: Bei Zählern mit Schnittstelle RS-485 (B2x yz2 – 1. .)?

Die Modbus Einstellungen hast du geprüft und so im Adapter stehen?

Wir hatten das ja auch schon im Forum. Da hast du geschrieben das ab Adressen >20000 Fehler kommen.

Die ist es also nicht gelungen für irgendeinen Wert die Daten zu bekommen?

Woher weißt du an welchen Pin am Zähler der Bus muss? ich habe nur dieses Bild gesehen oder habe ich etwas überlesen?

[image]

smartboart, 2018-05-17:
Ja in den Einstellungen an den Zählern kann ich RS485 Modbus einstellen und auch weitere Werte welche ich auf  9600 baud, Parität none und stoppits 1 eingestellt habe.
zu den Anschlüssen A B C habe ich gegoogelt und herausgefunden dass A + , B- und C Ground sein soll..
Was blöd ist, ist das am Gerät Selbst von links nach rechts 35 36 37 hinter einem Deckel aufgedruckt ist und nicht wie auf dem Bild von rechts nach links 37 36 35. Ich hatte es aber zuerst wie im Bild oben angeschlossen. Dadurch hatte ich + vom Digitus an ground am Gerät...Das habe ich aber mit dem Oszilloskop auch gemessen, das Signal wird dann total zerhackt... Ich vermute das dadurch die Geräteschnitstelle eine Macke hat. Den Digitus hatte ich danach neu bestellt.. . Beschuerte nicht eindeutige ABB Doku...
Ein sauberes Modbus Signal am Oszilloskop erhalte ich aber nur wenn ich es so anschließe A+ B- C Ground..

Sag mal die aufgedruckten  5 V auf der Adapterplatine welche beim Digitus mitgeliefert wurde, liegen da bei dir 5V an? Bei mir nämlich auf meinen beiden Adaptern nicht. Auch nicht auf dem Neuen.
Das wäre noch das einzige was ich eventuell ausprobieren würde...Das + Signal mit einem Widerstand gegen 5V anheben. Das - Signal habe ich bereits gegen Ground gezogen... ( Pull Up pull down )
Dadurch wird eine saubere Signalpegel trennung auf dem Bus erreicht... Kann man auf meinen Oszilloskop Messungen gut sehen. Ohne diesen Pill Up und Pull down liegen hier nur läppische 0,4V dazwischen...

wendy2702, 2018-05-17:
Gerade mal gemessen. Adapter am PC angestöpselt und zwischen GND und +5V liegen saubere 5V an.

TomSC1984, 2018-05-17:
könnt ihr mir mal in einem Screenshot zeigen wo die Einstellung für verschiedene slaves ist?

Gruß Thomas

smartboart, 2018-05-17:
> Gerade mal gemessen. Adapter am PC angestöpselt und zwischen GND und +5V liegen saubere 5V an.

wendy2702, 2018-05-18:
Ich kann keinen Screenshot liefern da ich die neue Version noch nicht nutze/benötige.

Läuft der Adapter, die Instanz denn noch bzw. lässt es sich noch starten?

TomSC1984, 2018-05-18:
Ja der adapter läuft noch. Ich hatte auch berucht über iobroker admin install via url das zu machen. Hat er dann auch gemacht. Seit dem kommt der oben genannte fehler. Beim ersten mal liefs durch. Jetzt nicht mehr. Aber adapter läuft :)

smartboart, 2018-05-18:
[image]

TomSC1984, 2018-05-18:
Hm... ich würde ja gerne das nochmal testen bekomme aber auf grund der fehlermeldungen nicjt mehr installiert. Sollte ichbevtl meine vorhandenen instanzen löschen?

ykuendig, 2018-05-19:
Ich misch nun auch mal etwas mit...
Adapter installieren klappt trotz Warnungen. Adapter läuft auch so weit.
Aber: Bei mehreren IDs werden die Werte nicht sauber ausgelesen. Will heissen, einige Felder liest er komischerweise; andere aber nicht.  Im Log kommt das:

```
modbus.0 | 2018-05-19 20:28:08.706 | info | Disconnected from slave
-- | -- | -- | --
modbus.0 | 2018-05-19 20:28:07.711 | warn | Poll error count: 15 code: {"err":"timeout"}
modbus.0 | 2018-05-19 20:28:07.711 | error | Request timed out.
modbus.0 | 2018-05-19 20:28:07.710 | warn | Error: undefined
modbus.0 | 2018-05-19 20:28:02.691 | info | Connected to slave
modbus.0 | 2018-05-19 20:27:02.679 | info | Disconnected from slave
modbus.0 | 2018-05-19 20:27:02.673 | warn | Poll error count: 14 code: {}
modbus.0 | 2018-05-19 20:27:02.577 | warn | Poll error count: 13 code: {}
modbus.0 | 2018-05-19 20:26:52.476 | warn | Poll error count: 12 code: {}
modbus.0 | 2018-05-19 20:26:52.380 | warn | Poll error count: 11 code: {}
modbus.0 | 2018-05-19 20:26:42.269 | warn | Poll error count: 10 code: {}
modbus.0 | 2018-05-19 20:26:42.170 | warn | Poll error count: 9 code: {}
modbus.0 | 2018-05-19 20:26:32.053 | warn | Poll error count: 8 code: {}
modbus.0 | 2018-05-19 20:26:31.944 | warn | Poll error count: 7 code: {}
modbus.0 | 2018-05-19 20:26:21.843 | warn | Poll error count: 6 code: {}
modbus.0 | 2018-05-19 20:26:21.747 | warn | Poll error count: 5 code: {}
modbus.0 | 2018-05-19 20:26:11.646 | warn | Poll error count: 4 code: {}
modbus.0 | 2018-05-19 20:26:11.536 | warn | Poll error count: 3 code: {}
modbus.0 | 2018-05-19 20:26:01.420 | warn | Poll error count: 2 code: {}
modbus.0 | 2018-05-19 20:26:01.307 | warn | Poll error count: 1 code: {}
modbus.0 | 2018-05-19 20:26:01.162 | info | Connected to slave
modbus.0 | 2018-05-19 20:25:59.928 | info | starting. Version 2.0.1 in /opt/iobroker/node_modules/iobroker.modbus, node: v4.8.3
host.raspi3 | 2018-05-19 20:25:57.495 | info | instance system.adapter.modbus.0 started with pid 9824
```

TomSC1984, 2018-06-15:
Hallo, 

ich habe mal mein System neu aufgesetzt und könnte nun wieder testen. Bräuchte nur eine kleine Anleitung dazu. Muss ich als erstes iobroker stoppen? die vorhandenen 2 Adapter löschen? den kompletten Modbus Adapter löschen? 

Bitte um kurze Hilfe wie ich installieren kann falls test gewünscht.
Danke
Gruß Thomas

Apollon77, 2018-06-15:
Gute Frage :-)
Ich habe (per Zufall zeitlich passend) das Thema auch mal zum Testen ins Forum gegeben: [quote=tstueben post_id=155873 time=1529044054 user_id=4352] am besten dort weiter Feedback posten anstelle hier

smartboart, 2018-06-15:
Ich habe einfach den adapter über die admin oberfläche den adapter gelöscht und dann über putty den neuen installiert...
cd /opt/iobroker 
npm i https://github.com/ioBroker/ioBroker.modbus/tarball/refactoring

Hat problemlos funktioniert ...

Apollon77, 2018-06-15:
Bitte jetzt nicht mehr den Branch installieren. Ist jetzt im Master!!

Also  https://github.com/ioBroker/ioBroker.modbus/

TomSC1984, 2018-06-18:
bekomme beim Verbinden leider diesen Fehler. 

`modbus.0	2018-06-18 18:22:33.734	error	Cannot create modbus master!
modbus.0	2018-06-18 18:22:33.733	error	Cannot connect to "192.168.178.50:8899": TypeError: Cannot read property 'timeout' of undefined
modbus.0	2018-06-18 18:22:33.503	info	starting. Version 2.0.3 in /opt/iobroker/node_modules/iobroker.modbus, node: v6.14.3`

woran könnte das liegen?

GermanBluefox, 2018-06-18:
Kannst du hier die Konfig posten? Screenshot?

TomSC1984, 2018-06-19:
[image]
Brauchst du mehr? Wenn ja was genau?

GermanBluefox, 2018-06-19:
Jetzt brauche ich schon mehr.
Ich sehe dass du alte Version hast.
Kannst du updaten?

GermanBluefox, 2018-06-19:
Das ist ein Fehler in der Doku. Gefixt.

TomSC1984, 2018-06-19:
[image]
2018-06-19 16:38:27.954 - info: host.ha-server instance system.adapter.modbus.0 started with pid 16871
--
2018-06-19 16:38:28.281 - debug: modbus.0 objectDB connected
2018-06-19 16:38:28.299 - debug: modbus.0 statesDB connected
2018-06-19 16:38:28.318 - info: modbus.0 starting. Version 2.0.3 in /opt/iobroker/node_modules/iobroker.modbus, node: v6.14.3
2018-06-19 16:38:28.416 - error: modbus.0 Cannot connect to "192.168.178.50:8899": TypeError: Cannot read property 'timeout' of undefined
2018-06-19 16:38:28.416 - error: modbus.0 Cannot create modbus master!

TomSC1984, 2018-06-21:
Hast du ne idee? Ich muss sonst wieder auf die alte version zurück, den aktuell bin ich blind :)

Apollon77, 2018-06-21:
Ich habe mal was versucht. Bitte mal GitHub-Version versuchen.

@GermanBluefox Hier auch ... durchblicke die Struktur noch nicht ganz und gerade nicht viel Zeit zum tieftauchen :-) Warum ist this.options nicht da ?!

Apollon77, 2018-06-21:
Ne lass ... im anderen Thread sind wir nicht wirklich viel weiter gekommen.
@GermanBluefox muss zuerst ran

GermanBluefox, 2018-06-22:
Could analyse the problem. Fix will come tonight.

TomSC1984, 2018-06-23:
Soll ich jetzt neu installieren? Aus welchem branch? Master oder refactoring

GermanBluefox, 2018-06-23:
Mastee

TomSC1984, 2018-06-27:
Jetzt funktioniert alles .Vielen Dank fürs Problem Lösen!

closed 2018-10-08 by GermanBluefox

KanteTaete, 2019-05-12:
Hallo, 
funkfioniert der Adapter jetzt mit mehreren Slaves, die per RTU over Serial angesprochen werden sollen?

@smartboart: Hat das bei Dir mit mehreren ABB-Zählern funktioniert? Ich habe nämlich 3phasige Zähler der gleichen Serie im Einsatz und da kann ich immer nur einen Zähler auslesen.
Siehe auch: https://forum.iobroker.net/post/260933