ioBroker/ioBroker.modbus #12: Rewrite Adapter to jsmodbus 3.x or other library

Apollon77, 2018-05-02:
The modbus adapter uses a mix of jsmodbus and "patched" files and such and uses jsmodbix 1.x as dep ... now jsmodbus is available in 3.x as big step ... we need to rebuild/rewrite the adapter for this I fear

GermanBluefox, 2018-05-03:
I modified jsmodbus completely to support multiple devcie Ids for one connection. It's no more possible to use jsmodbus directly. Because the maintainer of jsmodbus is very strange, I do not plan to merge the changes back to origin repo.

Apollon77, 2018-05-03:
As long as we have jsmodbus 1.2.x as dependency we will never get node10 copmatibility (jsmodbus 1.2.x has serialport 2.x as dependency)!
So we need to remove the dependency then and include directly into lib or such?

GermanBluefox, 2018-05-03:
Done

closed 2018-05-03 by GermanBluefox