ioBroker/ioBroker.modbus #13: "Uncaught exception: handler is not defined" in Version 2.0.3

stabilostick, 2018-06-20:
Das Update der ModBus-Adapters von 1.1.1 auf 2.0.3 führt zum unten stehenden Fehlerbild.

Die Einstellungen, die zum Fehler führen, sind [im ioBroker Forum](https://forum.iobroker.net/viewtopic.php?f=22&t=15044) zu finden.

```
2018-06-20 23:37:29.222  - info: host.ARBEITSZIMMER Restart adapter system.adapter.modbus.2 because enabled
2018-06-20 23:37:59.229  - info: host.ARBEITSZIMMER instance system.adapter.modbus.2 started with pid 17504
2018-06-20 23:37:59.598  - debug: modbus.2 objectDB connected
2018-06-20 23:37:59.608  - debug: modbus.2 statesDB connected
2018-06-20 23:37:59.618  - info: modbus.2 starting. Version 2.0.3 in C:/dev/iobroker/node_modules/iobroker.modbus, node: v8.11.1
2018-06-20 23:37:59.958  - error: modbus.2 uncaught exception: handler is not defined
2018-06-20 23:37:59.958  - error: modbus.2 ReferenceError: handler is not defined
at Object. (C:\dev\iobroker\node_modules\iobroker.modbus\lib\jsmodbus\handler\server\ReadDiscreteInputs.js:71:18)
at Module._compile (module.js:652:30)
at Object.Module._extensions..js (module.js:663:10)
at Module.load (module.js:565:32)
at tryModuleLoad (module.js:505:12)
at Function.Module._load (module.js:497:3)
at Module.require (module.js:596:17)
at require (internal/module.js:11:18)
at Modbus.fs.readdirSync.filter.forEach.file (C:\dev\iobroker\node_modules\iobroker.modbus\lib\jsmodbus\index.js:9:29)
at Array.forEach ()
2018-06-20 23:38:00.990  - error: Caught by controller[0]: ReferenceError: handler is not defined
2018-06-20 23:38:00.990  - error: Caught by controller[0]:     at Object. (C:\dev\iobroker\node_modules\iobroker.modbus\lib\jsmodbus\handler\server\ReadDiscreteInputs.js:71:18)
2018-06-20 23:38:00.990  - error: Caught by controller[0]:     at Module._compile (module.js:652:30)
2018-06-20 23:38:00.990  - error: Caught by controller[0]:     at Object.Module._extensions..js (module.js:663:10)
2018-06-20 23:38:00.990  - error: Caught by controller[0]:     at Module.load (module.js:565:32)
2018-06-20 23:38:00.990  - error: Caught by controller[0]:     at tryModuleLoad (module.js:505:12)
2018-06-20 23:38:00.990  - error: Caught by controller[0]:     at Function.Module._load (module.js:497:3)
2018-06-20 23:38:00.990  - error: Caught by controller[0]:     at Module.require (module.js:596:17)
2018-06-20 23:38:00.990  - error: Caught by controller[0]:     at require (internal/module.js:11:18)
2018-06-20 23:38:00.990  - error:  Caught by controller[0]:     at  Modbus.fs.readdirSync.filter.forEach.file  (C:\dev\iobroker\node_modules\iobroker.modbus\lib\jsmodbus\index.js:9:29)
2018-06-20 23:38:00.990  - error: Caught by controller[0]:     at Array.forEach ()
2018-06-20 23:38:00.990  - error: host.ARBEITSZIMMER instance system.adapter.modbus.2 terminated with code 6 (uncaught exception)
2018-06-20 23:38:00.990  - info: host.ARBEITSZIMMER Restart adapter system.adapter.modbus.2 because enabled
```

Apollon77, 2018-06-21:
This should be fixed by https://github.com/ioBroker/ioBroker.modbus/commit/59fcedbf20e66a26375b7d16ed9c21e6fe3654c0 ... but as discussed in forum then an error with tcp object comes ... I also did slight changes ... it seems that tcp was only suppose to work as slave for now.

@GermanBluefox I stop here to not confuse things ... fr TCP I only found parameters to edit port and bind and bind is called "Partner IP" in Admin ... so all a bit confusing ...

GermanBluefox, 2018-06-25:
Please retest with 2.0.3

stabilostick, 2018-06-25:
Error still not fixed. Sorry.
```
2018-06-25 21:17:59.794 - debug: modbus.2 objectDB connected
2018-06-25 21:17:59.804 - debug: modbus.2 statesDB connected
2018-06-25 21:17:59.814 - info: modbus.2 starting. Version 2.0.4 in C:/dev/iobroker/node_modules/iobroker.modbus, node: v8.11.1
2018-06-25 21:17:59.966 - error: modbus.2 uncaught exception: Cannot read property 'tcp' of undefined
2018-06-25 21:17:59.966 - error: modbus.2 TypeError: Cannot read property 'tcp' of undefined
at init (C:\dev\iobroker\node_modules\iobroker.modbus\lib\jsmodbus\transports\modbus-server-tcp.js:19:36)
at Object. (C:\dev\iobroker\node_modules\iobroker.modbus\lib\jsmodbus\transports\modbus-server-tcp.js:148:9)
at C:\dev\iobroker\node_modules\iobroker.modbus\node_modules\stampit\dist\stampit.js:185:31
at arrayEach (C:\dev\iobroker\node_modules\iobroker.modbus\node_modules\lodash\internal\arrayEach.js:15:9)
at C:\dev\iobroker\node_modules\iobroker.modbus\node_modules\lodash\internal\createForEach.js:15:9
at Factory (C:\dev\iobroker\node_modules\iobroker.modbus\node_modules\stampit\dist\stampit.js:177:48)
at Slave.initValues (C:\dev\iobroker\node_modules\iobroker.modbus\lib\slave.js:149:51)
at adapter.getStates (C:\dev\iobroker\node_modules\iobroker.modbus\lib\slave.js:141:13)
at C:\dev\iobroker\node_modules\iobroker.js-controller\lib\adapter.js:4036:61
at Socket. (C:\dev\iobroker\node_modules\iobroker.js-controller\lib\states\statesInMemClient.js:153:27)
2018-06-25 21:18:00.988 - error: Caught by controller[0]: TypeError: Cannot read property 'tcp' of undefined
2018-06-25 21:18:00.988 - error: Caught by controller[0]: at init (C:\dev\iobroker\node_modules\iobroker.modbus\lib\jsmodbus\transports\modbus-server-tcp.js:19:36)
2018-06-25 21:18:00.988 - error: Caught by controller[0]: at Object. (C:\dev\iobroker\node_modules\iobroker.modbus\lib\jsmodbus\transports\modbus-server-tcp.js:148:9)
2018-06-25 21:18:00.988 - error: Caught by controller[0]: at C:\dev\iobroker\node_modules\iobroker.modbus\node_modules\stampit\dist\stampit.js:185:31
2018-06-25 21:18:00.988 - error: Caught by controller[0]: at arrayEach (C:\dev\iobroker\node_modules\iobroker.modbus\node_modules\lodash\internal\arrayEach.js:15:9)
2018-06-25 21:18:00.988 - error: Caught by controller[0]: at C:\dev\iobroker\node_modules\iobroker.modbus\node_modules\lodash\internal\createForEach.js:15:9
2018-06-25 21:18:00.989 - error: Caught by controller[0]: at Factory (C:\dev\iobroker\node_modules\iobroker.modbus\node_modules\stampit\dist\stampit.js:177:48)
2018-06-25 21:18:00.989 - error: Caught by controller[0]: at Slave.initValues (C:\dev\iobroker\node_modules\iobroker.modbus\lib\slave.js:149:51)
2018-06-25 21:18:00.989 - error: Caught by controller[0]: at adapter.getStates (C:\dev\iobroker\node_modules\iobroker.modbus\lib\slave.js:141:13)
2018-06-25 21:18:00.989 - error: Caught by controller[0]: at C:\dev\iobroker\node_modules\iobroker.js-controller\lib\adapter.js:4036:61
2018-06-25 21:18:00.989 - error: Caught by controller[0]: at Socket. (C:\dev\iobroker\node_modules\iobroker.js-controller\lib\states\statesInMemClient.js:153:27)
2018-06-25 21:18:00.989 - error: host.ARBEITSZIMMER instance system.adapter.modbus.2 terminated with code 6 (uncaught exception)
```

stabilostick, 2018-06-25:
v2.0.4 breaks the passing of the ip address from the configuration dialoge the the jsmodbus-routines. Worked wit 2.0.3. The admin dialog displays the ip adress correctly.

Error now:

```
2018-06-25 21:27:03.848  - info: modbus.1 starting. Version 2.0.4 in C:/dev/iobroker/node_modules/iobroker.modbus, node: v8.11.1
[...]
2018-06-25 21:27:03.992  - info: history.0 enabled logging of modbus.1.inputRegisters.30873_Einverbrauchserhöhung
2018-06-25 21:27:05.066  - error: modbus.1 Socket Error
2018-06-25 21:27:05.067  - error: modbus.1 Client in error state.
2018-06-25 21:27:05.067  - warn: modbus.1 On error: {"code":"ECONNREFUSED","errno":"ECONNREFUSED","syscall":"connect","address":"127.0.0.1","port":502}
2018-06-25 21:28:07.072  - error: modbus.1 Socket Error
2018-06-25 21:28:07.072  - error: modbus.1 Client in error state.
2018-06-25 21:28:07.072  - warn: modbus.1 On error: {"code":"ECONNREFUSED","errno":"ECONNREFUSED","syscall":"connect","address":"127.0.0.1","port":502}
[...]
```

stabilostick, 2018-06-26:
There seems to be more to do. File slave.js, line 150 should be changed to something like 
```
     options: {
     tcp: {
            hostname: '127.0.0.1',  //??????
            port: parseInt(options.config.tcp.port, 10) || 502
     }},
```
to get rid of the "tcp" is undefined-error. But then there is line 163 `.compose(server.tcp.complete)`
with "server" is undefined...

GermanBluefox, 2018-06-28:
Slave works now by me.

stabilostick, 2018-06-28:
Sorry, no communication here. More information at [](https://forum.iobroker.net/viewtopic.php?f=22&t=15044&p=158496#p158496)

closed 2018-10-08 by GermanBluefox