ioBroker/ioBroker.modbus #16: Object naming issues within HoldingRegisters

FChristian, 2018-06-24:
Naming Convention for objects seems to follow rule ADDRESS_NAME. Like we can see here for example:
[image]

Within the HoldingRegisters, this doesn't seem to work, here it is only NAME:
[image]

I manually created the 300 new and gave it manually the name ADDRESS_NAME:
[image]

This leads to a new object called ADDRES_ADDRESS_NAME:
[image]

Funny more: EVERY(!) Object within the same instance is renamed to ADDRESS_ADDRESS_NAME:
[image]

Although the instance configuration has not changed (see example here):
[image]

More Info available here: https://forum.iobroker.net/viewtopic.php?f=20&t=15096&p=157812

Apollon77, 2018-06-25:
In former modbus versions name was completely free and so up to you ifyou added the ID before or not - but when generating the relas Object-ID the number was always prepending.

SO: you can enter whatever name you want in the name field

FChristian, 2018-06-25:
Did a full re-install. Now works fine, so this is not really an issue. Thank you!

closed 2018-06-25 by FChristian