ioBroker/ioBroker.modbus #17: Modbus (TCP?) broken with 2.0.4

JackWolfskin42, 2018-06-26:
SInce this update both modbus (TCP) instances stopped working:
`

modbus.0 | 2018-06-26 12:57:40.416 | warn | On error: {"code":"ECONNREFUSED","errno":"ECONNREFUSED","syscall":"connect","address":"127.0.0.1","port":502}
-- | -- | -- | --
modbus.0 | 2018-06-26 12:57:40.416 | error | Client in error state.
modbus.0 | 2018-06-26 12:57:40.414 | error | Socket Error
modbus.1 | 2018-06-26 12:57:29.935 | warn | On error: {"code":"ECONNREFUSED","errno":"ECONNREFUSED","syscall":"connect","address":"127.0.0.1","port":502}
modbus.1 | 2018-06-26 12:57:29.935 | error | Client in error state.
modbus.1 | 2018-06-26 12:57:29.934 | error | Socket Error

`
The addresses 127.0.0.1 are obviously wrong !

stabilostick, 2018-06-26:
File "modbus-client-tcp.js", line 25 must be
tcp.host = tcp.host || 'localhost';
instead of
tcp.host = tcp.bind || 'localhost';
to fix the IP error.

JackWolfskin42, 2018-06-27:
Fixed with 2.0.6 !

closed 2018-06-27 by JackWolfskin42