ioBroker/ioBroker.modbus #19: individual timeout settings per device

adavt, 2018-06-27:
It would be useful to have a possibility of individual control of frequency of poll of different devices on the tire in the multi ID mode

GermanBluefox, 2018-07-06:
https://github.com/contactless/wb-mqtt -serial take some from here

cburghardt, 2018-11-23:
I have a concrete use case for that. I am using modbus tcp to connect to Solaredge inverter. There is a master and salve inverter, querying the master is fast (poll interval a second) but to query the slave takes a lot more time. So I have set the timeout already to an enormous value but it still fails every 15 mins or so. If I have a chance to influence the devices separately this would help.

GermanBluefox, 2021-06-14:
You can install different instances and set the timeouts separatly

cburghardt, 2021-06-14:
Different devices don't work when the connection has to go via the master. 
Only the master accepts the modbus connection.

GermanBluefox, 2021-06-14:
How many devices IDs do you have?