ioBroker/ioBroker.modbus #21: Boolean state true/false swapped with version 2.0.6?

stabilostick, 2018-07-02:
Ich nutze zur Heizungssteuerung eine UVR16x2 von TA. Der aktuelle Status meiner Heizungspumpen soll an den ioBroker übergeben werden. Der modbus-Adapter ist Slave und bekommt die Daten über TCP vom UVR-Master. Der modbus-Adapter hat die Version 2.0.6.

Bei der UVR sehe ich den boolschen Status eine Heizungspumpe als Text "EIN" oder "AUS". Dabei betrachte ich den Modbus-Ausgangskonfigurationsdialog. Dort wird der aktuell übertragene Wert jeweils angezeigt.

Das zugehörige modbus-Objekt im ioBroker (egal ob vom Typ state oder value) zeigt als Text immer "true" oder "false" an. Die Coils sind dort als "Diskrete Ausgänge" definert. 

Das Objekt zeigt "true" an, wenn die Heizungssteuerung "AUS" anzeigt und "false", wenn die UVR den Wert "EIN" liefert.

**Vertauscht der ioBroker.modbus-Adapter true und false?!** 

Mit Wireshark sehe ich auf dem Netzwerkinterface des ioBrokers die folgenden Datenflüsse (die Modbus-Coil-Adresse ist die 2 und der Function-Code ist 5):

Übergang in der UVR von "AUS" nach stabilen "EIN":
```
modbus-Adapter empfängt "00 00 00 00 00 06 02 05 00 01 ff 00"
und sendet dann gleich  "00 00 00 00 00 06 02 05 00 01 00 00" 
zurück zur UVR
```

Übergang in der UVR von "EIN" nach stabilen "AUS":
```
modbus-Adapter empfängt "00 00 00 00 00 06 02 05 00 01 00 00"
und sendet dann gleich  "00 00 00 00 00 06 02 05 00 01 ff 00" 
zurück zur UVR
```

stabilostick, 2018-07-02:
Forum:  https://forum.iobroker.net/viewtopic.php?f=20&t=15202

digidax, 2019-10-28:
still open in Version 3.0.0

Apollon77, 2020-06-11:
still open

closed 2020-09-17 by stale[bot]