ioBroker/ioBroker.modbus #32: Add possibility to select self named device (udev rule) like Smartmeter Adapter

wendy2702, 2019-01-10:
Please add the possibility to select a self named device (via UDEV rule) like it is handled at Smartmeter Adapter.

TbsJah, 2020-11-19:
Is the selection by name planned (named via UDEV rule)?
I have connected 2 devices via S0 RS485. 
After each reboot the USB ports are randomly assigned.
It would be very helpful to assign a name here ( similar to Smartmeder Adapter)

Apollon77, 2020-11-19:
Planned in form of "when someone finds time", yes.

What you can do as workaround is to edit directly the "system.adapter.modbus.X" objects and enter that "by id" path directly there. Only editing adapter settings via admin might be problematic after that

TbsJah, 2020-11-20:
Thank you very much for your answer (and by the way for your great work)
Just to be sure what we are talking about
Your solution seems to be an easy way to choose the right USB port

[image]

Apollon77, 2020-11-20:
basically yes ... easier to edit when you use the Raw tab in the dialog of your screenshot

TbsJah, 2020-11-21:
Thanks a lot!
I can confirm that this workaround works

GermanBluefox, 2021-06-14:
In 3.4.0 you can enter your port manually

closed 2021-06-16 by GermanBluefox