ioBroker/ioBroker.modbus #39: Add a adjustable waiting time between response from RS485 slave and new Master Query Message

Lichte12, 2019-02-28:
Hi,
some Modbus-Slaves need a waiting time between response from the Slave and the new Querry message from the Master.
I think the modbus adapter is sometimes too fast and thats why the communication is terminated by timeout.
This happens when i read multiple registers in different adresses.
Maybe a adjustable waiting time could solve this problem.

Also see a screenshot from the Daikin Modbus Manual:
[image]

Also discussed here:
[https://forum.iobroker.net/topic/1259/modbus/496](url)

darkbrain85, 2019-03-28:
This could be the solution for all the timeout problems in the past. Maybe a hard coded waiting time can already fix this.

KanteTaete, 2019-04-30:
> 
> 
> This could be the solution for all the timeout problems in the past. Maybe a hard coded waiting time can already fix this.

What do you mean with "hard coded waiting"? - (Where) can this be changed?

darkbrain85, 2019-04-30:
It can not be changed at the moment. Someone with dev skills has to look at it...

darkbrain85, 2019-05-15:
I have done some investigation on this. I installed a Modbus TCP -> RTU Gateway on a Raspberry Pi mit a Software called "mbusd".

[https://github.com/3cky/mbusd](url)

In the Configuration of this Software, there are this options:

`# Maximum number of request retries
retries = 3

# Pause between requests in milliseconds
pause = 100

# Response wait time in milliseconds
wait = 500`

Iobroker ist now communicating with this Gateway and the timing problems are gone so far. I think, this timing options should implemented in the Biobroker Modbus adapter to solve all the connection issues. 

Specially the "response wait time" ist interesting. Is this not exactly what is mentioned here?

@GermanBluefox can you look at this to solve this issue? I think it will help a lot of people here.

GermanBluefox, 2021-06-14:
In 3.4.0 the Read interval was added.

Lichte12, 2021-06-16:
Tested with 3.4.2
The delay is working well. Thanks for this.

closed 2021-06-16 by Apollon77