ioBroker/ioBroker.modbus #41: Periodical Disconnected from slave

viallos, 2019-03-22:
Hi there

I defined connection to my Alarm Modbus TCP protocol and it is working fine, but from time to time I loose connection and can see following in the Modbus log.

modbus.0 | 2019-03-22 10:08:34.356 | info | Connected to slave 192.168.0.100
modbus.0 | 2019-03-22 10:07:25.325 | info | Disconnected from slave 192.168.0.100
modbus.0 | 2019-03-22 10:07:24.326 | warn | Poll error count: 1 code: {"err":"timeout"}
modbus.0 | 2019-03-22 10:07:24.325 | error | Client in error state.
modbus.0 | 2019-03-22 10:07:24.325 | error | Request timed out.
modbus.0 | 2019-03-22 10:07:24.325 | warn | Error: undefined
modbus.0 | 2019-03-22 10:07:12.654 | info | Connected to slave 192.168.0.100

Is there anything I can do to view more detailed log? Especially about that

 warn | Error: undefined

Maybe I should change some connection values like Pool delay or Read Timeout?
Any ideas?

At them moment I have default vales.

Thanks

viallos, 2019-03-22:
I also see following sequences:

modbus.0 | 2019-03-22 10:29:06.903 | info | Disconnected from slave 192.168.0.100
-- | -- | -- | --
modbus.0 | 2019-03-22 10:29:05.907 | warn | Poll error count: 1 code: {"err":"timeout"}
modbus.0 | 2019-03-22 10:29:05.907 | error | Client in error state.
modbus.0 | 2019-03-22 10:29:05.907 | error | Request timed out.
modbus.0 | 2019-03-22 10:29:05.905 | warn | Error: undefined
modbus.0 | 2019-03-22 10:28:29.976 | info | Connected to slave 192.168.0.100
modbus.0 | 2019-03-22 10:28:18.947 | warn | On error: {"errno":"ETIMEDOUT","code":"ETIMEDOUT","syscall":"connect","address":"192.168.0.100","port":502}
modbus.0 | 2019-03-22 10:28:18.947 | error | Client in error state.
modbus.0 | 2019-03-22 10:28:18.947 | error | Socket Error
modbus.0 | 2019-03-22 10:27:46.945 | warn | On error: {"errno":"ETIMEDOUT","code":"ETIMEDOUT","syscall":"connect","address":"192.168.0.100","port":502}
modbus.0 | 2019-03-22 10:27:46.945 | error | Client in error state.
modbus.0 | 2019-03-22 10:27:46.944 | error | Socket Error
modbus.0 | 2019-03-22 10:27:15.940 | info | Disconnected from slave 192.168.0.100

Signum, 2019-04-12:
Same here. A few polls work then I get a timeout again.

```
2019-04-12 13:58:43.639  - debug: modbus.0 Poll device 71
2019-04-12 13:58:43.639  - debug: modbus.0 Poll holdingRegs DevID(71) address 6 - 27 bytes
2019-04-12 13:58:43.891  - debug: modbus.0 Poll holdingRegs DevID(71) address 56 - 1 bytes
2019-04-12 13:58:43.899  - debug: modbus.0 Poll holdingRegs DevID(71) address 220 - 2 bytes
2019-04-12 13:58:43.910  - debug: modbus.0 Poll holdingRegs DevID(71) address 420 - 10 bytes
2019-04-12 13:58:46.048  - debug: modbus.0 Poll device 71
2019-04-12 13:58:46.048  - debug: modbus.0 Poll holdingRegs DevID(71) address 6 - 27 bytes
2019-04-12 13:58:51.052  - warn: modbus.0 Error: undefined
2019-04-12 13:58:51.052  - error: modbus.0 Request timed out.
2019-04-12 13:58:51.052  - error: modbus.0 Client in error state.
2019-04-12 13:58:51.052  - warn: modbus.0 Poll error count: 1 code: {"err":"timeout"}
2019-04-12 13:58:51.053  - debug: modbus.0 Socket closed with error
2019-04-12 13:58:51.053  - debug: modbus.0 Clearing timeout of the current request.
2019-04-12 13:58:51.053  - debug: modbus.0 Cleaning up request fifo.
2019-04-12 13:58:51.056  - silly: modbus.0 inMem message  modbus.0.* modbus.0.info.connection val=false, ack=true, ts=1555070331055, q=0, from=system.adapter.modbus.0, lc=1555070331055
2019-04-12 13:58:52.053  - debug: modbus.0 Closing client on purpose.
2019-04-12 13:58:52.053  - info: modbus.0 Disconnected from slave 192.168.13.8
2019-04-12 13:58:52.055  - silly: modbus.0 inMem message  modbus.0.* modbus.0.info.connection val=false, ack=true, ts=1555070332053, q=0, from=system.adapter.modbus.0, lc=1555070331055
2019-04-12 13:58:57.065  - info: modbus.0 Connected to slave 192.168.13.8
```

closed 2020-06-18 by stale[bot]

geek-at, 2023-02-28:
I'm having the same problem. Did you ever solve it @viallos @Signum ?