ioBroker/ioBroker.modbus #46: Problem with more than one slave

stener-io, 2019-05-06:
Reading from one slave (ID:2) works without fine, from the second slave (ID:3) or more slave there is a timeout.

2019-05-06 13:15:58.071  - info: modbus.0 starting. Version 2.0.9 in /opt/iobroker/node_modules/iobroker.modbus, node: v8.16.0
2019-05-06 13:15:58.526  - debug: modbus.0 connect to serial /dev/ttyUSB0 with 2400
2019-05-06 13:15:58.555  - info: modbus.0 Connected to slave
2019-05-06 13:15:58.560  - debug: modbus.0 Poll device 2
2019-05-06 13:15:58.565  - debug: modbus.0 Poll inputRegs DevID(2) address 0 - 2 bytes
2019-05-06 13:15:58.743  - debug: modbus.0 Poll inputRegs DevID(2) address 6 - 2 bytes
2019-05-06 13:15:58.908  - debug: modbus.0 Poll inputRegs DevID(2) address 12 - 2 bytes
2019-05-06 13:15:59.069  - debug: modbus.0 Poll inputRegs DevID(2) address 342 - 2 bytes
2019-05-06 13:15:59.237  - debug: modbus.0 Poll device 3
2019-05-06 13:15:59.238  - debug: modbus.0 Poll inputRegs DevID(3) address 0 - 2 bytes
2019-05-06 13:16:00.052  - debug: modbus.0 system.adapter.admin.0: logging true
2019-05-06 13:16:04.244  - warn: modbus.0 Error: undefined
2019-05-06 13:16:04.245  - error: modbus.0 Request timed out.
2019-05-06 13:16:04.246  - warn: modbus.0 Poll error count: 1 code: {"err":"timeout"}
2019-05-06 13:16:05.249  - info: modbus.0 Disconnected from slave

stener-io, 2019-05-06:
Another info, if the Device ID in the General Adapter Configuration is not equal, there is the following log entry 

2019-05-06 14:21:14.314  - info: modbus.0 starting. Version 2.0.9 in /opt/iobroker/node_modules/iobroker.modbus, node: v8.16.0
--
2019-05-06 14:21:14.745  - debug: modbus.0 connect to serial /dev/ttyUSB0 with 2400
2019-05-06 14:21:14.764  - info: modbus.0 Connected to slave
2019-05-06 14:21:14.766  - debug: modbus.0 Poll device 2
2019-05-06 14:21:14.769  - debug: modbus.0 Poll inputRegs DevID(2) address 0 - 2 bytes
2019-05-06 14:21:14.932  - debug: modbus.0 received answer for wrong ID 2, expected 0
2019-05-06 14:21:14.940  - debug: modbus.0 Poll inputRegs DevID(2) address 6 - 2 bytes
2019-05-06 14:21:15.104  - debug: modbus.0 received answer for wrong ID 2, expected 0
2019-05-06 14:21:15.106  - debug: modbus.0 Poll inputRegs DevID(2) address 12 - 2 bytes
2019-05-06 14:21:15.265  - debug: modbus.0 received answer for wrong ID 2, expected 0
2019-05-06 14:21:15.266  - debug: modbus.0 Poll inputRegs DevID(2) address 342 - 2 bytes
2019-05-06 14:21:15.428  - debug: modbus.0 received answer for wrong ID 2, expected 0
2019-05-06 14:21:15.435  - debug: modbus.0 Poll device 3
2019-05-06 14:21:15.435  - debug: modbus.0 Poll inputRegs DevID(3) address 0 - 2 bytes
2019-05-06 14:21:20.442  - warn: modbus.0 Error: undefined
2019-05-06 14:21:20.443  - error: modbus.0 Request timed out.
2019-05-06 14:21:20.444  - warn: modbus.0 Poll error count: 1 code: {"err":"timeout"}
2019-05-06 14:21:21.447  - info: modbus.0 Disconnected from slave

stener-io, 2019-05-18:
see also: https://forum.iobroker.net/topic/22126/adapter-modbus-mehrere-ger%C3%A4te-ids-timeout-beim-auslesen-von-mehr-als-einen-teilnehmern/6

BlackBird77, 2019-09-13:
My Version has this Problem fixed! It think it was a timing issue.
https://github.com/BlackBird77/ioBroker.modbus

Apollon77, 2019-09-14:
Would you provide a PR?

BlackBird77, 2019-09-14:
Yes i can create a PR after a little bit more testing.

Now i have some timeouts every 120 minutes. But this timeouts recover
perfectly to a working state automaticaly, without a manual Adapter
restart.

I want to try the default settings first.

Currently i use:
Poll delay: 1000
Reconnect time: 10000
Read timeout: 4000
Pulsetime: 1000
Max Read length float: 30
Max Read length bool: 30

Best regards
Andre

BlackBird77, 2019-09-28:
> Would you provide a PR?

https://github.com/ioBroker/ioBroker.modbus/pull/52

ok PR created!

Apollon77, 2019-09-28:
Cool. Thank you. I check soon

closed 2020-06-11 by Apollon77