ioBroker/ioBroker.modbus #48: Graceful disconnect

cburghardt, 2019-06-07:
My client (Solaredge inverter) regularly runs into a timeout when it is handling other requests. This seems to be normal modbus behaviour (collision detection).
I therefore have modified the flush function modbus-client-core.js like that:

```
                currentRequest.timeout = setTimeout(() => {
                    currentRequest.cb && currentRequest.cb();
                    this.log.warn('Request timed out');
                    this.setState('ready');
                }, this.options.timeout);
```

This way it does not disconnect but simply retries. This gives a much smoother behaviour.

Apollon77, 2019-06-07:
But should we then not have kind of a counter that make sure that a real reconnect happens when it stays timing out?

cburghardt, 2019-06-07:
You already have this in pollResult, the error count still works. So after the max number of errors it disconnects and connects again.

In addition I have pimped the modbus to also change the polling interval based upon daylight but that's probably a very specific use case ;)

closed 2020-06-18 by stale[bot]