ioBroker/ioBroker.modbus #54: Write Multiple Registers

TmShaz, 2019-11-20:
Trying to Change  a value in Holding Register  expacting "write multiple registers" leads to "unknown function code: 134"

Version 2.0.9

TmShaz, 2019-11-20:
Sollution of

TmShaz, 2019-11-20:
Set maxBlock to 1 doesn't work.
#23

GermanBluefox, 2020-06-05:
It is strange. You are trying to write a value, that is longer than 16 bit, but the slave says, that it does not support it. You can avoid this problem by not using the values longer than 16 bit.

TmShaz, 2020-06-25:
I don't know but I think the value are not longer than 16bit. I'm trying to write a "6" or "16". Unsigned 16bit (big endian)

TmShaz, 2020-06-25:
I try to use the checkbox "Do not use multiple registers". But it switches back to unchecked after leaving the configuration. Version 3.1.4

clausob, 2020-07-12:
I have the same problem with a MAICO WS 320. Reading holding register works, writing one uint16 fails with error
132 or 134 depending on the setup. Maico expects code 16. Tried maxBlock=1 and "Do not use multiple registers" 
(was changed in the config file even though I have the same behavior than TmShaz described above).

Added Screenshot of my configuration, Version 3.1.4
[image]

Am I missing something in my configuration?

Thanks a lot in advance,

Claus

mcmuller, 2020-09-11:
same sort of problem here, trying to write  Holding Register to a ELIWELL Heatpump that only accepts single writes. "Do not use multiple registers" switches back to unchecked when saved.
Question: Where to find "maxBlock" option? 
[image]

TmShaz, 2020-09-25:
Is there an option to force "write multiple registers (FC16)" even if the value doesn't need it? The Slave I'm using only accepts FC16.

clausob, 2020-09-25:
Hi,

I changed it in the sourcecode to force FC16 when the option is set, I 
hope that I found the right change:

diff -r ioBroker.modbus-3.1.4/lib/master.js 
/opt/iobroker/node_modules/iobroker.modbus/lib/master.js
407c407,408
<             if (objects[id].native.len > 1 && 
!options.config.doNotUseWriteMultipleRegisters) {
---
 >             //if (objects[id].native.len > 1 && 
!options.config.doNotUseWriteMultipleRegisters) {
 >             if ( !options.config.doNotUseWriteMultipleRegisters) {

Hope this helps,

Claus

TmShaz, 2020-09-25:
Hi Claus,
this seems to work for me too. Thank you.
But it makes no difference if i check or uncheck the option "Do not use multiple registers". It still switches back to unchecked.

Now someone has to check, if this change make sense for the next release?

clausob, 2020-10-13:
Hi,

my workaround foces to use "write multiple register" which is of course 
not a solution. It would be beneficial to have an option

"force use of write multiple register" in the user-interface. And second 
topic is that the switch "Do not use multiple registers"

does not work.

Best regards,

Claus

TmShaz, 2020-10-17:
Hi Claus, completely agree. This should be an option.

@GermanBluefox can you consider this for next release?

Apollon77, 2020-11-12:
then lets consider thjis as duplicate of #78

closed 2020-11-12 by Apollon77