ioBroker/ioBroker.modbus #58: Abfangen von unrealistisches Werten

mifricke, 2020-03-02:
Einige Modbus Geräte liefern in besonderen Betriebszuständen keine Plausiblen Werte. 
Beispielsweise liefern SMA Wechselrichter ohne Sonneneinstrahlung riesige Werte (z.B. 2147483648) statt 0. Das ganze kann auch ein negativer Wert sein.

Es wäre gut, wenn man dies direkt im Modbus Adapter pro Register abfangen könnte. Beispielsweise mit einem Maximal- und Minimalwert. Bei überschreiten dieser Grenzen sollte der Wert immer "0" sein. 

Aktuell geht das nur relativ "Aufwendig" über einen zweiten State und Javascript.

GermanBluefox, 2020-06-05:
Das zu implementieren ist kein grosses Problem, aber GUI ist aufwendig....

GermanBluefox, 2021-06-12:
Man kann Formel dafür verwenden. Z.b: `x > 2000000 ? null : x`

closed 2021-06-12 by Apollon77