ioBroker/ioBroker.modbus #60: Gerät erfordert 0.0.0.0 als IP und neue node JS

gidhap, 2020-03-28:
Hallo, ich nutze ein CMI interface von "[Technische Alternative](https://www.ta.co.at/en/)". Wenn ich dieses über Modbus TCP auslesen möchte klappt das nur mit der IP 0.0.0.0 und mit dem neuen Node.JS (z.B. 13.7.0). Die IP 0.0.0.0 ist schon sehr seltsam, und leider scheint der 13.7.0. instabil zu sein wodurch mein Javascript immer wieder einfriert. Wenn ich auf den empfohlenen 10.19.0 gehe, verbindet sich das Gerät nicht mehr (Instanz bleibt rot).  Alle anderen Modbus Instanzen funktionieren weiterhin mit der empfohlenen Version. Wie kann ich das Problem am besten einkreisen? Protokoll trace mit wireshark könnte ich zwar versuchen...bräuchte dann aber jemanden der das auch interpretieren kann...

Apollon77, 2020-03-28:
Nodejs 13 ist entwicklungsversion und daher nicht zu empfehlen. Es muss auch mit Versionen davor gehen. 

Bind auf 0.0.0.0 für den Server bedeutet das er auf alle ip Adressen hört. 

Zum Fehler: was sagt denn das log file? 

Das issue ist ohne genaue Fehler nicht wirklich zu bewerten.

gidhap, 2020-03-28:
So sieht das Log aus wenn ich die V10.19.0 verwende
`

modbus.2 | 2020-03-28 16:17:38.558 | info | (1547) Terminated (NO_ERROR): Without reason
-- | -- | -- | --
modbus.2 | 2020-03-28 16:17:38.558 | info | (1547) terminating
modbus.2 | 2020-03-28 16:17:38.545 | error | at process._tickCallback (internal/process/next_tick.js:63:19)
modbus.2 | 2020-03-28 16:17:38.545 | error | at doListen (net.js:1461:7)
modbus.2 | 2020-03-28 16:17:38.545 | error | at listenInCluster (net.js:1328:12)
modbus.2 | 2020-03-28 16:17:38.545 | error | at Server.setupListenHandle [as _listen2] (net.js:1263:19)
modbus.2 | 2020-03-28 16:17:38.545 | error | (1547) Error: listen EACCES: permission denied 0.0.0.0:502
modbus.2 | 2020-03-28 16:17:38.544 | error | (1547) uncaught exception: listen EACCES: permission denied 0.0.0.0:502
modbus.2 | 2020-03-28 16:17:37.936 | warn | (1547) Serial is not available
modbus.2 | 2020-03-28 16:17:37.856 | info | (1547) starting. Version 3.0.0 in /opt/iobroker/node_modules/iobroker.modbus, node: v10.19.0
modbus.2 | 2020-03-28 16:17:14.171 | info | (1055) Terminated (START_IMMEDIATELY_AFTER_STOP): Without reason

`

gidhap, 2020-03-28:
Wenn ich die 13.7.0. verwende
`

modbus.2 | 2020-03-28 16:51:33.824 | warn | (6769) Serial is not available
-- | -- | -- | --
modbus.2 | 2020-03-28 16:51:33.778 | info | (6769) starting. Version 3.0.0 in /opt/iobroker/node_modules/iobroker.modbus, node: v13.7.0

`

Apollon77, 2020-03-28:
Dann fehlen die Rechte das der nodejs Prozess auf ports <1024 binden darf. 

Bitte installation fixer laufen lassen. Der sollte das fixen. Was nodejs 13 da anders Macht weiß ich nicht.

gidhap, 2020-03-28:
Perfekt, das war die Lösung.
Danke !

closed 2020-03-28 by gidhap

gidhap, 2020-03-28:
@Apollon77 Wie kann ich denn einen einmal Betrag per PP an Dich Spenden ?

Apollon77, 2020-03-28:
Super. Freut mich das es tut. Schau mal auf mein GitHub Profil. Da ist der Link zu GitHub sponsors ... damit geht das glaube. Alternativ kann ich dir noch Paypal geben ;-) Danke!!

gidhap, 2020-03-28:
> 
> 
> Super. Freut mich das es tut. Schau mal auf mein GitHub Profil. Da ist der Link zu GitHub sponsors ... damit geht das glaube. Alternativ kann ich dir noch Paypal geben ;-) Danke!!

Da kann ich nur dauerhaft spenden. Ist aber nicht so meins, ich spende lieber "Fallbezogen", PP wäre also gut :)

Apollon77, 2020-03-30:
schick mir mal ne mail an iobroker@fischer-ka.de. Danke!