ioBroker/ioBroker.modbus #61: Kann "coil" Status nicht korrekt auslesen

gidhap, 2020-03-31:
Hallo Ich kämpfe mit den Einstellungen der "coils" beim Modbus Adapter (V 3.0.0)
Mein Modbus Device (CMI) stellt analoge Werte wie auch Schaltzustände zur Verfügung die ich mit dem ioBroker auswerte und visualisiere. Die Analogen Werte werden sauber übermittelt (ich nutze Holding Register) und auch in der vis problemlos dargestellt.

Bei den Schaltzuständen versuche ich nun über die Diskreten Eingänge mit Rolle "State" zu gehen, komme aber ich nicht weiter. Es wird als Objektzustand im ioBroker immer "false" angezeigt. Nun habe ich mal mit Wireshark ins Modbus Protokoll geschaut und versucht damit schlauer zu werden. Hat nur bedingt geklappt.

Steht der entsprechende Ausgang auf "AUS", sehe ich im Protokoll, daß "Data: 0000" an den ioBroker geschickt wird, das Objekt im ioB steht auf "FALSE"
Ist der Ausgang "EIN", sehe ich im Protokoll, daß "Data: ff00" an den ioBroker geschickt wird, das Objekt im ioB steht unverändert auf "FALSE"
[image]
[image]
Ich hoffe dass ich die Daten hier richtig interpretiere und das nur eine Kleinigkeit ist die ich irgendwo falsch eingestellt habe !?
Danke für jede Unterstützung

gidhap, 2020-06-14:
@GermanBluefox  Hi, took me a long time to come back on this one. Today I tested the V 3.1.3, but same issue. 
ff00 is not recognized as "true", object in ioBroker remains on state "false". 
I am using "Diskrete Eingänge" and Role "State" hope this is still the way to go!?

Anything I can do to support  ? Happy to do so. ( I am not a coder ;o ) 

Node.js v10.19.0
NPM 6.13.4
Modbus Adapter 3.1.3.

compton-git, 2020-09-11:
@gidhap 
Apart from an inversion of the state itself also the adresses of the coils seem to be rotated.
IOBroker, Modbus-Slave adress -> CMI Modbus Master adress
Coil-adress 1 -> CMI Adress 15
Coil-adress 2 -> CMI Adress 14
The connection works for me with these setting then, but looks like the implementation for the adresses still needs to be fixed.

closed 2020-12-31 by stale[bot]