ioBroker/ioBroker.modbus #62: Poll Error bei mehrere Geräten (Slave)

NemoMarkus, 2020-04-01:
Ich habe 3 Modbus Stromzähler SDM120 von der Firma EASTRON bei mir zuhause verbaut und wollte diese über den Modbus Adapter in ioBroker auslesen. 
 Ich habe dazu einen ATC-1000 TCP/IP to RS485 Converter verwendet.

Systeminformation ioBroker
NPM: 6.13.4
Node.js: v10.18.1
Modbus Adapter: 3.0.0
 
Wenn ich im Adapter nur einen Stromzähler konfiguriert funktioniert alles wie es soll. Man sieht im Log zwar immer folgende Meldung aber das Auslesen funktioniert.

(10865) received answer for wrong ID 11, expected 1
(10865) Sending pdu to the   socket.
(10865) Poll inputRegs   DevID(11) address 70 - 4 bytes
(10865) received answer for   wrong ID 11, expected 1
(10865) Sending pdu to the   socket.
(10865) Poll inputRegs   DevID(11) address 0 - 32 bytes
(10865) Poll device 11

 Konfiguriere ich nun die anderen 2 Stromzähler fällt der Adapter immer wieder aus und ich erhalte  Timeouts, daraufhin startet der Adapter neu.
 

(10865) Disconnected from slave
(10865) Poll error count: 1 code:   {"err":"timeout"}
(10865) Client in error state.
(10865) Request timed out.
(10865) Error: undefined
(10865) Connected to slave
(10865) Disconnected from slave

Dieses Verhalten habe ich auch mit anderen Modbus Geräten nicht nur mit den Stromzählern.
Zu Testzwecken habe ich auch einen USB to RS485 Konverter versucht, dasselbe Verhalten.
Es spielt auch keine Rolle ob ich Eingangsregister oder Holding Register auslese.

Könnt ihr mi helfen. Habe ich etwas falsch konfiguriert oder bin ich hier auf einen BUG gestoßen

mp84, 2020-04-14:
Hallo Markus,

ich kann das von dir beschriebene Verhalten bestätigen und bin ebenfalls an einer Lösung interessiert.

GermanBluefox, 2020-06-05:
Please test 3.0.3

closed 2020-09-11 by stale[bot]