ioBroker/ioBroker.modbus #66: Keine Serielle Verbindung mit 3.0.1

lastoks, 2020-04-14:
Der Modbus Adapter bleibt gelb und verbindet sich nicht. Mit 3.0.0 und gleichen Einstellungen funktioniert die Verbindung, ist allerdings auf dauer nicht stabil.

Hier der Log:
`modbus.0	2020-04-13 12:01:57.100	info	(1126) starting. Version 3.0.1 in /opt/iobroker/node_modules/iobroker.modbus, node: v10.20.0
modbus.0	2020-04-13 12:01:56.275	debug	(1126) statesDB connected
modbus.0	2020-04-13 12:01:56.272	debug	(1126) States connected to redis: 192.168.0.117:9000
modbus.0	2020-04-13 12:01:56.248	debug	(1126) States create System PubSub Client
modbus.0	2020-04-13 12:01:56.236	debug	(1126) States create User PubSub Client
modbus.0	2020-04-13 12:01:56.216	debug	(1126) Redis States: Use Redis connection: 192.168.0.117:9000
modbus.0	2020-04-13 12:01:56.207	debug	(1126) objectDB connected
modbus.0	2020-04-13 12:01:56.183	debug	(1126) Objects connected to redis: 192.168.0.117:9001
modbus.0	2020-04-13 12:01:56.145	debug	(1126) Objects client initialize lua scripts
modbus.0	2020-04-13 12:01:56.142	debug	(1126) Objects create User PubSub Client
modbus.0	2020-04-13 12:01:56.138	debug	(1126) Objects create System PubSub Client
modbus.0	2020-04-13 12:01:56.122	debug	(1126) Objects client ready ... initialize now
modbus.0	2020-04-13 12:01:55.991	debug	(1126) Redis Objects: Use Redis connection: 192.168.0.117:9001
host.iobroker2	2020-04-13 12:01:53.219	info	instance system.adapter.modbus.0 started with pid 1126
host.iobroker2	2020-04-13 12:01:51.344	info	instance system.adapter.modbus.0 terminated with code 156 (156)
host.iobroker2	2020-04-13 12:01:50.682	info	stopInstance system.adapter.modbus.0 send kill signal
modbus.0	2020-04-13 12:01:50.654	info	(1111) Terminated (START_IMMEDIATELY_AFTER_STOP): Without reason
modbus.0	2020-04-13 12:01:50.648	info	(1111) terminating
modbus.0	2020-04-13 12:01:50.640	info	(1111) Got terminate signal TERMINATE_YOURSELF`

Apollon77, 2020-05-29:
solved or still current?

BigBadUlf, 2020-06-02:
not solved. Same in Version 3.0.2.

AlfredSch, 2020-06-02:
not solved. Same in Version 3.0.2.

GermanBluefox, 2020-06-05:
Please test 3.0.3

BigBadUlf, 2020-06-05:
Version 3.0.3 works fine.
The modbus objects are updatet.

Here my log:
modbus.0 | 2020-06-05 09:42:40.846 | info | (4403) Connected to slave 192.168.178.86
-- | -- | -- | --
modbus.0 | 2020-06-05 09:42:39.822 | warn | (4403) Serial is not available
modbus.0 | 2020-06-05 09:42:39.622 | info | (4403) starting. Version 3.0.3 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.16.3, js-controller: 3.1.4

Thx @GermanBluefox

closed 2020-06-05 by GermanBluefox