ioBroker/ioBroker.modbus #68: Irregular error messages

thgonline, 2020-05-24:
I get the following error messages.

What could be the reason?

modbus.1 2020-05-23 08:16:57.506 info (1733) Terminated (NO_ERROR): Without reason
modbus.1 2020-05-23 08:16:57.504 info (1733) terminating
modbus.1 2020-05-23 08:16:57.491 error at processTimers (internal/timers.js:492:7)
modbus.1 2020-05-23 08:16:57.491 error at listOnTimeout (internal/timers.js:549:17)
modbus.1 2020-05-23 08:16:57.491 error at Timeout._onTimeout (/opt/iobroker/node_modules/iobroker.modbus/lib/master.js:286:17)
modbus.1 2020-05-23 08:16:57.491 error at pollResult (/opt/iobroker/node_modules/iobroker.modbus/lib/master.js:254:23)
modbus.1 2020-05-23 08:16:57.491 error (1733) Error: Reconnect did not help, restart adapter
modbus.1 2020-05-23 08:16:57.489 error (1733) uncaught exception: Reconnect did not help, restart adapter
modbus.1 2020-05-23 08:16:57.486 warn (1733) Poll error count: 13 code: "App Timeout"
modbus.1 2020-05-23 08:16:57.289 error (1733) Cannot write [40236]: {"err":"timeout"}
modbus.1 2020-05-23 08:16:57.288 error (1733) Client in error state.
modbus.1 2020-05-23 08:16:57.287 error (1733) Request timed out.
modbus.1 2020-05-23 08:16:57.287 warn (1733) Error: undefined
modbus.1 2020-05-23 08:16:52.285 info (1733) Connected to slave 192.168.178.82
modbus.1 2020-05-23 08:15:52.283 info (1733) Disconnected from slave 192.168.178.82
modbus.1 2020-05-23 08:15:51.480 warn (1733) Poll error count: 12 code: "App Timeout"
modbus.1 2020-05-23 08:15:51.283 error (1733) Cannot write [40236]: {"err":"timeout"}
modbus.1 2020-05-23 08:15:51.282 error (1733) Client in error state.
modbus.1 2020-05-23 08:15:51.282 error (1733) Request timed out.
modbus.1 2020-05-23 08:15:51.281 warn (1733) Error: undefined
modbus.1 2020-05-23 08:15:46.278 info (1733) Connected to slave 192.168.178.82

But the most common error message that occurs very often is this one:

modbus.1	2020-05-20 13:07:04.998	info	(6850) Connected to slave 192.168.178.82
modbus.1	2020-05-20 13:06:04.995	info	(6850) Disconnected from slave 192.168.178.82
modbus.1	2020-05-20 13:06:03.995	warn	(6850) Poll error count: 1 code: {"err":"timeout"}
modbus.1	2020-05-20 13:06:03.994	error	(6850) Client in error state.
modbus.1	2020-05-20 13:06:03.993	error	(6850) Request timed out.
modbus.1	2020-05-20 13:06:03.992	warn	(6850) Error: undefined

How can this be explained or what are the solutions?

Apollon77, 2020-05-24:
In fact that error means that the adapter for a timeout while reading from the server.  So please Check the Server device ...

thgonline, 2020-05-24:
How long may this time be?

According to Ping, the SMA PV inverter always has noticeably long response times of sometimes more than 30/40ms - normal is about 1ms

Apollon77, 2020-05-24:
You can set the timeout  in the adapter settings.

thgonline, 2020-05-24:
I know this but with all the settings you would have to know which setting affects this error message?
 
What would be the recommendation?

Currently the data are as follows:

Poll delay: 5000ms
Reconnect time: 60000ms
Read Timeout: 5000ms
Pulsetime: 1000ms

GermanBluefox, 2020-06-05:
If this TCP connection, I would suggest to sniff the communication with wireshark.

thgonline, 2020-06-05:
Unfortunately I do not have the knowledge to work with Wireshark.

Network problems can now be ruled out because everything has been replaced.

In the ioBroker forum I have now created a script that automatically restarts the adapter in case of an error. I can live with that for the first time. But these errors should be solved with the Modbus adapter.

GermanBluefox, 2020-06-05:
> But these errors should be solved with the Modbus adapter.

It is not the error in Adapter! It is an error with your system. As I cannot reproduce an error and I have no wireshark sniff, I will close this issue.

closed 2020-06-05 by GermanBluefox