ioBroker/ioBroker.modbus #69: Seit Update auf 3.0.2 keine Verbindung mit SMA Tripower Wechselrichter

TJJochen, 2020-06-02:
Nach dem Update bleiben alle Modbus Instanzen auf gelb. 
Aktuelle Version 3.0.2
ioBroker 4.1.1
Node.js 12.17.0
JS Controller 3.1.4
NPM 6.14.4

itProfi, 2020-06-02:
Same problem! After update not work!

Apollon77, 2020-06-02:
Could you please share some logs? Pot. enable debug log?

Apollon77, 2020-06-02:
But yes, verified!

GermanBluefox, 2020-06-02:
Only one line was modified from 3.0.1 to 3.0.2.... it cannot be because of change

Apollon77, 2020-06-02:
3.0.1 was never released!! So it is between 3.0.0 and 3.0.2!

TJJochen, 2020-06-02:
Now i'm back on version 3.0.0 in ioBroker and it's work perfectly. I don't know, what's the problem of 3.0.2 is

ingiz71, 2020-06-02:
Same behavior for me. After Upgrade to 3.0.2 both Modbus TCP instances (one master one slave) stay yellow. For me it look like that version 3.0.2 does only work if serial is available even if I use onle MODBUS TCP. At least after each resart of a 3.0.2 instance I get the following Message in the log: "Serial is not available". I downgraded to 3.0.0 and everything works now again. Except this the adapter works perfectly! :-)

BigBadUlf, 2020-06-03:
Same behavior for me. Version 3.0.2 does not work. The adapter stays yellow. I can‘t connect to E3/DC Storage. Downgrade to version 3.0.0 brings back the full funtionality.
In my log is: „Serial is not available“. This message was in both logs. Maybe in Version 3.0.2 this is a problem.

BigBadUlf, 2020-06-03:
Maybe the change from Blackbird77 in Version 3.0.1 produce the problem.

(BlackBird77) Fixes for Serial Timeouts done

Garfonso, 2020-06-03:
Yes, sadly TCP does not work currently.
This is the log output:
```
2020-06-04 00:57:05.220  - info: host.Server instance system.adapter.modbus.0 started with pid 28667
2020-06-04 00:57:05.450  - info: modbus.0 (28667) starting. Version 3.0.2 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.16.3, js-controller: 3.1.4
2020-06-04 00:57:05.461  - warn: modbus.0 (28667) Serial is not available
```

smartboart, 2020-06-04:
Not only tcp...

Anzic23, 2020-06-04:
if you need statistics, then it doesn’t work for me either. TCP connection.

Matzebhv, 2020-06-04:
Same Problem here with serial connection to our Growatt PV-Inverter. Switching back to 3.0.0 fixes this issue. Even in silly nothing wrong in log.

BigBadUlf, 2020-06-05:
Version 3.0.3 works fine.
The modbus objects are updatet.

Here my log:
modbus.0 | 2020-06-05 09:42:40.846 | info | (4403) Connected to slave 192.168.178.86
-- | -- | -- | --
modbus.0 | 2020-06-05 09:42:39.822 | warn | (4403) Serial is not available
modbus.0 | 2020-06-05 09:42:39.622 | info | (4403) starting. Version 3.0.3 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.16.3, js-controller: 3.1.4

Thx @GermanBluefox

closed 2020-06-05 by GermanBluefox