ioBroker/ioBroker.modbus #70: after update from 3.0.2 to 3.0.4 lose connection

superpoli, 2020-06-08:
after update from 3.0.2 to 3.0.4 lose connection after about 30s
`

modbus.1 | 2020-06-08 09:37:00.040 | debug | (6052) Cleaning up request fifo.
-- | -- | -- | --
modbus.1 | 2020-06-08 09:37:00.040 | debug | (6052) Clearing timeout of the current request.
modbus.1 | 2020-06-08 09:37:00.039 | debug | (6052) Socket closed with error
modbus.1 | 2020-06-08 09:37:00.022 | info | (6052) terminating
modbus.1 | 2020-06-08 09:37:00.010 | info | (6052) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
modbus.1 | 2020-06-08 09:37:00.007 | debug | (6052) Closing client on purpose.
modbus.1 | 2020-06-08 09:37:00.006 | info | (6052) Scheduled restart.
modbus.1 | 2020-06-08 09:36:59.239 | debug | (6052) Poll inputRegs DevID(3) address 30953 - 10 bytes

`

Node-Version: | v12.18.0
Nodejs-Version: | v12.18.0
NPM-Version: | 6.14.4

Apollon77, 2020-06-08:
What is the tipic? The fact that the connection was closed or that the scheduled restart did not worked? If second, I currently prepare an update for this.

superpoli, 2020-06-09:
I think the problem is that the connection was closed, but i can test your update first.

closed 2020-06-11 by Apollon77

superpoli, 2020-06-12:
the update did not solve the problem. the connection is established and at the full minute it disconnects.
`
host.iobroker-ubuntu | 2020-06-12 08:49:00.536 | info | Do not restart adapter system.adapter.modbus.1 because desired by instance
-- | -- | -- | --
host.iobroker-ubuntu | 2020-06-12 08:49:00.536 | error | instance system.adapter.modbus.1 terminated by request of the instance itself and will not be restarted, before user restarts it.
mqtt.0 | 2020-06-12 08:49:00.518 | info | (14052) Terminated (START_IMMEDIATELY_AFTER_STOP): Without reason
mqtt.0 | 2020-06-12 08:49:00.516 | info | (14052) terminating
modbus.1 | 2020-06-12 08:49:00.046 | debug | (14387) Cleaning up request fifo.
modbus.1 | 2020-06-12 08:49:00.045 | debug | (14387) Clearing timeout of the current request.
modbus.1 | 2020-06-12 08:49:00.040 | debug | (14387) Socket closed with error
modbus.1 | 2020-06-12 08:49:00.010 | info | (14387) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
modbus.1 | 2020-06-12 08:49:00.009 | debug | (14387) Plugin sentry destroyed
modbus.1 | 2020-06-12 08:49:00.007 | debug | (14387) Closing client on purpose.
modbus.1 | 2020-06-12 08:49:00.005 | info | (14387) Scheduled restart.
mqtt.0 | 2020-06-12 08:49:00.005 | info | (14052) Scheduled restart.
modbus.1 | 2020-06-12 08:48:59.908 | debug | (14387) Poll inputRegs DevID(3) address 30773 - 4 bytes
modbus.1 | 2020-06-12 08:48:59.757 | debug | (14387) Poll inputRegs DevID(3) address 30531 - 8 bytes
modbus.1 | 2020-06-12 08:48:59.756 | debug | (14387) Poll device 3
modbus.1 | 2020-06-12 08:48:58.564 | debug | (14387) Poll inputRegs DevID(3) address 30953 - 10 bytes
modbus.1 | 2020-06-12 08:48:58.408 | debug | (14387) Poll inputRegs DevID(3) address 30773 - 4 bytes
modbus.1 | 2020-06-12 08:48:58.261 | debug | (14387) Poll inputRegs DevID(3) address 30531 - 8 bytes
modbus.1 | 2020-06-12 08:48:58.260 | debug | (14387) Poll device 3
modbus.1 | 2020-06-12 08:48:57.059 | debug | (14387) Poll inputRegs DevID(3) address 30953 - 10 bytes
modbus.1 | 2020-06-12 08:48:56.908 | debug | (14387) Poll inputRegs DevID(3) address 30773 - 4 bytes
modbus.1 | 2020-06-12 08:48:56.747 | debug | (14387) Poll inputRegs DevID(3) address 30531 - 8 bytes
`

Apollon77, 2020-06-12:
I see a scheduled restart here. Do you have something configured? Maybe wrong not that you have configured it to restart every minute. Also please show a Full log from such a run

superpoli, 2020-06-12:
A restart was actually configured every minute for both controllers, but not by me. I took them out, now everything is ok. thank you

Apollon77, 2020-06-12:
such an auto restart is never configured by the system ;-) So should be done by you ... but now ok

commit 761befae (2020-06-11): * (Apollon77) Make sure that regular adapter stops do not terminate the process, so that scheduled restarts still work, fixes #70
* Add issue templates
* Add stalebot configuration
* Update github action script
* update serialport to also support nodejs 12/14