ioBroker/ioBroker.modbus #71: Serielle Schnittstelle nicht wählbar

AlfredSch, 2020-06-12:
Hallo, nach der Installation der Adapter Version 3.1.1 ist es nicht mehr möglich, den Serial Port auszuwählen. Das Auswahl Menu bleibt leer. 
Nach dem Downgrade auf Version 3.0.4 ist alles wieder ok.
Node.js   12.18.0
NPM        6.14.4

Gruss und Danke

Apollon77, 2020-06-12:
Bitte installations log Posten.

Apollon77, 2020-06-12:
Und auch log vom adapterstart bitte posten.

AlfredSch, 2020-06-12:
**Installations Log:**

$ ./iobroker upgrade modbus
Update modbus from @3.0.4 to @3.1.1
NPM version: 6.14.4
npm install iobroker.modbus@3.1.1 --loglevel error --prefix "/opt/iobroker" (System call)
upload [11] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/words.js words.js application/javascript
Update "system.adapter.modbus.0"
Update "system.adapter.modbus.1"
upload [10] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/modbus.png modbus.png image/png
upload [9] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/lib/js/jsgrid.js lib/js/jsgrid.js application/javascript
upload [8] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/lib/js/grid.locale-ru.js lib/js/grid.locale-ru.js application/javascript
upload [7] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/lib/js/grid.locale-de.js lib/js/grid.locale-de.js application/javascript
upload [6] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/lib/css/jsgrid.min.css lib/css/jsgrid.min.css text/css
upload [5] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/lib/css/jsgrid.css lib/css/jsgrid.css text/css
upload [4] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/lib/css/jsgrid-theme.min.css lib/css/jsgrid-theme.min.css text/css
upload [3] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/lib/css/jsgrid-theme.css lib/css/jsgrid-theme.css text/css
upload [2] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/index.html index.html text/html
upload [1] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/img/plc_back_opacity.png img/plc_back_opacity.png image/png
upload [0] modbus.admin /opt/iobroker/node_modules/iobroker.modbus/admin/img/plc_back.png img/plc_back.png image/png
Adapter "modbus" updated
process exited with code 0

**Nach der Installation läuft der Adapter ohne Problem, erst wenn die Konfiguration aufgerufen wird bleibt bei der Auswahl des Ports alles leer.
Danach läuft der Adapter auf Fehler.**

  **Adapter Log :**
modbus.0	2020-06-12 11:33:53.348	debug	(13914) Poll inputRegs DevID(1) address 342 - 10 bytes
modbus.0	2020-06-12 11:33:53.307	debug	(13914) Poll inputRegs DevID(1) address 224 - 2 bytes
modbus.0	2020-06-12 11:33:53.251	debug	(13914) Poll inputRegs DevID(1) address 200 - 6 bytes
modbus.0	2020-06-12 11:33:53.209	debug	(13914) Poll inputRegs DevID(1) address 70 - 4 bytes
modbus.0	2020-06-12 11:33:53.148	debug	(13914) Poll inputRegs DevID(1) address 48 - 10 bytes
modbus.0	2020-06-12 11:33:53.064	debug	(13914) Poll inputRegs DevID(1) address 0 - 18 bytes
modbus.0	2020-06-12 11:33:53.063	debug	(13914) Poll device 1
modbus.0	2020-06-12 11:33:52.741	warn	at process.topLevelDomainCallback (domain.js:137:15)
modbus.0	2020-06-12 11:33:52.741	warn	at processImmediate (internal/timers.js:456:21)
modbus.0	2020-06-12 11:33:52.741	warn	at Immediate._onImmediate (/opt/iobroker/node_modules/iobroker.js-controller/lib/states/statesInRedis.js:234:41)
modbus.0	2020-06-12 11:33:52.741	warn	at change (/opt/iobroker/node_modules/iobroker.js-controller/lib/adapter.js:5313:34)
modbus.0	2020-06-12 11:33:52.741	warn	at Adapter.EventEmitter.emit (domain.js:482:12)
modbus.0	2020-06-12 11:33:52.741	warn	at Adapter.emit (events.js:315:20)
modbus.0	2020-06-12 11:33:52.741	warn	at Adapter.<anonymous> (/opt/iobroker/node_modules/iobroker.modbus/main.js:36:45)
modbus.0	2020-06-12 11:33:52.741	warn	(13914) TypeError: serialport.list.then is not a function
modbus.0	2020-06-12 11:33:52.740	warn	(13914) States system pmessage io.messagebox.system.adapter.modbus.0 {"command":"listUart","message":null,"from":"system.adapter.admin.0","callback":{"message":null,"id":236,"ack":false,"time":159195
modbus.0	2020-06-12 11:33:50.698	warn	at process.topLevelDomainCallback (domain.js:137:15)
modbus.0	2020-06-12 11:33:50.698	warn	at processImmediate (internal/timers.js:456:21)
modbus.0	2020-06-12 11:33:50.698	warn	at Immediate._onImmediate (/opt/iobroker/node_modules/iobroker.js-controller/lib/states/statesInRedis.js:234:41)
modbus.0	2020-06-12 11:33:50.698	warn	at change (/opt/iobroker/node_modules/iobroker.js-controller/lib/adapter.js:5313:34)
modbus.0	2020-06-12 11:33:50.698	warn	at Adapter.EventEmitter.emit (domain.js:482:12)
modbus.0	2020-06-12 11:33:50.698	warn	at Adapter.emit (events.js:315:20)
modbus.0	2020-06-12 11:33:50.698	warn	at Adapter.<anonymous> (/opt/iobroker/node_modules/iobroker.modbus/main.js:36:45)
modbus.0	2020-06-12 11:33:50.698	warn	(13914) TypeError: serialport.list.then is not a function
modbus.0	2020-06-12 11:33:50.696	warn	(13914) States system pmessage io.messagebox.system.adapter.modbus.0 {"command":"listUart","message":null,"from":"system.adapter.admin.0","callback":{"message":null,"id":235,"ack":false,"time":159195
modbus.0	2020-06-12 11:33:47.712	warn	at process.topLevelDomainCallback (domain.js:137:15)
modbus.0	2020-06-12 11:33:47.712	warn	at processImmediate (internal/timers.js:456:21)
modbus.0	2020-06-12 11:33:47.712	warn	at Immediate._onImmediate (/opt/iobroker/node_modules/iobroker.js-controller/lib/states/statesInRedis.js:234:41)
modbus.0	2020-06-12 11:33:47.712	warn	at change (/opt/iobroker/node_modules/iobroker.js-controller/lib/adapter.js:5313:34)
modbus.0	2020-06-12 11:33:47.712	warn	at Adapter.EventEmitter.emit (domain.js:482:12)
modbus.0	2020-06-12 11:33:47.712	warn	at Adapter.emit (events.js:315:20)
[... cut ...]
modbus.0	2020-06-12 11:33:45.697	warn	at Adapter.emit (events.js:315:20)
modbus.0	2020-06-12 11:33:45.697	warn	at Adapter.<anonymous> (/opt/iobroker/node_modules/iobroker.modbus/main.js:36:45)
modbus.0	2020-06-12 11:33:45.697	warn	(13914) TypeError: serialport.list.then is not a function
modbus.0	2020-06-12 11:33:45.696	warn	(13914) States system pmessage io.messagebox.system.adapter.modbus.0 {"command":"listUart","message":null,"from":"system.adapter.admin.0","callback":{"message":null,"id":233,"ack":false,"time":159195

Apollon77, 2020-06-12:
Ahh ok. Fixe ich heute Abend.

AlfredSch, 2020-06-12:
Super und Danke :-)

Apollon77, 2020-06-12:
Bitte 3.1.2 versuchen

closed 2020-06-12 by Apollon77