ioBroker/ioBroker.modbus #75: Can't select serial port.  The pull down menu only shows undefined.  /dev/ttyUSB0 seems to be working and available.

kskenyon, 2020-09-20:
**Describe the bug**  
Can't select serial port.  The pull down menu only shows undefined.  /dev/ttyUSB0 seems to be working and available.
**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '.modbus adapter general configuration page.'
2. Click on '.port..'
3. Scroll down to '...Only Undefined is shown, should show /dev/ttyUSB0.'
4. See error

**Expected behavior**  
should show /dev/ttyUSB0.'
**Screenshots & Logfiles**  

modbus.0 | 2020-09-20 00:18:54.165 | warn | (2061) On error: {}
-- | -- | -- | --
modbus.0 | 2020-09-20 04:17:54.163 | warn | (2061) On error: {}
modbus.0 | 2020-09-20 04:17:53.162 | warn | (2061) On error: {}
modbus.0 | 2020-09-20 04:17:20.397 | info | (2061) List of port: [{"comName":"/dev/ttyUSB0"}]
modbus.0 | 2020-09-20 04:16:53.160 | warn | (2061) On error: {}
modbus.0 | 2020-09-20 04:16:52.157 | warn | (2061) On error: {}
modbus.0 | 2020-09-20 04:15:52.156 | warn | (2061) On error: {}
modbus.0 | 2020-09-20 04:15:51.155 | warn | (2061) On error: {}
modbus.0 | 2020-09-20 04:14:51.153 | warn | (2061) On error: {}
modbus.0 | 2020-09-20 04:14:50.152 | warn | (2061) On error: {}
modbus.0 | 2020-09-20 04:13:50.150 | warn | (2061) On error: {}
modbus.0 | 2020-09-20 04:13:49.149 | warn | (2061) On error: {}

**Versions:**  
 - Adapter version: <adapter-version>3.1.9
 - JS-Controller version: <js-controller-version> 3.1.4
 - Node version: <node-version> 12.16.3
 - Operating system: <os-name>Raspbian Stretch

**Additional context**  
Serialport is installed.

cyberax17, 2020-12-01:
Dear All, any solution to this bug? Ran into it all of a sudden today. No change except restart of the adapter.

Lichte12, 2020-12-01:
Have you tried to delete the adpater and reinstall? This fixed this problem for me recently.

cyberax17, 2020-12-01:
Will this delete all my register assignments? How would I preserve those?

Lichte12, 2020-12-01:
Export all register to CSV first. After reinstall import again.
[image]

cyberax17, 2020-12-02:
Just deleted and re-installed the modes adapter. no effect unfortunately. same issue: ports are all "undefined" .

kskenyon, 2020-12-02:
I did the same thing. No change.

cyberax17, 2020-12-03:
I now use mbusd  as a TCP/MODBUS bridge and this seems to work fine.

Apollon77, 2020-12-03:
I think I have an idea ... let me check something

Apollon77, 2020-12-03:
Please try 3.1.11

closed 2020-12-03 by Apollon77

cyberax17, 2020-12-04:
thanks for the quick update, unfortunately it did not do the job. same issue. Do I need to do something else?

Apollon77, 2020-12-05:
Please try to "iobroker upload modbus" ... really not working?

Apollon77, 2020-12-05:
hm ... I see ... i check again tonight

Apollon77, 2020-12-05:
check Github version please ... or wait for 3.1.12 in latest soon

kskenyon, 2020-12-06:
I still don’t seem to be able to connect.  Serial Port not available even after a reset.  Works OK from a node-red modbus node so hardware is OK.

Kevin

2020-12-06 02:30:50.913	info	(30859) Disconnected from slave
modbus.0	2020-12-06 02:30:50.911	info	(30859) Connected to slave
modbus.0	2020-12-06 02:30:50.490	info	(30859) starting. Version 3.1.12 in /opt/iobroker/node_modules/iobroker.modbus, node: v10.21.0, js-controller: 3.1.6
modbus.0	2020-12-06 02:30:47.172	info	(30837) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
modbus.0	2020-12-06 02:30:47.169	info	(30837) terminating
modbus.0	2020-12-06 02:30:47.163	info	(30837) Got terminate signal TERMINATE_YOURSELF
modbus.0	2020-12-06 02:30:39.741	info	(30837) List of port: [{"path":"/dev/ttyUSB0"}]
modbus.0	2020-12-06 02:30:39.741	info	(30837) Check /dev/ttyUSB0 : false
modbus.0	2020-12-06 02:30:39.739	info	(30837) Verify []
modbus.0	2020-12-06 02:30:19.928	info	(30837) Disconnected from slave
modbus.0	2020-12-06 02:30:19.927	info	(30837) Connected to slave
modbus.0	2020-12-06 02:29:19.922	info	(30837) Disconnected from slave
modbus.0	2020-12-06 02:29:19.920	info	(30837) Connected to slave
modbus.0	2020-12-06 02:29:19.099	info	(30837) starting. Version 3.1.12 in /opt/iobroker/node_modules/iobroker.modbus, node: v10.21.0, js-controller: 3.1.6
modbus.0	2020-12-06 02:29:14.242	info	(30816) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
modbus.0	2020-12-06 02:29:14.240	info	(30816) terminating
modbus.0	2020-12-06 02:29:14.238	info	(30816) Got terminate signal TERMINATE_YOURSELF
modbus.0	2020-12-06 02:28:52.584	info	(30816) Disconnected from slave
modbus.0	2020-12-06 02:28:52.583	info	(30816) Connected to slave
modbus.0	2020-12-06 02:27:52.579	info	(30816) Disconnected from slave
modbus.0	2020-12-06 02:27:52.576	info	(30816) Connected to slave
modbus.0	2020-12-06 02:27:52.075	info	(30816) starting. Version 3.1.12 in /opt/iobroker/node_modules/iobroker.modbus, node: v10.21.0, js-controller: 3.1.6
modbus.0	2020-12-06 02:27:48.858	info	(30352) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
modbus.0	2020-12-06 02:27:48.851	info	(30352) terminating
modbus.0	2020-12-06 02:27:48.846	info	(30352) Got terminate signal TERMINATE_YOURSELF
modbus.0	2020-12-06 02:27:35.445	info	(30352) List of port: [{"path":"/dev/ttyUSB0"}]
modbus.0	2020-12-06 02:27:35.445	info	(30352) Check /dev/ttyUSB0 : false
modbus.0	2020-12-06 02:27:35.442	info	(30352) Verify []
modbus.0	2020-12-06 02:26:51.935	info	(30352) Disconnected from slave
modbus.0	2020-12-06 02:26:51.934	info	(30352) Connected to slave
modbus.0	2020-12-06 02:25:51.928	info	(30352) Disconnected from slave
modbus.0	2020-12-06 02:25:51.927	info	(30352) Connected to slave

Apollon77, 2020-12-06:
Ok, so : In admin now all is ok and you can choose the port, but now connection does not work?

Please first run "Iobroker fix" to make sure all needed permissions are correct.

If still not work please disable the adapter in admin, open a shell and run "iobroker debug modbus.0". Post output here

SSonder0815, 2021-02-11:
I had the same issue with missing port descriptions that lead to an error executing the modbus adapter in serial mode, it worked for me to update the linux os on the raspberry, restart the iobroker and voila, iobroker connected to the Modbus RTU server.

ViPtAd, 2022-02-25:
Hi all,

i am experience the same issue with nearly the same configuration; installed iobroker docker from buanet repo, created the device connection as shown below:

[image]
[image]
(ok, it is used in portainer but this should not be the deal...)

I checked the devices in the console (e voila):
```
root@iobroker:/opt/iobroker# ls -al /dev/serial/by-id/*
crw-rw---- 1 root dialout 188, 0 Feb 25 13:02 /dev/serial/by-id/rs485port1
crw-rw---- 1 root dialout 188, 1 Feb 25 13:02 /dev/serial/by-id/rs485port2
```

But the selection is empty:
[image]

best regards and thank you for your help!

ViPtAd, 2022-02-25:
OK, solved the problem for me:
I wanted to give it a try if I change the alias, now the adapters are showing up:
[image]