ioBroker/ioBroker.modbus #77: Holdingregisters without poll aren't created

k3mpaxl, 2020-10-18:
I've created a holdingregister for an write-only address and set the checkbox for polling to false. With the restart of the adapter, all states are erases and are re-created. For holdingregisters with polling enabled this works fine. For holdingregisters without polling enabled, the states aren't created.
Manually adding the states through the GUI works and the adapter sends the values as expected if i edit the state.

Could you please fix this issue ? 

Thx.

anhomb, 2020-12-13:
I've got the same problem. The manual adding of the state works, but If I restart the modbus adapter the state is gone. 

Any Solution?

Thanks!

k3mpaxl, 2021-01-25:
Any update here?

riedesign, 2021-01-30:
Hi,
I have the exact same problem.
Hopefully there will be fix of this bug. 

Thanks a lot!

Starter4711, 2021-03-02:
Same for Keba P30c Wallbox, need write-only and no polling.
THX

k3mpaxl, 2021-03-08:
Hey @Apollon77, if you have time and resources available, do you mind having a look at this? It might be a quick win/fix? Thank you very much!

MarcelMertens, 2021-03-08:
For clarification:
Register 262 is writable, but not readable. So Disabled "poll"
[image]

But Datapoint is not created:
[image]

samot13, 2021-03-13:
Hi,
I have also the same problem.
Hopefully this bug will be fixed soon.

Thanks!

Apollon77, 2021-04-11:
Ok, lets dig into it :-)

Please install version from Github. Version number will not change. Set to debug log (admin- Instances - Expert mode - column loglevel) and restart adapter (onky needed if you had loglevel debug before. Checck if it happens again and provide debug log. Take log from disk /opt/iobroker/log/... please

riedesign, 2021-04-12:
Great! Thanks.

Here is my log:
`2021-04-12 09:56:27.972  - ^[[32minfo^[[39m: modbus.2 (211053) Plugin sentry Sentry Plugin disabled for this process because sending of statistic data is disabled for the system
2021-04-12 09:56:28.022  - ^[[32minfo^[[39m: modbus.2 (211053) starting. Version 3.2.6 in /opt/iobroker/node_modules/iobroker.modbus, node: v10.22.0, js-controller: 3.2.16
2021-04-12 09:56:28.159  - ^[[34mdebug^[[39m: modbus.2 (211053) Initialize Objects for disInputs: []
2021-04-12 09:56:28.160  - ^[[34mdebug^[[39m: modbus.2 (211053) Initialize Objects for coils: []
2021-04-12 09:56:28.160  - ^[[34mdebug^[[39m: modbus.2 (211053) Initialize Objects for inputRegs: [{"_address":0,"address":0,"name":"Time","description":"Unix timestamp: higher word gets  transmit  in $
2021-04-12 09:56:28.161  - ^[[34mdebug^[[39m: modbus.2 (211053) Add inputRegisters.0_Time: {"_id":"inputRegisters.0_Time","type":"state","common":{"name":"Unix timestamp: higher word gets  transmit  in$
2021-04-12 09:56:28.177  - ^[[34mdebug^[[39m: modbus.2 (211053) Initialize Objects for holdingRegs: [{"_address":0,"name":"SetMaxPower","description":"Set maximum Power to be drained from grid","unit":$
2021-04-12 09:56:28.177  - ^[[34mdebug^[[39m: modbus.2 (211053) Add undefined: {"type":"state","common":{"name":"Set maximum Power to be drained from grid","role":"value","type":"number","read":false,"$
2021-04-12 09:56:28.426  - ^[[32minfo^[[39m: modbus.2 (211053) Connected to slave 192.168.11.128
2021-04-12 09:56:28.426  - ^[[34mdebug^[[39m: modbus.2 (211053) Poll device 1
2021-04-12 09:56:28.427  - ^[[34mdebug^[[39m: modbus.2 (211053) Initialization of scale factors done!
2021-04-12 09:56:28.427  - ^[[34mdebug^[[39m: modbus.2 (211053) Poll inputRegs DevID(1) address 0 - 19 bytes`

Problem is SetMaxPower

Apollon77, 2021-04-12:
Ok, next try please ... and ideally also add a polling datapoint that I can see that still all works for that too ...

riedesign, 2021-04-13:
The value which is not polled was now generated, but if I write a value to this I get following error:
> modbus.4 | 2021-04-13 06:44:27.122 | warn | (213426) Poll error count: 1 code: "App Timeout"
modbus.4 | 2021-04-13 06:44:26.605 | error | (213426) Cannot write [5004]: {"err":"timeout","timeout":5000}
modbus.4 | 2021-04-13 06:44:26.604 | error | (213426) Client in error state.
modbus.4 | 2021-04-13 06:44:26.604 | error | (213426) Request timed out.
modbus.4 | 2021-04-13 06:44:26.603 | warn | (213426) Error: undefined

All values which are polled, are working correct.

Here is the start-up log:

> 2021-04-13 06:32:41.649  - [34mdebug[39m: modbus.4 (213305) Redis Objects: Use Redis connection: 127.0.0.1:9001
2021-04-13 06:32:41.696  - [34mdebug[39m: modbus.4 (213305) Objects client ready ... initialize now
2021-04-13 06:32:41.701  - [34mdebug[39m: modbus.4 (213305) Objects create System PubSub Client
2021-04-13 06:32:41.701  - [34mdebug[39m: modbus.4 (213305) Objects create User PubSub Client
2021-04-13 06:32:41.702  - [34mdebug[39m: modbus.4 (213305) Objects client initialize lua scripts
2021-04-13 06:32:41.751  - [34mdebug[39m: modbus.4 (213305) Objects connected to redis: 127.0.0.1:9001
2021-04-13 06:32:41.757  - [34mdebug[39m: modbus.4 (213305) objectDB connected
2021-04-13 06:32:41.759  - [34mdebug[39m: modbus.4 (213305) Redis States: Use Redis connection: 127.0.0.1:9000
2021-04-13 06:32:41.822  - [34mdebug[39m: modbus.4 (213305) States create System PubSub Client
2021-04-13 06:32:41.822  - [34mdebug[39m: modbus.4 (213305) States create User PubSub Client
2021-04-13 06:32:41.942  - [34mdebug[39m: modbus.4 (213305) States connected to redis: 127.0.0.1:9000
2021-04-13 06:32:41.943  - [34mdebug[39m: modbus.4 (213305) statesDB connected
2021-04-13 06:32:42.520  - [34mdebug[39m: modbus.4 (213305) Plugin sentry Initialize Plugin (enabled=true)
2021-04-13 06:32:42.523  - [32minfo[39m: modbus.4 (213305) Plugin sentry Sentry Plugin disabled for this process because sending of statistic data is disabled for the system
2021-04-13 06:32:42.576  - [32minfo[39m: modbus.4 (213305) starting. Version 3.2.6 in /opt/iobroker/node_modules/iobroker.modbus, node: v10.22.0, js-controller: 3.2.16
2021-04-13 06:32:42.758  - [34mdebug[39m: modbus.4 (213305) Initialize Objects for disInputs: []
2021-04-13 06:32:42.763  - [34mdebug[39m: modbus.4 (213305) Initialize Objects for coils: []
2021-04-13 06:32:42.764  - [34mdebug[39m: modbus.4 (213305) Initialize Objects for inputRegs: []
2021-04-13 06:32:42.764  - [34mdebug[39m: modbus.4 (213305) Initialize Objects for holdingRegs: [{"_address":1000,"name":"Charging state","description":"Charging state","unit":"","type":"uint32be","len":2,"factor":1,"offset":0,"formula":"","role":"state","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":1000,"deviceId":255,"id":"holdingRegisters.1000_Charging_state"},{"_address":1004,"name":"Cable State","description":"Cable State","unit":"","type":"uint32be","len":2,"factor":1,"offset":0,"formula":"","role":"state","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":1004,"deviceId":255,"id":"holdingRegisters.1004_Cable_State"},{"_address":1008,"name":"Charging current phase 1","description":"Charging current phase 1","unit":"mA","type":"uint32be","len":2,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":1008,"deviceId":255,"id":"holdingRegisters.1008_Charging_current phase 1"},{"_address":1010,"name":"Charging current phase 2","description":"Charging current phase 2","unit":"mA","type":"uint32be","len":2,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":1010,"deviceId":255,"id":"holdingRegisters.1010_Charging_current phase 2"},{"_address":1012,"name":"Charging current phase 3","description":"Charging current phase 1","unit":"mA","type":"uint32be","len":2,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":1012,"deviceId":255,"id":"holdingRegisters.1012_Charging_current phase 3"},{"_address":1020,"name":"Active Power","description":"Active Power","unit":"kW","type":"uint32be","len":2,"factor":0.000001,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":1020,"deviceId":255,"id":"holdingRegisters.1020_Active_Power"},{"_address":1100,"name":"Max charging current","description":"Max charging current","unit":"mA","type":"uint32be","len":2,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":1100,"deviceId":255,"id":"holdingRegisters.1100_Max_charging current"},{"_address":1500,"name":"RFID card","description":"RFID card","unit":"","type":"uint32be","len":2,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":1500,"deviceId":255,"id":"holdingRegisters.1500_RFID_card"},{"_address":1502,"name":"Charged energy","description":"Charged energy","unit":"kWh","type":"uint32be","len":2,"factor":0.0001,"offset":1,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":1502,"deviceId":255,"id":"holdingRegisters.1502_Charged_energy"},{"_address":5004,"name":"Set_charging current","description":"","unit":"","type":"uint16be","len":1,"factor":"1","offset [... cut ...]
2021-04-13 06:32:43.892  - [34mdebug[39m: modbus.4 (213305) Poll holdingRegs DevID(255) address 1020 - 2 bytes
2021-04-13 06:32:43.896  - [34mdebug[39m: modbus.4 (213305) Poll holdingRegs DevID(255) address 1100 - 2 bytes
2021-04-13 06:32:43.900  - [34mdebug[39m: modbus.4 (213305) Poll holdingRegs DevID(255) address 1500 - 2 bytes
2021-04-13 06:32:43.903  - [34mdebug[39m: modbus.4 (213305) Poll holdingRegs DevID(255) address 1502 - 2 bytes
2021-04-13 06:32:44.965  - [34mdebug[39m: modbus.4 (213305) Poll device 255

Apollon77, 2021-04-13:
Is the write log anything or did you cutted stuff out?

If there is normally more please include. And also restore the official version of adapter and add the state manually as you did before and send a log for write casse to compare ...
Honestly I'm doing blind exoeriments with this rght now!

riedesign, 2021-04-13:
I tried it once more:

write log:

> 2021-04-13 07:56:57.680  - [34mdebug[39m: modbus.4 (2039) Poll device 255
2021-04-13 07:56:57.681  - [34mdebug[39m: modbus.4 (2039) Poll holdingRegs DevID(255) address 1000 - 2 bytes
2021-04-13 07:57:02.144  - [33mwarn[39m: modbus.4 (2039) Error: undefined
2021-04-13 07:57:02.146  - [31merror[39m: modbus.4 (2039) Request timed out.
2021-04-13 07:57:02.146  - [31merror[39m: modbus.4 (2039) Client in error state.
2021-04-13 07:57:02.147  - [31merror[39m: modbus.4 (2039) Cannot write [5004]: {"err":"timeout","timeout":5000}
2021-04-13 07:57:02.148  - [34mdebug[39m: modbus.4 (2039) Socket closed with error
2021-04-13 07:57:02.148  - [34mdebug[39m: modbus.4 (2039) Clearing timeout of the current request.
2021-04-13 07:57:02.148  - [34mdebug[39m: modbus.4 (2039) Cleaning up request fifo.
2021-04-13 07:57:02.882  - [33mwarn[39m: modbus.4 (2039) Poll error count: 1 code: "App Timeout"
2021-04-13 07:57:03.146  - [34mdebug[39m: modbus.4 (2039) Closing client on purpose.
2021-04-13 07:57:03.147  - [32minfo[39m: modbus.4 (2039) Disconnected from slave 192.168.11.16

Write log with official version and manally created state: (write value is '11000')

> 2021-04-13 08:04:49.382  - [34mdebug[39m: modbus.4 (2115) Poll device 255
2021-04-13 08:04:49.382  - [34mdebug[39m: modbus.4 (2115) Poll holdingRegs DevID(255) address 1000 - 2 bytes
2021-04-13 08:04:49.385  - [34mdebug[39m: modbus.4 (2115) Poll holdingRegs DevID(255) address 1004 - 2 bytes
2021-04-13 08:04:49.387  - [34mdebug[39m: modbus.4 (2115) Poll holdingRegs DevID(255) address 1008 - 2 bytes
2021-04-13 08:04:49.390  - [34mdebug[39m: modbus.4 (2115) Poll holdingRegs DevID(255) address 1010 - 2 bytes
2021-04-13 08:04:49.393  - [34mdebug[39m: modbus.4 (2115) Poll holdingRegs DevID(255) address 1012 - 2 bytes
2021-04-13 08:04:49.396  - [34mdebug[39m: modbus.4 (2115) Poll holdingRegs DevID(255) address 1020 - 2 bytes
2021-04-13 08:04:49.398  - [34mdebug[39m: modbus.4 (2115) Poll holdingRegs DevID(255) address 1100 - 2 bytes
2021-04-13 08:04:49.401  - [34mdebug[39m: modbus.4 (2115) Poll holdingRegs DevID(255) address 1500 - 2 bytes
2021-04-13 08:04:49.403  - [34mdebug[39m: modbus.4 (2115) Poll holdingRegs DevID(255) address 1502 - 2 bytes
2021-04-13 08:04:50.023  - [34mdebug[39m: modbus.4 (2115) Write successfully [5004]: 11000

riedesign, 2021-04-13:
Perhaps that helps ... i used debub "silly" mode

Official version:
> 2021-04-13 08:13:22.718  - [34mdebug[39m: modbus.4 (2322) Poll device 255
2021-04-13 08:13:22.718  - [34mdebug[39m: modbus.4 (2322) Poll holdingRegs DevID(255) address 1000 - 2 bytes
2021-04-13 08:13:22.722  - [34mdebug[39m: modbus.4 (2322) Poll holdingRegs DevID(255) address 1004 - 2 bytes
2021-04-13 08:13:22.725  - [34mdebug[39m: modbus.4 (2322) Poll holdingRegs DevID(255) address 1008 - 2 bytes
2021-04-13 08:13:22.728  - [34mdebug[39m: modbus.4 (2322) Poll holdingRegs DevID(255) address 1010 - 2 bytes
2021-04-13 08:13:22.730  - [34mdebug[39m: modbus.4 (2322) Poll holdingRegs DevID(255) address 1012 - 2 bytes
2021-04-13 08:13:22.732  - [34mdebug[39m: modbus.4 (2322) Poll holdingRegs DevID(255) address 1020 - 2 bytes
2021-04-13 08:13:22.732  - [35msilly[39m: modbus.4 (2322) States user redis pmessage modbus.4.*/modbus.4.holdingRegisters.1008_Charging_current phase 1:{"val":10637,"ack":true,"ts":1618294402729,"q":0,"from":"system.adapter.modbus.4","user":"system.user.admin","lc":1618294402729}
2021-04-13 08:13:22.734  - [34mdebug[39m: modbus.4 (2322) Poll holdingRegs DevID(255) address 1100 - 2 bytes
2021-04-13 08:13:22.736  - [35msilly[39m: modbus.4 (2322) States user redis pmessage modbus.4.*/modbus.4.holdingRegisters.1010_Charging_current phase 2:{"val":10656,"ack":true,"ts":1618294402734,"q":0,"from":"system.adapter.modbus.4","user":"system.user.admin","lc":1618294402734}
2021-04-13 08:13:22.736  - [34mdebug[39m: modbus.4 (2322) Poll holdingRegs DevID(255) address 1500 - 2 bytes
2021-04-13 08:13:22.738  - [35msilly[39m: modbus.4 (2322) States user redis pmessage modbus.4.*/modbus.4.holdingRegisters.1012_Charging_current phase 3:{"val":10571,"ack":true,"ts":1618294402735,"q":0,"from":"system.adapter.modbus.4","user":"system.user.admin","lc":1618294402735}
2021-04-13 08:13:22.739  - [34mdebug[39m: modbus.4 (2322) Poll holdingRegs DevID(255) address 1502 - 2 bytes
2021-04-13 08:13:22.824  - [35msilly[39m: modbus.4 (2322) States user redis pmessage modbus.4.*/modbus.4.holdingRegisters.1020_Active_Power:{"val":7.29,"ack":true,"ts":1618294402821,"q":0,"from":"system.adapter.modbus.4","user":"system.user.admin","lc":1618294402821}
2021-04-13 08:13:22.831  - [35msilly[39m: modbus.4 (2322) States user redis pmessage modbus.4.*/modbus.4.holdingRegisters.Set_charging_current:{"val":11100,"ack":false,"ts":1618294402828,"q":0,"from":"system.adapter.admin.0","user":"system.user.admin","lc":1618294402828}
2021-04-13 08:13:22.838  - [34mdebug[39m: modbus.4 (2322) Write successfully [5004]: 11100

**Your modified version:**

> 2021-04-13 08:16:01.694  - [35msilly[39m: modbus.4 (2384) States user redis pmessage modbus.4.*/modbus.4.holdingRegisters.5004_Set_charging_current:{"val":11200,"ack":false,"ts":1618294561690,"q":0,"from":"system.adapter.admin.0","user":"system.user.admin","lc":1618294561690}
2021-04-13 08:16:02.454  - [34mdebug[39m: modbus.4 (2384) Poll device 255
2021-04-13 08:16:02.455  - [34mdebug[39m: modbus.4 (2384) Poll holdingRegs DevID(255) address 1000 - 2 bytes
2021-04-13 08:16:03.203  - [35msilly[39m: modbus.4 (2384) States user redis pmessage modbus.4.*/modbus.4.holdingRegisters.5004_Set_charging_current:{"val":null,"ack":true,"ts":1618294563200,"q":0,"from":"system.adapter.modbus.4","user":"system.user.admin","lc":1618294563200}
2021-04-13 08:16:06.700  - [33mwarn[39m: modbus.4 (2384) Error: undefined
2021-04-13 08:16:06.700  - [31merror[39m: modbus.4 (2384) Request timed out.
2021-04-13 08:16:06.701  - [31merror[39m: modbus.4 (2384) Client in error state.
2021-04-13 08:16:06.702  - [31merror[39m: modbus.4 (2384) Cannot write [5004]: {"err":"timeout","timeout":5000}
2021-04-13 08:16:06.703  - [34mdebug[39m: modbus.4 (2384) Socket closed with error
2021-04-13 08:16:06.703  - [34mdebug[39m: modbus.4 (2384) Clearing timeout of the current request.
2021-04-13 08:16:06.708  - [34mdebug[39m: modbus.4 (2384) Cleaning up request fifo.
2021-04-13 08:16:07.656  - [33mwarn[39m: modbus.4 (2384) Poll error count: 1 code: "App Timeout"
2021-04-13 08:16:07.659  - [35msilly[39m: modbus.4 (2384) States user redis pmessage modbus.4.*/modbus.4.info.connection:{"val":false,"ack":true,"ts":1618294567658,"q":0,"from":"system.adapter.modbus.4","user":"system.user.admin","lc":1618294567658}
2021-04-13 08:16:07.701  - [34mdebug[39m: modbus.4 (2384) Closing client on purpose.
2021-04-13 08:16:07.702  - [32minfo[39m: modbus.4 (2384) Disconnected from slave 192.168.11.16

Apollon77, 2021-04-13:
I have an issue with the logs and that you say that all normal polling work. because I see in the last two logs that when polling only adress 1000 is polled and then you get kind of an timeout ... is that true? because then it is more nor working then just the write ...

Can you please provide a complete log (not cutting areround and making oarts on top and stuff)

Apollon77, 2021-04-13:
Please also get anew from hithub, I added some more debug

Apollon77, 2021-04-13:
PS: kann es sein das Du geschrieben hast als er gelesen hatte ... weil "App timeout" kommt beim polling ... starnge

riedesign, 2021-04-13:
Hoffe dieses Mal ist alles drinnen.
Logfile geht vom Neustart des Adapters bis zum update des States 5004 auf den Wert 11999

[log_1.txt](https://github.com/ioBroker/ioBroker.modbus/files/6304838/log_1.txt)

Apollon77, 2021-04-14:
@riedesign Also wenns bei einem tut und bei einem nicht ist es sehr strange weil der Adapter macht nix anderes ... keine Idee ehrlich gesagt

@anhomb Hast Du denn so ein gerät? Weil bei dir kommt ein klarer Fehler
2021-04-13 21:48:16.932  - [31merror[39m: modbus.1 (10695) Cannot write [5004]: {"errorCode":131,"exceptionCode":1,"message":"ILLEGAL FUNCTION"}

Starter4711, 2021-04-17:
Hallo, hätte Zeit könnte mich einbringen.
Log ohne write only parameter:
`

2021-04-17 13:21:27.682 - info: host.raspi4iobrokerpv "system.adapter.modbus.2" enabled
--
2021-04-17 13:21:27.803 - info: host.raspi4iobrokerpv instance system.adapter.modbus.2 started with pid 20045
2021-04-17 13:21:28.762 - debug: modbus.2 (20045) Redis Objects: Use Redis connection: 127.0.0.1:9001
2021-04-17 13:21:28.796 - debug: modbus.2 (20045) Objects client ready ... initialize now
2021-04-17 13:21:28.798 - debug: modbus.2 (20045) Objects create System PubSub Client
2021-04-17 13:21:28.799 - debug: modbus.2 (20045) Objects create User PubSub Client
2021-04-17 13:21:28.800 - debug: modbus.2 (20045) Objects client initialize lua scripts
2021-04-17 13:21:28.814 - debug: modbus.2 (20045) Objects connected to redis: 127.0.0.1:9001
2021-04-17 13:21:28.824 - debug: modbus.2 (20045) objectDB connected
2021-04-17 13:21:28.826 - debug: modbus.2 (20045) Redis States: Use Redis connection: 127.0.0.1:9000
2021-04-17 13:21:28.835 - debug: modbus.2 (20045) States create System PubSub Client
2021-04-17 13:21:28.836 - debug: modbus.2 (20045) States create User PubSub Client
2021-04-17 13:21:28.844 - debug: modbus.2 (20045) States connected to redis: 127.0.0.1:9000
2021-04-17 13:21:28.846 - debug: modbus.2 (20045) statesDB connected
2021-04-17 13:21:29.104 - debug: modbus.2 (20045) Plugin sentry Initialize Plugin (enabled=true)
2021-04-17 13:21:29.379 - info: modbus.2 (20045) starting. Version 3.2.4 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.22.0, js-controller: 3.2.16
2021-04-17 13:21:30.095 - info: modbus.2 (20045) Connected to slave 192.168.0.97
2021-04-17 13:21:30.096 - debug: modbus.2 (20045) Poll device 255
2021-04-17 13:21:30.108 - debug: modbus.2 (20045) Initialization of scale factors done!
2021-04-17 13:21:30.108 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1000 - 2 bytes
2021-04-17 13:21:30.122 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1004 - 2 bytes
2021-04-17 13:21:30.128 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1006 - 2 bytes
2021-04-17 13:21:30.132 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1008 - 2 bytes
2021-04-17 13:21:30.137 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1010 - 2 bytes
2021-04-17 13:21:30.141 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1012 - 2 bytes
2021-04-17 13:21:30.145 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1013 - 2 bytes
2021-04-17 13:21:30.150 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1014 - 2 bytes
2021-04-17 13:21:30.155 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1016 - 2 bytes
2021-04-17 13:21:30.159 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1020 - 2 bytes
2021-04-17 13:21:30.163 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1036 - 2 bytes
2021-04-17 13:21:30.169 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1040 - 2 bytes
2021-04-17 13:21:30.173 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1042 - 2 bytes
2021-04-17 13:21:30.178 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1044 - 2 bytes
2021-04-17 13:21:30.183 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1046 - 2 bytes
2021-04-17 13:21:30.187 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1100 - 2 bytes
2021-04-17 13:21:30.191 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1110 - 2 bytes
2021-04-17 13:21:30.195 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1500 - 2 bytes
2021-04-17 13:21:30.200 - debug: modbus.2 (20045) Poll holdingRegs DevID(255) address 1502 - 2 bytes
2021-04-17 13:21:31.258 - debug: modbus.2 (20045) Poll device 255

`

log mit einem write only param:
`

2021-04-17 13:23:16.602 - info: host.raspi4iobrokerpv "system.adapter.modbus.2" enabled
--
2021-04-17 13:23:16.705 - info: host.raspi4iobrokerpv instance system.adapter.modbus.2 started with pid 20061
2021-04-17 13:23:17.712 - debug: modbus.2 (20061) Redis Objects: Use Redis connection: 127.0.0.1:9001
2021-04-17 13:23:17.820 - debug: modbus.2 (20061) Objects client ready ... initialize now
2021-04-17 13:23:17.845 - debug: modbus.2 (20061) Objects create System PubSub Client
2021-04-17 13:23:17.847 - debug: modbus.2 (20061) Objects create User PubSub Client
2021-04-17 13:23:17.848 - debug: modbus.2 (20061) Objects client initialize lua scripts
2021-04-17 13:23:17.973 - debug: modbus.2 (20061) Objects connected to redis: 127.0.0.1:9001
2021-04-17 13:23:17.992 - debug: modbus.2 (20061) objectDB connected
2021-04-17 13:23:17.997 - debug: modbus.2 (20061) Redis States: Use Redis connection: 127.0.0.1:9000
2021-04-17 13:23:18.011 - debug: modbus.2 (20061) States create System PubSub Client
2021-04-17 13:23:18.013 - debug: modbus.2 (20061) States create User PubSub Client
2021-04-17 13:23:18.021 - debug: modbus.2 (20061) States connected to redis: 127.0.0.1:9000
2021-04-17 13:23:18.022 - debug: modbus.2 (20061) statesDB connected
2021-04-17 13:23:18.241 - debug: modbus.2 (20061) Plugin sentry Initialize Plugin (enabled=true)
2021-04-17 13:23:18.540 - info: modbus.2 (20061) starting. Version 3.2.4 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.22.0, js-controller: 3.2.16
2021-04-17 13:23:19.096 - info: modbus.2 (20061) Connected to slave 192.168.0.97
2021-04-17 13:23:19.098 - debug: modbus.2 (20061) Poll device 255
[... cut ...]
2021-04-17 13:23:19.195 - debug: modbus.2 (20061) Poll holdingRegs DevID(255) address 1500 - 2 bytes
2021-04-17 13:23:19.207 - debug: modbus.2 (20061) Poll holdingRegs DevID(255) address 1502 - 2 bytes
2021-04-17 13:23:19.218 - debug: modbus.2 (20061) Poll holdingRegs DevID(255) address 5004 - 1 bytes
2021-04-17 13:23:19.227 - debug: modbus.2 (20061) Received pdu describes an error.
2021-04-17 13:23:19.229 - warn: modbus.2 (20061) Poll error count: 1 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}
2021-04-17 13:23:20.280 - debug: modbus.2 (20061) Poll device 255

`

Apollon77, 2021-04-17:
Bitte 3.3.0 versuchen. Ich habe es jetzt mal released weil es ja eh nur probleme macht wenn mal "poll=false "states hat und nutzt ... bitte mal damit versuchen

Und ein "illegal Address "fehler heisst das das modbux device das antwortet ... dann ist der Ounkt falsch für das Gerät

riedesign, 2021-04-17:
@Apollon77 vielen Dank für deine Hilfe hier!
> @riedesign Also wenns bei einem tut und bei einem nicht ist es sehr strange weil der Adapter macht nix anderes ... keine Idee ehrlich gesagt
Echt seht strange. Bei der 3.2.6. official habe ich (wenn ich den state manuell hinzugefügt habe) diesen erfolgreichen schreibvorgang durchführen können - dies ist sehr merkwürdig, dass es da klappt, aber jetzt nicht mehr.

Testlog bei Version 3.2.6
>2021-04-17 22:19:28.662  - [34mdebug[39m: modbus.5 (41939) Poll holdingRegs DevID(255) address 1000 - 2 bytes
2021-04-17 22:19:28.666  - [34mdebug[39m: modbus.5 (41939) Poll holdingRegs DevID(255) address 1004 - 2 bytes
2021-04-17 22:19:28.676  - [34mdebug[39m: modbus.5 (41939) Poll holdingRegs DevID(255) address 1008 - 2 bytes
2021-04-17 22:19:28.679  - [34mdebug[39m: modbus.5 (41939) Poll holdingRegs DevID(255) address 1010 - 2 bytes
2021-04-17 22:19:28.682  - [34mdebug[39m: modbus.5 (41939) Poll holdingRegs DevID(255) address 1012 - 2 bytes
2021-04-17 22:19:28.686  - [34mdebug[39m: modbus.5 (41939) Poll holdingRegs DevID(255) address 1020 - 2 bytes
2021-04-17 22:19:28.689  - [34mdebug[39m: modbus.5 (41939) Poll holdingRegs DevID(255) address 1100 - 2 bytes
2021-04-17 22:19:28.695  - [34mdebug[39m: modbus.5 (41939) Poll holdingRegs DevID(255) address 1500 - 2 bytes
2021-04-17 22:19:28.697  - [34mdebug[39m: modbus.5 (41939) Poll holdingRegs DevID(255) address 1502 - 2 bytes
2021-04-17 22:19:33.426  - [35msilly[39m: modbus.5 (41939) States user redis pmessage modbus.5.*/modbus.5.holdingRegisters.Set_charging_current:{"val":11000,"ack":false,"ts":1618690773423,"q":0,"from":"system.adapter.admin.0","user":"system.user.admin","lc":1618690773423}
2021-04-17 22:19:33.439  - [34mdebug[39m: modbus.5 (41939) Write successfully [5004]: 11000
2021-04-17 22:19:33.750  - [34mdebug[39m: modbus.5 (41939) Poll device 255

Habe nun die 3.3.0 getestet, jedoch noch immer das gleiche Problem, dass dieser Error kommt. Habs auch mit verschiedenen Einstellungen (Schreibintervalle,...) getestet - immer der gleihce Fehler.
>21-04-17 21:46:49.666  - [33mwarn[39m: modbus.5 (41109) Error: undefined
2021-04-17 21:46:49.669  - [31merror[39m: modbus.5 (41109) Request timed out.
2021-04-17 21:46:49.670  - [31merror[39m: modbus.5 (41109) Client in error state.
2021-04-17 21:46:49.671  - [31merror[39m: modbus.5 (41109) Cannot write [5004]: {"err":"timeout","timeout":5000}
2021-04-17 21:46:49.673  - [34mdebug[39m: modbus.5 (41109) Socket closed with error

Hier das ganze Logfile 
[iobroker.2021-04-17.log](https://github.com/ioBroker/ioBroker.modbus/files/6329950/iobroker.2021-04-17.log)

Hab in der Anleitung der Ladestation nochmals nachgelesen. Da steht, dass der Schreibvorgang als FC6 durchgeführt werden muss. Leider kenne ich mich hier zu wenig aus, aber vielleicht sagt es dir was.
Quelle: [https://www.keba.com/download/x/cfaaba8c72/kecontactp30modbustcp_pgen.pdf](url) Seite 8

Apollon77, 2021-04-17:
Am Ende ist in der Version nur noch https://github.com/ioBroker/ioBroker.modbus/commit/260360e757b7afdfb24643fed411ec6edbb0db04#diff-7457e37f73144333d9f610292348a1b7a4e3d1872d373723831cbc4d0d145f7cL439 drin was in Issue https://github.com/ioBroker/ioBroker.modbus/issues/78 diskutiert wurde ... ggf mal da syncen?!

k3mpaxl, 2021-04-18:
Hallo zusammen, ich habe mit der Version mal probiert, ob es bei mir klappt. Tatsächlich werden die States nun angelegt, aber das Schreiben schlägt fehl.
Ich teste noch einmal ausführlich durch und danach gebe hier auch meine Rückmeldung.

Starter4711, 2021-04-18:
Hallo, ich habe nub 3.3.0 installiert, der ow Wert wird nun als Objekt angelegt, super.
Wenn ich dort einen zulässigen Wert (6000-32000) Bsp. 8000 eintrage, verschwindet der Wert u der Adapter geht offline.
[image]
[image]
`2021-04-18 13:53:23.424  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1000 - 2 bytes
2021-04-18 13:53:23.427  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1004 - 2 bytes
2021-04-18 13:53:23.430  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1006 - 2 bytes
2021-04-18 13:53:23.433  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1008 - 2 bytes
2021-04-18 13:53:23.440  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1010 - 2 bytes
2021-04-18 13:53:23.443  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1012 - 2 bytes
2021-04-18 13:53:23.446  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1013 - 2 bytes
2021-04-18 13:53:23.452  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1014 - 2 bytes
2021-04-18 13:53:23.455  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1016 - 2 bytes
2021-04-18 13:53:23.459  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1020 - 2 bytes
2021-04-18 13:53:23.462  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1036 - 2 bytes
2021-04-18 13:53:23.465  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1040 - 2 bytes
2021-04-18 13:53:23.468  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1042 - 2 bytes
2021-04-18 13:53:23.471  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1044 - 2 bytes
2021-04-18 13:53:23.476  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1046 - 2 bytes
2021-04-18 13:53:23.481  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1100 - 2 bytes
2021-04-18 13:53:23.484  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1110 - 2 bytes
2021-04-18 13:53:23.487  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1500 - 2 bytes
2021-04-18 13:53:23.490  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1502 - 2 bytes
2021-04-18 13:53:23.652  - [34mdebug[39m: modbus.2 (11487) state Changed ack=false: modbus.2.holdingRegisters.5004_set_charging current: {"val":8000,"ack":false,"ts":1618746803649,"q":0,"from":"system.adapter.admin.0","user":"system.user.admin","lc":1618746803649}
2021-04-18 13:53:23.657  - [34mdebug[39m: modbus.2 (11487) Received pdu describes an error.
2021-04-18 13:53:23.658  - [31merror[39m: modbus.2 (11487) Cannot write [5004]: {"errorCode":131,"exceptionCode":1,"message":"ILLEGAL FUNCTION"}
2021-04-18 13:53:23.659  - [34mdebug[39m: modbus.2 (11487) Closing client on purpose.
2021-04-18 13:53:23.660  - [32minfo[39m: modbus.2 (11487) Disconnected from slave 192.168.0.97
2021-04-18 13:53:23.664  - [34mdebug[39m: modbus.2 (11487) Socket closed with error
2021-04-18 13:53:23.664  - [34mdebug[39m: modbus.2 (11487) Cleaning up request fifo.
2021-04-18 13:53:24.662  - [32minfo[39m: modbus.2 (11487) Connected to slave 192.168.0.97
2021-04-18 13:53:24.663  - [34mdebug[39m: modbus.2 (11487) Poll device 255
2021-04-18 13:53:24.663  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1000 - 2 bytes
2021-04-18 13:53:24.666  - [34mdebug[39m: modbus.2 (11487) Closing client on purpose.
2021-04-18 13:53:24.666  - [32minfo[39m: modbus.2 (11487) Disconnected from slave 192.168.0.97
2021-04-18 13:53:24.669  - [34mdebug[39m: modbus.2 (11487) Poll holdingRegs DevID(255) address 1004 - 2 bytes
2021-04-18 13:53:24.681  - [31merror[39m: modbus.2 (11487) Socket Error
2021-04-18 13:53:24.682  - [31merror[39m: modbus.2 (11487) Client in error state.
2021-04-18 13:53:24.683  - [33mwarn[39m: modbus.2 (11487) On error: {"code":"ERR_STREAM_WRITE_AFTER_END"}
2021-04-18 13:53:24.684  - [34mdebug[39m: modbus.2 (11487) Socket closed with error
2021-04-18 13:53:24.684  - [34mdebug[39m: modbus.2 (11487) Clearing timeout of the current request.
2021-04-18 13:53:24.685  - [34mdebug[39m: modbus.2 (11487) Cleaning up request fifo.
2021-04-18 13:53:25.684  - [34mdebug[39m: modbus.2 (11487) Closing client on purpose.`

Apollon77, 2021-04-18:
> Cannot write [5004]: {"errorCode":131,"exceptionCode":1,"message":"ILLEGAL FUNCTION"}

Das ist der Fehler vom Device und danach disconnected es ... und der Adater sollte danach neu verbinden.

Apollon77, 2021-04-18:
WIe oben geschrieben macht doch mal die eine andere Änderung rückgängig .-... geht es vllt dann? Dann muss man doch da nochmal schauen

Starter4711, 2021-04-18:
Weiß nicht was du meinst mit anderer Änderung?
schon klar was der Fehler sagt, die Frage ist eher warum der kommt, die Keba sollte den Wert ja nehmen.

Apollon77, 2021-04-18:
In https://github.com/ioBroker/ioBroker.modbus/commit/260360e757b7afdfb24643fed411ec6edbb0db04#diff-7457e37f73144333d9f610292348a1b7a4e3d1872d373723831cbc4d0d145f7cL439 wurde "objects[id].native.len > 1 && " aus dem "if" entfernt. macht es mal wieder rein, tut es dann?

Starter4711, 2021-04-18:
OK, da ich kein Java kann, müsste ich genau Anweisungen haben, das zu ändern.

riedesign, 2021-04-19:
> 
> 
> In [260360e#diff-7457e37f73144333d9f610292348a1b7a4e3d1872d373723831cbc4d0d145f7cL439](https://github.com/ioBroker/ioBroker.modbus/commit/260360e757b7afdfb24643fed411ec6edbb0db04#diff-7457e37f73144333d9f610292348a1b7a4e3d1872d373723831cbc4d0d145f7cL439) wurde "objects[id].native.len > 1 && " aus dem "if" entfernt. macht es mal wieder rein, tut es dann?

DANKE! das war der Fehler. Nun klappt es wunderbar.

Starter4711, 2021-04-19:
Super, ich kann das nicht ändern, wird hoff. dann ein neues Release ;)

riedesign, 2021-04-19:
ich hab das repository geforkt und diese eine Änderung dran vorgenommen. 
Zum testen: https://github.com/riedesign/ioBroker.modbus (über Katze installieren)

Apollon77, 2021-04-19:
Ja überlegen gerade wie am besten ...

Starter4711, 2021-05-02:
Hi @Apollon77 möchte freundlich nachfragen, ob eine Lösung in Sicht wäre. Vielen Dank.

Apollon77, 2021-05-02:
Es gab andere Themen die meine Zeit recht arg auffressen, ich muss schauen wann ich dazu komme

GermanBluefox, 2021-06-12:
Please test 3.3.2