ioBroker/ioBroker.modbus #78: Read/Write Multible Registers

EtzeSte13, 2020-10-23:
Hello,

I tried to use a SPR8 temrature controler from Skrjl via modbus adapter with the IO-Broker. Unfortunately it did not work since it is not possible to send the modbus function 03 - Read holding registers  or 16 - set/write multiple registers  --> Or I was not able to send it. 

Enclosed you can find the protocol which is needed for that sensor. Maybe you have tip for me how this can be solved. 

Just for your information. The python module minimal modbus is working with that sensor by using the multiple read and multiple register read function.

[communication_spr8.pdf](https://github.com/ioBroker/ioBroker.modbus/files/5426542/communication_spr8.pdf)

Since I want to introduce that sensors in io Broker it would be very helpful to find some good tip here. 

My best regards
Etze

TmShaz, 2020-11-02:
See #54

EtzeSte13, 2020-11-03:
Thank you very much for this link. This descripes the same problem I have. Also my sensor this SPR8 excepts only FC03 for read  and  FC16 to write. So I guess a option "use read/write multiple registers" which uses only FC03 and FC16 would solve my issue.

Would be great to have this in on of the next versions..... 

my best regards!!

EtzeSte13, 2020-11-03:
> 
> 
> See #54

Can you tell me in detail where and how I need to introduce the suggested change in the code? I have already installed the modbus adapter in my iobroker so I need to know in which on my Raspi 4 i need to uptade to run the changed file. I could not find this file in the file structure yet....

Can you please let me know how this works?

Thanks for your help

TmShaz, 2020-11-03:
Go to /opt/iobroker/node_modules/iobroker.modbus/lib
There you will find a file master.js

Open this file in editor and change the line 410 of the code from:
if (objects[id].native.len > 1 && !options.config.doNotUseWriteMultipleRegisters) {

to :
if ( !options.config.doNotUseWriteMultipleRegisters) {

This worked for me but I have no idea what I am doing :)

mcmuller, 2020-11-04:
@TmShaz  solved my write problem on an ELIWELL Heatpump Controller as well!

EtzeSte13, 2020-11-06:
Thank you for explaination!! W will try this soon...

Apollon77, 2020-11-12:
I could implement this change, but honestly ... I have no idea what this change can break. Can anyone tell me more ?

clausob, 2020-11-12:
It forces to use the FC16 even when only one register is to be written. Not a clean solution, that's the reason 
why I proposed to introduce a an option "force to write multiple register" in thread #54 in order to address the situation 
when a device only accepts FC16.

Best regards,

Claus

peter605, 2021-01-22:
@TmShaz solved my write problem on an Nilan Heatpump Controller as well

modbus.0 | 2021-01-22 16:22:28.395 | info | (24432) Disconnected from slave
-- | -- | -- | --
modbus.0 | 2021-01-22 16:22:27.394 | error | (24432) Cannot write [1003]: {"err":"timeout","timeout":5000}
modbus.0 | 2021-01-22 16:22:27.393 | error | (24432) Request timed out.
modbus.0 | 2021-01-22 16:22:27.391 | warn | (24432) Error: undefined
modbus.0 | 2021-01-22 16:22:22.420 | error | (24432) unknown function code: 134

from the Manual 
Supported functions

Function | Name | Description
 -- |  -- | -- 
03 | Read Holding Registers | Read one or more holding registers
04 | Read Input Registers | Read one or more input registers
16 | Preset Multiple Registers | Write one or more holding registers

adavt, 2021-03-24:
@TmShaz solved my write problem on OWEN TPM210 PID controller as well! There are no other problems associated with changing the file.

Apollon77, 2021-04-17:
siehe https://github.com/ioBroker/ioBroker.modbus/issues/77#issuecomment-821881994 ... scheinbar gibts doch Issues?!

Apollon77, 2021-04-19:
So, confirmed ... der Obige Change macht andere Fälle kaputt :-((

https://github.com/ioBroker/ioBroker.modbus/issues/77#issuecomment-822167758

Was machmer nun?

Ich denke ein Setting ist dann das einzieg was sinnmacht.
Ist "force to write multiple register" genug info? oder sollte noch was anderes hin?

TmShaz, 2021-04-19:
Ja, setting sollte es auch sein. So wie es Claus vorgeschlagen hatte. Sorry, falls das anders rüber kam. “force to write multiple registers” reicht denke ich. In der Beschreibung kann man dann näher drauf eingeben.

Apollon77, 2021-04-19:
> In der Beschreibung kann man dann näher drauf eingeben.
Die liest doch eh keiner ;-)) Daher war die frage ob man in der checkbox bescfhreibung gleich mehr infos geben kann, wie "always use Fc6 also when requesting single datapoint" (oder so ... bin in modbus nicht tief drin

TmShaz, 2021-05-30:
Ok, überzeugt. Vorschlag: "Always use FC16, even when writing single datapoint. (Some slaves only accepts FC16)"

Apollon77, 2021-05-30:
Ja das sterht nicht aus .. korrekt ... Zeit ist einfach zu knapp für alles. Sorry

Apollon77, 2021-06-12:
Will be fixed in coming update 3.4.0

closed 2021-06-12 by Apollon77

HJS72, 2021-06-16:
Hallo Appollon77,
Hab exakt das gleiche Problem. Meine Lüftungsanlage von Maico verlangt nach multi register (FC16)
 beim schreiben. Hab den aktuellen Adapter 3.4.2 und 3.4.4 versucht. Option nur FC16 ist aktiviert. Lesen funktioniert, beim Schreiben bekomme ich aber leider folgenden Fehler: 

(23251) Cannot write [550]: {"errorCode":134,"exceptionCode":1,"message":"ILLEGAL FUNCTION"}.

Hast du vielleicht noch eine Idee? Mit ModbusPoll konnte ich das Register erfolgreich ändern.

Schon mal vielen Dank für die Hilfe
Gruß Stefan

danimeierch, 2022-03-03:
Hi! I have still the same issue / error showing up... is it already fixed and it's something else or is still pending? Thanks for your help and thank you for the awesome plugin at all. Regards, Daniel

Apollon77, 2022-03-03:
Which options are set how? Did you tried several options? COmmenting on closed issues is not that helpful ... so open a new one with exact details and logs please

commit 260360e7 (2021-04-11): * (Apollon77/TmShaz) Fix Write multiple registers, fixes #78
* prepare weblate
* update deps
* add debug