ioBroker/ioBroker.modbus #83: Modbus adapter wird in Admin nicht korrekt angezeigt

nkleber78, 2020-12-12:
Im Admin wird der modbus Adapter nicht korrekt angezeigt. Es scheint dass irgendwo eine Kleinigkeit im Code nicht passt sodass z.b. löschen von Instanzen usw. je nach Ansicht nicht oder nicht sauber dargestellt wird. Siehe screenshots
[image]
[image]

GermanBluefox, 2020-12-12:
Ich kann nicht reproduzieren

nkleber78, 2020-12-13:
Hallo @GermanBluefox 
Ich habe die Ursache gefunden. Das problem war dass bei system.adapter.modbus.upload 1% drin war. Dies führt dann dazu dass es nicht korrekt angezeigt wird. Ich vermute der Zahlenwert habe ich versehentlich dort eingetragen... Vielleicht kann sich ja jemand dieses Thema im Admin anschauen. Denn es ist reproduzierbar für jeden Adapter...

closed 2020-12-13 by nkleber78