ioBroker/ioBroker.modbus #84: Port not available for TCP -Slave

NiessenJuergen, 2020-12-25:
**Describe the bug**  
Configuerd as TCP Slave, no port is availible. I tried port 502, 1502 and 2502, tried with a Raspberry Pi2 B, Raspberry Pi4 and Ubuntu 18.04.5 LTS, Tried 3.2.2 and 3.1.10. 

Tried partner IP address, 0.0.0.0, 127.0.0.1, Ubuntu 18.04.5 LTS and own IP as partner IP address

telnet to port x52 from other PC won't work, telnet from system to localhost:502 wont work,  netstat -tulpn | grep 502 shows no result, adapter stays yellow

**Versions:**  

 - JS-Controller version: <js-controller-version>  3.1.6
 - Node version: <node-version>  v15.3.0
 - Operating system: Ubuntu 18.04.5 LTS

Apollon77, 2020-12-26:
In the Admin confog a port entry should be shown to enter the port. Please show admin screenshot with "Slave" selected

NiessenJuergen, 2020-12-26:
[image]

Apollon77, 2020-12-26:
Hm ... as slave the adapter itself connects to the IP and port provided ... that port is not offered ... because a slave is just a client device that conects to the master. WHat was your expectation?

NiessenJuergen, 2020-12-26:
That was, what I also thought first. But then i found out, that on my master port 502 is not open. Then i dug a little deeper in the topic of modbus and found out, that the master(client) connects to several slaves (server) on port 502. Then i tried it with several tools (diagslave, modpoll) and found out, the it was that way round. 

Then i tested with FHEM. 
`define cmi_modbus ModbusAttr 123 slave 172.19.10.123:3502 TCP`
An as I expected a "telnet 172.19.10.123 3502" worked.

Then i tested ioBroker.modbus with type "Master" because i thought, maybe there was a mix up with master/slave in the dropdown, but even then the port wasn't open.

I chose ports above 1024, as FHEM (and ioBroker i think) cant open elevated ports bellow 1024 without root access.

GermanBluefox, 2021-06-14:
Does the problem still exist?
You should set the IP to 0.0.0.0

closed 2021-06-24 by GermanBluefox