ioBroker/ioBroker.modbus #117: Holding Register write error with 3.3.0

Strobelix80, 2021-05-07:
Error when writing holding register in version 3.3.0.

Works until v 3.2.6

Logfile:

2021-05-07 22:00:43.067  - [34mdebug[39m: modbus.0 (3017) state Changed ack=false: modbus.0.holdingRegisters.8052_Betriebsart_Heizkreis 7: {"val":0,"ack":false,"ts":1620417643058,"q":0,"from":"system.adapter.admin.0","user":"system.user.admin","lc":1620417643058}
2021-05-07 22:00:43.097  - [31merror[39m: modbus.0 (3017) unknown function code: 144
2021-05-07 22:00:44.797  - [34mdebug[39m: modbus.0 (3017) system.adapter.admin.0: logging true
2021-05-07 22:00:48.085  - [33mwarn[39m: modbus.0 (3017) Error: undefined
2021-05-07 22:00:48.087  - [31merror[39m: modbus.0 (3017) Request timed out.
2021-05-07 22:00:48.089  - [31merror[39m: modbus.0 (3017) Cannot write [8052]: {"err":"timeout","timeout":5000}
2021-05-07 22:00:49.091  - [32minfo[39m: modbus.0 (3017) Disconnected from slave

Versions:
 - Adapter version: 3.3.0
 - JS-Controller version: 3.3.8
 - Node version: 12.22.1
 - Operating system: linux

Apollon77, 2021-05-07:
@GermanBluefox I know the issue and will fix it soon!!

Apollon77, 2021-06-12:
Will be fixed in upcoming 3.4.0

closed 2021-06-12 by Apollon77