ioBroker/ioBroker.modbus #121: "modbus.0.info.connection" has to be type "boolean" but received type "number"

Diginix, 2021-05-26:
> 2021-05-22 02:28:37.967  - info: modbus.0 (1649) State value to set for "modbus.0.info.connection" has to be type "boolean" but received type "number"

History:
`true	true	modbus.0	2021-05-22 02:28:38.335
false	true	modbus.0	2021-05-22 02:28:38.320
0	true	modbus.0	2021-05-22 02:28:37.973
true	true	modbus.0	2021-05-20 11:57:20.014
false	true	modbus.0	2021-05-20 11:57:20.005
0	true	modbus.0	2021-05-20 11:57:19.748
true	true	modbus.0	2021-05-20 10:58:34.423
false	true	modbus.0	2021-05-20 10:58:34.409
0	true	modbus.0	2021-05-20 10:58:34.097
true	true	modbus.0	2021-05-20 00:34:26.457`

RAW:
`"_id": "modbus.0.info.connection",
"type": "state",
"common": {
  "role": "indicator.connected",
  "name": "If master connected",
  "type": "boolean",
  "read": true,
  "write": false,
  "def": false
}`

Adapter 3.3.1
JS-Controller 3.3.10
NodeJS 12.22.1

GermanBluefox, 2021-06-12:
Fixed

Diginix, 2021-06-16:
I still get this error with 3.4.1 and 3.4.2

```
2021-06-16 08:08:47.652 - info: modbus.0 (1073146) starting. Version 3.4.2 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.22.1, js-controller: 3.3.12
2021-06-16 08:08:47.718 - info: modbus.0 (1073146) State value to set for "modbus.0.info.connection" has to be type "boolean" but received type "number" 
2021-06-16 08:08:48.046 - info: modbus.0 (1073146) Connected to slave 192.168.2.106
```

GermanBluefox, 2021-06-16:
Please check the 3.4.3

Diginix, 2021-06-16:
First start with 3.4.4 looks good. No wrong data type warnings.

closed 2021-06-17 by GermanBluefox

sv-fepa, 2021-08-17:
I got the same error warning for an own adapter. 
The problem is there, on Vis toggle button, the changing values are 0 and 1 for example for lamp on and off.
But internal logic of the adapter works with boolean true and false to toggle lamp on and off.

How can one fix the "wrong" vis toggle?

Apollon77, 2021-08-17:
Please create an own issue where it belongs to, adding a comment to a closed issue in a complete different adapter makes no sense