ioBroker/ioBroker.modbus #132: "Request timed out" if two or more parallel polls

5G7K, 2021-06-14:
**Describe the bug**  
If a request is splitted into two or more polls, in my configuration, a timeout occurs.
See: [Problem description](https://forum.iobroker.net/post/642134)

**To Reproduce**  
Maybe only to reproduce in my configuration? Maybe a problem of my modbus slave?

**Versions:**  
 - Adapter version: 3.3.1
 - JS-Controller version: js-controller: 3.2.16
 - Node version: node: v12.22.1
 - Operating system: Raspberry

**Additional context**  
I solved the problem by a "dirty" implementation of a 50ms sleep between each request.
See my changes in master.js

Please check if this is a good solution or how this can be implemented in a better way.

Thank you!
Andreas

`    function sleep(milliseconds) {       
      var start = new Date().getTime();
      for (var i = 0; i < 1e7; i++) {
        if ((new Date().getTime() - start) > milliseconds){
          break;
        }
      }
    }

    function pollBinariesBlock(device, regType, func, block, callback) {
        let regs = device[regType];
        if (block >= regs.blocks.length) {
            return callback();
        }

        const regBlock = regs.blocks[block];

        if (regBlock.startIndex === regBlock.endIndex) {
            regBlock.endIndex++;
        }

        adapter.log.debug(`Poll ${regType} DevID(${regs.deviceId}) address ${regBlock.start} - ${regBlock.count} bits`);
        **sleep(50);**
        
        if (modbusClient) {
            modbusClient[func](regs.deviceId, regBlock.start, regBlock.count)
                .then(response => {
                    if (response.data && response.data.length) {
                        for (let n = regBlock.startIndex; n < regBlock.endIndex; n++) {
                            let id = regs.config[n].id;
                            let val = response.data[regs.config[n].address - regBlock.start];

                            if (options.config.alwaysUpdate || ackObjects[id] === undefined || ackObjects[id].val !== val) {
                                ackObjects[id] = {val: val};
                                adapter.setState(id, !!val, true, err => {
                                    // analyse if the state could be set (because of permissions)
                                    err && adapter.log.error('Can not set state ' + id + ': ' + err);
                                });
                            }
                        }
                    } else {
                        adapter.log.warn(`Null buffer length ${func} for ${regType} ${regBlock.start}`);
                    }
                    setImmediate(() => pollBinariesBlock(device, regType, func, block + 1, callback));
                })
                .catch(err => callback(err));
        } else {
            adapter.log.debug(`Poll canceled, because no connection`);
            callback('No connection');
        }
    }

    function pollBinariesBlocks(device, regType, func, callback) {
        let regs = device[regType];
        if (regs.length) {
            pollBinariesBlock(device, regType, func, 0, err =>
                callback(err));
        } else {
            callback();
        }
    }

    function pollFloatBlock(device, regType, func, block, callback) {
        let regs = device[regType];

        if (block >= regs.blocks.length) {
            return callback();
        }
        const regBlock = regs.blocks[block];

        if (regBlock.startIndex === regBlock.endIndex) {
            regBlock.endIndex++;
        }
        if (!scaleFactors[regs.deviceId]){
            adapter.log.debug('Initialization of scale factors done!');
            scaleFactors[regs.deviceId] = {};
        }

        adapter.log.debug(`Poll ${regType} DevID(${regs.deviceId}) address ${regBlock.start} - ${regBlock.count} bytes`);
        **sleep(50);**          
        if (modbusClient) {
            modbusClient[func](regs.deviceId, regBlock.start, regBlock.count)
                .then(response => {
                    if (response.payload && response.payload.length) {
                        for (let n = regBlock.startIndex; n < regBlock.endIndex; n++) {
                            let id = regs.config[n].id;
                            let val = common.extractValue(regs.config[n].type, regs.config[n].len, response.payload, regs.config[n].address - regBlock.start);
                            **adapter.log.debug('ID: ' + id + ' Val: ' + val);**    //just for logging                            
                            if (regs.config[n].type !== 'string' && regs.config[n].type !== 
                            'stringle') {
                                // If value must be calculated with formula
                                if (regs.config[n].formula) {
                                    adapter.log.debug('Input Value = ' + val);
                                    adapter.log.debug('Formula = ' + regs.config[n].formula);
                                    try {
                                        // calculate value from formula or report an error
                                        const func = new Function('x', 'sf', 'return ' + regs.config[n].formula);
                                        val = func(val, scaleFactors[regs.deviceId]);
                                        val = Math.round(val * options.config.round) / options.config.round;
                                    } catch (e) {
                                        adapter.log.warn('Calculation: eval(' + regs.config[n].formula + ') not possible: ' + e);
                                    }
                                } else {
                                    val = val * regs.config[n].factor + regs.config[n].offset;
                                    val = Math.round(val * options.config.round) / options.config.round;
                                }

                                // If this value is used as scale factor => store it
                                if (regs.config[n].isScale) {
                                    scaleFactors[regs.deviceId][regs.config[n].address] = val;
                                    adapter.log.debug('Scale factor value stored from address ' + regs.config[n].address + ' = ' + scaleFactors[regs.deviceId][regs.config[n].address]);
                                }
                            }

                            if (options.config.alwaysUpdate || ackObjects[id] === undefined || ackObjects[id].val !== val) {
                                ackObjects[id] = {val: val};
                                adapter.setState(id, val, true, err => {
                                    // analyse if the state could be set (because of permissions)
                                    err && adapter.log.error(`Can not set state ${id}: ${err}`);
                                });
                            }
                        }
                    } else {
                        adapter.log.warn(`Null buffer length ${func} for ${regType} ${regBlock.start}`);
                    }
                    // special case
                    if (options.config.maxBlock < 2 && regs.config[regBlock.startIndex].cw) {
                        // write immediately the current value
                        writeFloatsReg(device, regType, objects[regs.config[regBlock.startIndex].fullId], () => {
                            pollFloatBlock(device, regType, func, block + 1, callback);
                        });
                    } else {
                        setImmediate(() => pollFloatBlock(device, regType, func, block + 1, callback));
                    }
                })
                .catch(err => callback(err));
        } else {
            adapter.log.debug(`Poll canceled, because no connection`);
            callback('No connection');
        }
    }
`

GermanBluefox, 2021-06-14:
Please test 3.4.1. You can set now the "Read interval" that do the same.

5G7K, 2021-06-16:
Works with 3.4.2
Thank you!

closed 2021-06-16 by 5G7K