ioBroker/ioBroker.modbus #133: Keine Verbindung // Cannot create modbus master!

fuchs-1978, 2021-06-15:
Mit einem Update von 3.3.1 auf 3.4.1 verbindet sich der Adapter mit mehr mit meinem SMA Wechselrichter. Es kommt die Fehlermeldung:

Cannot create modbus master! C
Cannotread property 'timeout' of undefined

jk-andersen, 2021-06-15:
Habe das selbe hier:

`

modbus.1 | 2021-06-15 08:37:13.154 | error | Cannot create modbus master!
-- | -- | -- | --
modbus.1 | 2021-06-15 08:37:13.154 | error | Cannot connect to "192.168.178.103:502": TypeError: Cannot read property 'timeout' of undefined
modbus.0 | 2021-06-15 08:37:06.797 | error | Cannot create modbus master!
modbus.0 | 2021-06-15 08:37:06.797 | error | Cannot connect to "192.168.178.102:502": TypeError: Cannot read property 'timeout' of undefined

`

adavt, 2021-06-15:
Same problem. When the old version is returned, everything starts to work as before, the version number is lowered, but the new interface remains in the modbus driver settings.

PalmManiac, 2021-06-15:
Das gleiche bei mir. Bin erst mal wieder zurück auf die V3.3.1, die sich sofort wieder verbunden hat.

GermanBluefox, 2021-06-15:
Fixed in 3.4.2

fuchs-1978, 2021-06-16:
Funktioniert wieder, vielen Dank

jk-andersen, 2021-06-16:
Verbindung generell klappt, bekomme jetzt aber zyklische Fehler:

Hast du eine Idee? Konfig nicht verändert.

<div class="MuiGrid-root-272 MuiGrid-container-273 MuiGrid-item-274 MuiGrid-align-items-xs-center-281" style="box-sizing: border-box; width: 1549.43px; display: flex; flex-wrap: wrap; margin: 0px; align-items: center; color: rgba(0, 0, 0, 0.87); font-family: &quot;Segoe UI&quot;, Tahoma, Arial, &quot;Courier New&quot;, monospace; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><span class="MuiBadge-root-106" title="Fehler anzeigen" style="box-sizing: inherit; display: inline-flex; position: relative; flex-shrink: 0; vertical-align: middle;"><span class="MuiBadge-badge-107 jss2923 jss2924 MuiBadge-anchorOriginTopRightRectangle-112 MuiBadge-colorError-110" style="box-sizing: border-box; height: 20px; display: flex; padding: 0px 6px; z-index: 1; position: absolute; flex-flow: row wrap; font-size: 0.75rem; min-width: 20px; transition: transform 225ms cubic-bezier(0.4, 0, 0.2, 1) 0ms; align-items: center; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 500; line-height: 1; place-content: center; border-radius: 10px; color: rgb(255, 255, 255); background-color: rgb(244, 67, 54); top: 10px; right: 10px; transform: scale(1) translate(50%, -50%); transform-origin: 100% 0%;">4</span></span><span class="MuiBadge-root-106" title="Fehler und Warnungen anzeigen" style="box-sizing: inherit; display: inline-flex; position: relative; flex-shrink: 0; vertical-align: middle;"><button class="MuiButtonBase-root-85 MuiIconButton-root-77" tabindex="0" type="button" style="box-sizing: inherit; color: rgba(0, 0, 0, 0.54); border: 0px; cursor: pointer; margin: 0px; display: inline-flex; outline: 0px; padding: 12px; position: relative; align-items: center; user-select: none; border-radius: 50%; vertical-align: middle; justify-content: center; text-decoration: none; background-color: transparent; appearance: none; -webkit-tap-highlight-color: transparent; flex: 0 0 auto; overflow: visible; font-size: 1.5rem; text-align: center; transition: background-color 150ms cubic-bezier(0.4, 0, 0.2, 1) 0ms;"><span class="MuiIconButton-label-84" style="box-sizing: inherit; width: 23.9915px; display: flex; align-items: inherit; justify-content: inherit;"><svg class="MuiSvgIcon-root-86" focusable="false" viewBox="0 0 24 24" aria-hidden="true"><path d="M1 21h22L12 2 1 21zm12-3h-2v-2h2v2zm0-4h-2v-4h2v4z"></path></svg></span><span class="MuiTouchRipple-root-173" style="box-sizing: inherit; inset: 0px; z-index: 0; overflow: hidden; position: absolute; border-radius: inherit; pointer-events: none;"></span></button><span class="MuiBadge-badge-107 jss2923 jss2925 MuiBadge-anchorOriginTopRightRectangle-112" style="box-sizing: border-box; height: 20px; display: flex; padding: 0px 6px; z-index: 1; position: absolute; flex-flow: row wrap; font-size: 0.75rem; min-width: 20px; transition: transform 225ms cubic-bezier(0.4, 0, 0.2, 1) 0ms; align-items: center; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 500; line-height: 1; place-content: center; border-radius: 10px; top: 10px; right: 10px; transform: scale(1) translate(50%, -50%); transform-origin: 100% 0%; background-color: rgb(255, 193, 7);">99+</span></span><div class="jss2912" style="box-sizing: inherit; flex-grow: 1;"></div><div style="box-sizing: inherit;"><button class="MuiButtonBase-root-85 MuiButton-root-128 MuiButton-contained-136 MuiButton-containedPrimary-137" tabindex="0" type="button" style="box-sizing: border-box; color: rgb(255, 255, 255); border: 0px; cursor: pointer; margin: 0px; display: inline-flex; outline: 0px; padding: 6px 16px; position: relative; align-items: center; user-select: none; border-radius: 4px; vertical-align: middle; justify-content: center; text-decoration: none; background-color: rgb(51, 153, 204); appearance: none; -webkit-tap-highlight-color: transparent; font-size: 0.875rem; min-width: 64px; transition: background-color 250ms cubic-bezier(0.4, 0, 0.2, 1) 0ms, box-shadow 250ms cubic-bezier(0.4, 0, 0.2, 1) 0ms, border 250ms cubic-bezier(0.4, 0, 0.2, 1) 0ms; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 500; line-height: 1.75; letter-spacing: 0.02857em; text-transform: uppercase; box-shadow: rgba(0, 0, 0, 0.2) 0px 3px 1px -2px, rgba(0, 0, 0, 0.14) 0px 2px 2px 0px, rgba(0, 0, 0, 0.12) 0px 1px 5px 0px;"><span class="MuiButton-label-129" style="box-sizing: inherit; width: 188.736px; display: inherit; align-items: inherit; justify-content: inherit;"><span class="MuiButton-startIcon-150 MuiButton-iconSizeMedium-153" style="box-sizing: inherit; display: inherit; margin-left: -4px; margin-right: 8px;"><svg class="MuiSvgIcon-root-86" focusable="false" viewBox="0 0 24 24" aria-hidden="true"><path d="M19 12v7H5v-7H3v7c0 1.1.9 2 2 2h14c1.1 0 2-.9 2-2v-7h-2zm-6 .67l2.59-2.58L17 11.5l-5 5-5-5 1.41-1.41L11 12.67V3h2z"></path></svg></span>LOG HERUNTERLADEN</span><span class="MuiTouchRipple-root-173" style="box-sizing: inherit; inset: 0px; z-index: 0; overflow: hidden; position: absolute; border-radius: inherit; pointer-events: none;"></span></ [... cut ...]
modbus.0 | 2021-06-16 08:18:57.844 | warn | The current range for reading the values was from adress 30513 up to adress 30545!
modbus.0 | 2021-06-16 08:18:57.844 | warn | The used scaleFactor from address 600 is not inside the same read block as the parameter on address 30541
modbus.0 | 2021-06-16 08:18:57.844 | warn | Please make sure to configure the read process that both adresses are read in the same block!
modbus.0 | 2021-06-16 08:18:57.843 | warn | The current range for reading the values was from adress 30513 up to adress 30545!

</div></div>

GermanBluefox, 2021-06-16:
> Verbindung generell klappt, bekomme jetzt aber zyklische Fehler:
> 
> Hast du eine Idee? Konfig nicht verändert.
> 
> 499+
> LOG HERUNTERLADEN
> Log-Größe: 764.8 KB
> 
> Quelle	Zeit	debug	Nachricht
> modbus.0	2021-06-16 08:18:57.844	warn	The used scaleFactor from address 600 is not inside the same read block as the parameter on address 30543
> modbus.0	2021-06-16 08:18:57.844	warn	Please make sure to configure the read process that both adresses are read in the same block!
> modbus.0	2021-06-16 08:18:57.844	warn	The current range for reading the values was from adress 30513 up to adress 30545!
> modbus.0	2021-06-16 08:18:57.844	warn	The used scaleFactor from address 600 is not inside the same read block as the parameter on address 30541
> modbus.0	2021-06-16 08:18:57.844	warn	Please make sure to configure the read process that both adresses are read in the same block!
> modbus.0	2021-06-16 08:18:57.843	warn	The current range for reading the values was from adress 30513 up to adress 30545!
> **Edit: Konnte ich selbst fixen - Die Formel war vorher x/3600 - Ist jetzt x/(3600).**

Fixed in 3.4.4

closed 2021-06-17 by GermanBluefox