ioBroker/ioBroker.modbus #134: New interface

adavt, 2021-06-15:
There are comments on the version 3.4.2 interface.
1. When you select the "Slave ID" sort, the addresses are sorted in descending order. Uncomfortable.
2. The meaning of the "Edit Mode" slider is not clear. His condition is not remembered. Switches to editing simply by accessing any register field. If there wasn't a slider, nothing would have changed.
3. The extended mode is little functional. If there were no switching to this mode, it would not have gotten worse. An extra entity that, however, does not interfere.

GermanBluefox, 2021-06-16:
1. fixed
2. On some slow browsers the edit mode is too slow. And with edit mode off the user can at least see the settings
3. There are too many expert fields: formula, scale factor, offset, multiplier, write pulse. In non "extended" mode all these fields are hidden and table is not overloaded with controls

3.4.3

adavt, 2021-06-16:
In the 3.4.4 version, the comments are corrected - checked. Thank you! As a sentence - is it not better to leave only "ID" in the name of the field "Slave ID" (in russian "ID ведомого")? That's enough. And narrow the width of the field, which will make it possible to make wider other fields more meaningful.

GermanBluefox, 2021-06-17:
No. ID could be everything. But Slave ID clearly

closed 2021-06-17 by GermanBluefox