ioBroker/ioBroker.modbus #135: Write value no more possible with 3.4.2 - unknown function code: 134

5G7K, 2021-06-16:
**Describe the bug**  
With versions 3.2.6 it was possible to write values.
Now with 3.4.2 following error message occurs:

`modbus.0	2021-06-16 10:39:21.379	debug	(26828) Cleaning up request fifo.
modbus.0	2021-06-16 10:39:21.379	debug	(26828) Clearing timeout of the current request.
modbus.0	2021-06-16 10:39:21.371	info	(26828) Disconnected from slave
modbus.0	2021-06-16 10:39:21.190	debug	(26828) Poll holdingRegs DevID(2) address 1 - 20 bytes
modbus.0	2021-06-16 10:39:21.190	debug	(26828) Poll device 2
modbus.0	2021-06-16 10:39:20.371	error	(26828) Cannot write [265]: {"err":"timeout","timeout":5000}
modbus.0	2021-06-16 10:39:20.370	error	(26828) Request timed out.
modbus.0	2021-06-16 10:39:20.369	warn	(26828) Error: undefined
modbus.0	2021-06-16 10:39:15.392	error	(26828) unknown function code: 134
modbus.0	2021-06-16 10:39:15.362	debug	(26828) state Changed ack=false: modbus.0.holdingRegisters.40266_Dauer_Spitzendrehzahl_Küche: {"val":60,"ack":false,"ts":1623832755359,"q":0,"from":"system.adapter.linkeddevices.0","user":"system.use
modbus.0	2021-06-16 10:39:10.958	debug	(26828) Poll holdingRegs DevID(2) address 384 - 9 bytes
modbus.0	2021-06-16 10:39:10.881	debug	(26828) Poll holdingRegs DevID(2) address 336 - 16 bytes
modbus.0	2021-06-16 10:39:10.796	debug	(26828) Poll holdingRegs DevID(2) address 256 - 30 bytes
modbus.0	2021-06-16 10:39:10.717	debug	(26828) Poll holdingRegs DevID(2) address 1 - 20 bytes
modbus.0	2021-06-16 10:39:10.716	debug	(26828) Poll device 2
modbus.0	2021-06-16 10:39:00.488	debug	(26828) Poll holdingRegs DevID(2) address 384 - 9 bytes
modbus.0	2021-06-16 10:39:00.411	debug	(26828) Poll holdingRegs DevID(2) address 336 - 16 bytes
modbus.0	2021-06-16 10:39:00.329	debug	(26828) Poll holdingRegs DevID(2) address 256 - 30 bytes
modbus.0	2021-06-16 10:39:00.250	debug	(26828) Poll holdingRegs DevID(2) address 1 - 20 bytes
modbus.0	2021-06-16 10:39:00.249	debug	(26828) Poll device 2`

Apollon77, 2021-06-16:
There are new settings to always use "Do not use "Write multiple registers\"" command or such to query ... maybe check if this helps to change that?

5G7K, 2021-06-16:
Both options are unchecked. I tried both ones - always with the same result
I downgraded to 3.3.1 and everything works fine with the same settings
Log of 3.3.1

modbus.0 | 2021-06-16 14:12:17.216 | debug | (15863) Write successfully [265]: 72
-- | -- | -- | --
modbus.0 | 2021-06-16 14:12:17.184 | debug | (15863) state Changed ack=false: modbus.0.holdingRegisters.40266_Dauer_Spitzendrehzahl_Küche: {"val":360,"ack":false,"ts":1623845537180,"q":0,"from":"system.adapter.linkeddevices.0","user":"system.use
modbus.0 | 2021-06-16 14:12:09.036 | debug | (15863) Poll holdingRegs DevID(2) address 384 - 9 bytes
modbus.0 | 2021-06-16 14:12:08.957 | debug | (15863) Poll holdingRegs DevID(2) address 336 - 16 bytes
modbus.0 | 2021-06-16 14:12:08.876 | debug | (15863) Poll holdingRegs DevID(2) address 256 - 30 bytes
modbus.0 | 2021-06-16 14:12:08.795 | debug | (15863) Poll holdingRegs DevID(2) address 1 - 20 bytes
modbus.0 | 2021-06-16 14:12:08.743 | debug | (15863) Poll device 2

[image]

GermanBluefox, 2021-06-16:
I added in 3.4.3 some error codes. May be they will help to identify the problem

HJS72, 2021-06-16:
i've tested 3.4.4 with Option "Write multiple registers".
Reading works fine, but when I try to write a Register I get following error:

(24322) Cannot write [550]: {"errorCode":134,"exceptionCode":1,"message":"ILLEGAL FUNCTION"}

5G7K, 2021-06-17:
my test with 3.4.4.

2021-06-17 08:32:03.055 -&nbsp;debug: modbus.0 (13806) Redis Objects: Use Redis connection: 0.0.0.0:9001
2021-06-17 08:32:03.106 -&nbsp;debug: modbus.0 (13806) Objects client ready ... initialize now
2021-06-17 08:32:03.109 -&nbsp;debug: modbus.0 (13806) Objects create System PubSub Client
2021-06-17 08:32:03.110 -&nbsp;debug: modbus.0 (13806) Objects create User PubSub Client
2021-06-17 08:32:03.112 -&nbsp;debug: modbus.0 (13806) Objects client initialize lua scripts
2021-06-17 08:32:03.127 -&nbsp;debug: modbus.0 (13806) Objects connected to redis: 0.0.0.0:9001
2021-06-17 08:32:03.137 -&nbsp;debug: modbus.0 (13806) objectDB connected
2021-06-17 08:32:03.139 -&nbsp;debug: modbus.0 (13806) Redis States: Use Redis connection: 127.0.0.1:6379
2021-06-17 08:32:03.147 -&nbsp;debug: modbus.0 (13806) States create System PubSub Client
2021-06-17 08:32:03.149 -&nbsp;debug: modbus.0 (13806) States create User PubSub Client
2021-06-17 08:32:03.158 -&nbsp;debug: modbus.0 (13806) States connected to redis: 127.0.0.1:6379
2021-06-17 08:32:03.159 -&nbsp;debug: modbus.0 (13806) statesDB connected
2021-06-17 08:32:03.554 -&nbsp;debug: modbus.0 (13806) Plugin sentry Initialize Plugin (enabled=true)
2021-06-17 08:32:03.812 -&nbsp;info: modbus.0 (13806) starting. Version 3.4.4 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.22.1, js-controller: 3.2.16

**write a value with both options unchecked:**

2021-06-17 08:32:16.050 -&nbsp;debug: modbus.0 (13806) Poll device 2
2021-06-17 08:32:16.052 -&nbsp;debug: modbus.0 (13806) Poll holdingRegs DevID(2) address 1 - 20 bytes
2021-06-17 08:32:16.160 -&nbsp;debug: modbus.0 (13806) Poll holdingRegs DevID(2) address 256 - 30 bytes
2021-06-17 08:32:16.267 -&nbsp;debug: modbus.0 (13806) Poll holdingRegs DevID(2) address 336 - 16 bytes
2021-06-17 08:32:16.375 -&nbsp;debug: modbus.0 (13806) Poll holdingRegs DevID(2) address 384 - 9 bytes
2021-06-17 08:32:22.343 -&nbsp;debug: modbus.0 (13806) state Changed ack=false: modbus.0.holdingRegisters.40267_Dauer_Boost: {"val":600,"ack":false,"ts":1623911542340,"q":0,"from":"system.adapter.linkeddevices.0","user":"system.user.admin","lc":1623911542340}
2021-06-17 08:32:22.371 -&nbsp;error: modbus.0 (13806) Error response for FCx6: Illegal Function
2021-06-17 08:32:22.372 -&nbsp;error: modbus.0 (13806) Error response: Function code received in the query is not recognized or allowed by slave
2021-06-17 08:32:22.849 -&nbsp;warn: modbus.0 (13806) Error: undefined
2021-06-17 08:32:22.849 -&nbsp;error: modbus.0 (13806) Request timed out.
2021-06-17 08:32:22.850 -&nbsp;error: modbus.0 (13806) Cannot write [266]: {"err":"timeout","timeout":500}
2021-06-17 08:32:23.851 -&nbsp;info: modbus.0 (13806) Disconnected from slave
2021-06-17 08:32:23.858 -&nbsp;debug: modbus.0 (13806) Clearing timeout of the current request.
2021-06-17 08:32:23.859 -&nbsp;debug: modbus.0 (13806) Cleaning up request fifo.

**with option: without multiple register**

2021-06-17 08:36:15.452 -&nbsp;debug: modbus.0 (19840) Poll device 2
2021-06-17 08:36:15.454 -&nbsp;debug: modbus.0 (19840) Poll holdingRegs DevID(2) address 1 - 20 bytes
2021-06-17 08:36:15.560 -&nbsp;debug: modbus.0 (19840) Poll holdingRegs DevID(2) address 256 - 30 bytes
2021-06-17 08:36:15.667 -&nbsp;debug: modbus.0 (19840) Poll holdingRegs DevID(2) address 336 - 16 bytes
2021-06-17 08:36:15.773 -&nbsp;debug: modbus.0 (19840) Poll holdingRegs DevID(2) address 384 - 9 bytes
2021-06-17 08:36:25.346 -&nbsp;debug: modbus.0 (19840) state Changed ack=false: modbus.0.holdingRegisters.40267_Dauer_Boost: {"val":600,"ack":false,"ts":1623911785344,"q":0,"from":"system.adapter.linkeddevices.0","user":"system.user.admin","lc":1623911785344}
2021-06-17 08:36:25.375 -&nbsp;error: modbus.0 (19840) Error response for FCx6: Illegal Function
2021-06-17 08:36:25.375 -&nbsp;error: modbus.0 (19840) Error response: Function code received in the query is not recognized or allowed by slave
2021-06-17 08:36:25.851 -&nbsp;warn: modbus.0 (19840) Error: undefined
2021-06-17 08:36:25.852 -&nbsp;error: modbus.0 (19840) Request timed out.
2021-06-17 08:36:25.853 -&nbsp;error: modbus.0 (19840) Cannot write [266]: {"err":"timeout","timeout":500}
2021-06-17 08:36:26.005 -&nbsp;debug: modbus.0 (19840) Poll device 2
2021-06-17 08:36:26.006 -&nbsp;debug: modbus.0 (19840) Poll holdingRegs DevID(2) address 1 - 20 bytes
2021-06-17 08:36:26.708 -&nbsp;warn: modbus.0 (19840) Poll error count: 1 code: "App Timeout"
2021-06-17 08:36:26.854 -&nbsp;info: modbus.0 (19840) Disconnected from slave
2021-06-17 08:36:26.865 -&nbsp;debug: modbus.0 (19840) Clearing timeout of the current request.
2021-06-17 08:36:26.866 -&nbsp;debug: modbus.0 (19840) Cleaning up request fifo.

**with option: write only multiple register**

2021-06-17 08:37:43.999 -&nbsp;debug: modbus.0 (22070) Poll device 2
2021-06-17 08:37:44.000 -&nbsp;debug: modbus.0 (22070) Poll holdingRegs DevID(2) address 1 - 20 bytes
2021-06-17 08:37:44.109 -&nbsp;debug: modbus.0 (22070) Poll holdingRegs DevID(2) address 256 - 30 bytes
2021-06-17 08:37:44.217 -&nbsp;debug: modbus.0 (22070) Poll holdingRegs DevID(2) address 336 - 16 bytes
2021-06-17 08:37:44.322 -&nbsp;debug: modbus.0 (22070) Poll holdingRegs DevID(2) address 384 - 9 bytes
2021-06-17 08:37:49.982 -&nbsp;debug: modbus.0 (22070) state Changed ack=false: modbus.0.holdingRegisters.40267_Dauer_Boost: {"val":600,"ack":false,"ts":1623911869978,"q":0,"from":"system.adapter.linkeddevices.0","user":"system.user.admin","lc": [... cut ...]
2021-06-17 08:37:50.491 -&nbsp;warn: modbus.0 (22070) Error: undefined
2021-06-17 08:37:50.492 -&nbsp;error: modbus.0 (22070) Request timed out.
2021-06-17 08:37:50.493 -&nbsp;error: modbus.0 (22070) Cannot write [266]: {"err":"timeout","timeout":500}
2021-06-17 08:37:51.493 -&nbsp;info: modbus.0 (22070) Disconnected from slave
2021-06-17 08:37:51.501 -&nbsp;debug: modbus.0 (22070) Clearing timeout of the current request.
2021-06-17 08:37:51.502 -&nbsp;debug: modbus.0 (22070) Cleaning up request fifo.

GermanBluefox, 2021-06-17:
It will be always written with FC6.... no matter what options are set... Strange.

GermanBluefox, 2021-06-19:
Please test 3.4.5

HJS72, 2021-06-19:
Test with Maico WS75 Powerbox (Write multiple registers only) was successful. Thanks for the very fast support.

5G7K, 2021-06-19:
Test successful. Thank you very much!

closed 2021-06-19 by 5G7K