ioBroker/ioBroker.modbus #136: error | Invalid holdingRegisters address nach update auf 3.4.5

Schimi1983, 2021-06-19:
Nach Update von 3.4.4 auf 3.4.5
bekomme ich folgenden Fehler....

Zurück auf 3.4.4 und alles funktioniert wie es soll

`

modbus.4 | 2021-06-19 19:28:01.761 | error | Invalid holdingRegisters address: -40001
-- | -- | -- | --
modbus.4 | 2021-06-19 19:28:01.761 | error | Invalid holdingRegisters address: -40000
modbus.4 | 2021-06-19 19:28:01.761 | error | Invalid holdingRegisters address: -39999
modbus.4 | 2021-06-19 19:28:01.761 | error | Invalid holdingRegisters address: -39998
modbus.4 | 2021-06-19 19:28:01.761 | error | Invalid holdingRegisters address: -39997
modbus.4 | 2021-06-19 19:28:01.761 | error | Invalid holdingRegisters address: -39996
modbus.4 | 2021-06-19 19:28:01.761 | error | Invalid holdingRegisters address: -39995
modbus.4 | 2021-06-19 19:28:01.761 | error | Invalid holdingRegisters address: -39994
modbus.4 | 2021-06-19 19:28:01.760 | error | Invalid holdingRegisters address: -39993
modbus.4 | 2021-06-19 19:28:01.760 | error | Invalid holdingRegisters address: -39978
modbus.4 | 2021-06-19 19:28:01.760 | error | Invalid holdingRegisters address: -39977
modbus.4 | 2021-06-19 19:28:01.760 | error | Invalid holdingRegisters address: -39976
modbus.4 | 2021-06-19 19:28:01.760 | error | Invalid holdingRegisters address: -39975
modbus.4 | 2021-06-19 19:28:01.760 | error | Invalid holdingRegisters address: -39974
modbus.4 | 2021-06-19 19:28:01.760 | error | Invalid holdingRegisters address: -39973
modbus.4 | 2021-06-19 19:28:01.760 | error | Invalid holdingRegisters address: -39972
modbus.4 | 2021-06-19 19:28:01.759 | error | Invalid holdingRegisters address: -39970
modbus.4 | 2021-06-19 19:28:01.759 | error | Invalid holdingRegisters address: -39969
modbus.4 | 2021-06-19 19:28:01.759 | error | Invalid holdingRegisters address: -39968
modbus.4 | 2021-06-19 19:28:01.759 | error | Invalid holdingRegisters address: -39967
modbus.4 | 2021-06-19 19:28:01.759 | error | Invalid holdingRegisters address: -39966
modbus.4 | 2021-06-19 19:28:01.759 | error | Invalid holdingRegisters address: -39965
modbus.4 | 2021-06-19 19:28:01.759 | error | Invalid holdingRegisters address: -39964
modbus.4 | 2021-06-19 19:28:01.758 | error | Invalid holdingRegisters address: -39963
modbus.4 | 2021-06-19 19:28:01.758 | error | Invalid holdingRegisters address: -39962
modbus.4 | 2021-06-19 19:28:01.758 | error | Invalid holdingRegisters address: -39961
modbus.4 | 2021-06-19 19:28:01.758 | error | Invalid holdingRegisters address: -39960
modbus.4 | 2021-06-19 19:28:01.758 | error | Invalid holdingRegisters address: -39959
modbus.4 | 2021-06-19 19:28:01.758 | error | Invalid holdingRegisters address: -39958
modbus.4 | 2021-06-19 19:28:01.758 | error | Invalid holdingRegisters address: -39957
modbus.4 | 2021-06-19 19:28:01.757 | error | Invalid holdingRegisters address: -39992
modbus.4 | 2021-06-19 19:28:01.757 | error | Invalid holdingRegisters address: -39991
modbus.4 | 2021-06-19 19:28:01.757 | error | Invalid holdingRegisters address: -39990
modbus.4 | 2021-06-19 19:28:01.757 | error | Invalid holdingRegisters address: -39989
modbus.4 | 2021-06-19 19:28:01.757 | error | Invalid holdingRegisters address: -39988
modbus.4 | 2021-06-19 19:28:01.757 | error | Invalid holdingRegisters address: -39987
modbus.4 | 2021-06-19 19:28:01.756 | error | Invalid holdingRegisters address: -39986
modbus.4 | 2021-06-19 19:28:01.756 | error | Invalid holdingRegisters address: -39984
modbus.4 | 2021-06-19 19:28:01.756 | error | Invalid holdingRegisters address: -39982
modbus.4 | 2021-06-19 19:28:01.756 | error | Invalid holdingRegisters address: -39981
modbus.4 | 2021-06-19 19:28:01.756 | error | Invalid holdingRegisters address: -39985
modbus.4 | 2021-06-19 19:28:01.756 | error | Invalid holdingRegisters address: -39980
modbus.4 | 2021-06-19 19:28:01.756 | error | Invalid holdingRegisters address: -39983
modbus.4 | 2021-06-19 19:28:01.755 | error | Invalid holdingRegisters address: -39979
modbus.4 | 2021-06-19 19:28:01.755 | error | Invalid holdingRegisters address: -39971
modbus.4 | 2021-06-19 19:28:01.755 | error | Invalid holdingRegisters address: -39956
modbus.4 | 2021-06-19 19:28:01.755 | error | Invalid holdingRegisters address: -39955
modbus.4 | 2021-06-19 19:28:01.755 | error | Invalid holdingRegisters address: -39954
modbus.4 | 2021-06-19 19:28:01.754 | error | Invalid holdingRegisters address: -39953

`

GermanBluefox, 2021-06-21:
Kann dein Fehler nicht reproduzieren.
Kannst du dein Objekt `system.adapter.modbus.4` hierher exportieren?

Schimi1983, 2021-06-21:
Ich hoffe es ist das was du meinst :-)

`{
  "modbus.4": {
    "_id": "modbus.4",
    "type": "meta",
    "common": {
      "type": "meta"
    },
    "native": {},
    "from": "system.adapter.modbus.4",
    "user": "system.user.admin",
    "ts": 1624293408831,
    "acl": {
      "object": 1636,
      "owner": "system.user.admin",
      "ownerGroup": "system.group.administrator"
    }
  },
  "modbus.4.holdingRegisters": {
    "type": "channel",
    "common": {
      "name": "Holding registers"
    },
    "native": {},
    "from": "system.adapter.modbus.4",
    "user": "system.user.admin",
    "ts": 1624293409655,
    "_id": "modbus.4.holdingRegisters",
    "acl": {
      "object": 1636,
      "owner": "system.user.admin",
      "ownerGroup": "system.group.administrator"
    }
  },
  "modbus.4.holdingRegisters.40001_Type_R": {
    "_id": "modbus.4.holdingRegisters.40001_Type_R",
    "type": "state",
    "common": {
      "name": "1 = Basis 2 = Bypass 3 = Comfort",
      "role": "level",
      "type": "number",
      "read": true,
      "write": true,
      "def": 0,
      "unit": ""
    },
    "native": {
      "regType": "holdingRegs",
      "address": 0,
      "deviceId": 1,
      "type": "uint16be",
      "len": 1,
      "offset": 0,
      "factor": 1,
      "poll": true
    },
    "from": "system.adapter.modbus.4",
    "user": "system.user.admin",
    "ts": 1624293409550,
    "acl": {
      "object": 1636,
      "state": 1636,
      "owner": "system.user.admin",
      "ownerGroup": "system.group.administrator"
    }
  },
  "modbus.4.holdingRegisters.40002_On/Off_R/W": {
    "_id": "modbus.4.holdingRegisters.40002_On/Off_R/W",
    "type": "state",
[... 1570 lines cut ...]
  },
  "modbus.4.info.pollTime": {
    "type": "state",
    "common": {
      "name": "Poll time",
      "type": "number",
      "role": "",
      "write": false,
      "read": true,
      "def": 0,
      "unit": "ms"
    },
    "native": {},
    "from": "system.adapter.modbus.4",
    "user": "system.user.admin",
    "ts": 1624293409548,
    "_id": "modbus.4.info.pollTime",
    "acl": {
      "object": 1636,
      "state": 1636,
      "owner": "system.user.admin",
      "ownerGroup": "system.group.administrator"
    }
  }
}`

Schimi1983, 2021-06-21:
habe es gerade nochmal getestet (wollte dir Infos mit 3.4.5 schicken... doch sobald ich die Instanz starte, wird der Objektbaum leer und im Protokoll die Fehler "oben"

Schimi1983, 2021-06-21:
*update*

Lege ich eine neue Instanz in 3.4.5 an und geb die register neu ein... scheint alles zu klappen....

muss aber kurz weg und würde in ca. einer Std nochmal nen update geben wenn ich das alles neu gemacht habe....

die anderen modbus instanzen laufen problemlos...

GermanBluefox, 2021-06-21:
> Ich hoffe es ist das was du meinst :-)

Du hast modbus.4.* exported und ich habe gebeten ein Objekt `system.adapter.modbus.4`

Schimi1983, 2021-06-21:
Hoffe es ist jetzt richtig :-)

> {
  "system.adapter.modbus.4": {
    "_id": "system.adapter.modbus.4",
    "type": "instance",
    "common": {
      "name": "modbus",
      "version": "3.4.5",
      "news": {
        "3.3.1": {
          "en": "fixed the configuration dialog for \"input registers\" in slave mode",
          "de": "Der Konfigurationsdialog für \"Eingangsregister\" im Slave-Modus wurde korrigiert",
          "ru": "исправлен диалог конфигурации для \"входных регистров\" в ведомом режиме",
          "pt": "corrigiu a caixa de diálogo de configuração para \"registros de entrada\" no modo escravo",
          "nl": "de configuratiedialoog voor \"invoerregisters\" in slaafmodus gerepareerd",
          "fr": "correction de la boîte de dialogue de configuration des \"registres d'entrée\" en mode esclave",
          "it": "corretta la finestra di dialogo di configurazione per \"registri di ingresso\" in modalità slave",
          "es": "Se corrigió el diálogo de configuración para \"registros de entrada\" en modo esclavo.",
          "pl": "naprawiono okno dialogowe konfiguracji dla \"rejestrów wejściowych\" w trybie slave",
          "zh-cn": "修复了从模式下“输入寄存器”的配置对话框"
        },
        "3.3.0": {
          "en": "Allow to use write-only (no poll) states\n(Apollon77/TmShaz) F Write multiple registers\ncreate states of type string with default value of type string",
          "de": "Erlauben Sie die Verwendung von Nur-Schreib-Zuständen (keine Abfrage)\n(Apollon77 / TmShaz) F Schreiben Sie mehrere Register\nErstellen Sie Zustände vom Typ Zeichenfolge mit dem Standardwert vom Typ Zeichenfolge",
          "ru": "Разрешить использовать состояния только для записи (без опроса)\n(Apollon77 / TmShaz) F Запись нескольких регистров\nсоздавать состояния строки типа со значением по умолчанию строки типа",
          "pt": "Permitir o uso de estados somente gravação (sem pesquisa)\n(Apollon77 / TmShaz) F Grava registros múltiplos\ncria estados do tipo string com o valor padrão do tipo string",
          "nl": "Sta toe om alleen-schrijven (geen poll) staten te gebruiken\n(Apollon77 / TmShaz) F Schrijf meerdere registers\nmaak staten van het type string met de standaardwaarde van het type string",
          "fr": "Autoriser à utiliser des états en écriture seule (pas d'interrogation)\n(Apollon77 / TmShaz) F Ecriture de plusieurs registres\ncréer des états de type string avec la valeur par défaut de type string",
          "it": "Consenti l'utilizzo degli stati di sola scrittura (nessun sondaggio)\n(Apollon77 / TmShaz) F Scrive più registri\ncrea stati di tipo stringa con il valore predefinito di tipo stringa",
          "es": "Permitir usar estados de solo escritura (sin encuesta)\n(Apollon77 / TmShaz) F Escribir varios registros\ncrear estados de tipo cadena con el valor predeterminado de tipo cadena",
          "pl": "Zezwalaj na używanie stanów tylko do zapisu (bez ankiety)\n(Apollon77 / TmShaz) F Zapisz wiele rejestrów\nstwórz stany typu string z domyślną wartością typu string",
          "zh-cn": "允许使用只写（无轮询）状态\n（Apollon77 / TmShaz）F写入多个寄存器\n使用默认类型的字符串创建字符串类型的状态"
        },
        "3.2.6": {
          "en": "Prevent a crash case (Sentry IOBROKER-MODBUS-20)\nBetter handle invalid responses",
          "de": "Verhindern Sie einen Absturz (Sentry IOBROKER-MODBUS-20)\nBehandeln Sie ungültige Antworten besser",
          "ru": "Предотвращение сбоя (Sentry IOBROKER-MODBUS-20)\nЛучше обрабатывать недействительные ответы",
          "pt": "Previna um caso de travamento (Sentry IOBROKER-MODBUS-20)\nMelhor lidar com respostas inválidas",
          "nl": "Voorkom een crashcase (Sentry IOBROKER-MODBUS-20)\nBeter omgaan met ongeldige reacties",
          "fr": "Prévenir un cas de crash (Sentry IOBROKER-MODBUS-20)\nMieux gérer les réponses invalides",
          "it": "Prevenire un crash case (Sentry IOBROKER-MODBUS-20)\nGestire meglio le risposte non valide",
          "es": "Evitar un caso de accidente (Sentry IOBROKER-MODBUS-20)\nManeje mejor las respuestas no válidas",
          "pl": "Zapobiegaj awariom (Sentry IOBROKER-MODBUS-20)\nLepiej radzić sobie z nieprawidłowymi odpowiedziami",
          "zh-cn": "防止发生事故（Sentry IOBROKER-MODBUS-20）\n更好地处理无效回复"
        },
        "3.2.4": {
          "en": "also support ttyXRUSB0 style devices",
          "de": "unterstützt auch Geräte im ttyXRUSB0-Stil",
          "ru": "также поддерживает устройства в стиле ttyXRUSB0",
          "pt": "também suporta dispositivos estilo ttyXRUSB0",
          "nl": "ondersteunt ook ttyXRUSB0-stijl apparaten",
          "fr": "prend également en charge les périphériques de style ttyXRUSB0",
          "it": "supporta anche dispositivi in stile ttyXRUSB0",
          "es": "también es compatible con dispositivos de estilo ttyXRUSB0",
          "pl": "obsługuje również urządzenia w stylu ttyXRUSB0",
          "zh-cn": "还支持ttyXRUSB0样式的设备"
        },
        "3.2.3": {
          "en": "Catch value encoding error and do not crash adapter (Sentry IOBROKER-MODBUS-1W)\nadd a meta object as instance object",
          "de": "Catch-Wert-Codierungsfehler und Adapter nicht abstürzen (Sentry IOBROKER-MODBUS-1W)\nFügen Sie ein Metaobjekt als Instanzobjekt hinzu",
          "ru": "Выявить ошибку кодирования значения и не вызывать сбой адаптера (Sentry IOBROKER-MODBUS-1W)\nдобавить мета-объект как объект-экземпляр",
[... cut ...]
    "_id": "system.adapter.modbus.4.uptime",
    "type": "state",
    "common": {
      "name": "modbus.4 uptime",
      "type": "number",
      "read": true,
      "write": false,
      "role": "indicator.state",
      "unit": "seconds"
    },
    "native": {},
    "from": "system.adapter.modbus.4",
    "ts": 1624294905267,
    "acl": {
      "object": 1636,
      "state": 1636,
      "owner": "system.user.admin",
      "ownerGroup": "system.group.administrator"
    },
    "user": "system.user.admin"
  }
}

Schimi1983, 2021-06-21:
so, mein versprochenes Update....

habe komplett alles neu eingetragen, jetzt geht es auch mit der 3.4.5.... vielleicht war durch das "experementieren" damals was durcheinander gekommen?

GermanBluefox, 2021-06-22:
Ich kann das Problem nicht reporoduzieren

Schimi1983, 2021-06-23:
hat sich nach einem erneuten eingeben der Register auch erledigt... kann es nun auch nicht mehr reproduzieren....

War vielleicht ein überbleibsel aus der "Probierphase" welches sich "quer gestellt" hat nach dem update auf 3.4.5

closed 2021-06-24 by GermanBluefox

Nibbels, 2022-09-01:
Ich hatte den Fehler eben auch.
Version 5.0.5 und ich sah keine Objekte im Baum.
[image]
Dann wieder auf Instanz gegangen, etwas an den Optionen rumgeklickt. (Ich weiß nicht ob das eine Wirkung hatte.)
Anschließend habe ich bei 
[image]
tatsächlich eine negative Zahl gesehen.

Dann Try-and-Error:
[image]
und die Adresse neu eingegeben. Anschließend hats wohl gepasst.

Ich vermute, dass irgendeine Settings-Initialisierungs-Problematik vorlag?

Nibbels, 2022-09-01:
Oder lags an 
[image]
On oder Off?