ioBroker/ioBroker.modbus #137: v3.4.6

adavt, 2021-06-21:
Glad they fixed the old alias error. But fixed not everywhere. With coils recalculation, the error has not been fixed. Register address correction must be "-1" without using alias. With the rest of the types it became normal and "-1" was added to the recount to what was.

GermanBluefox, 2021-06-22:
Is it about GUI or about backend?

adavt, 2021-06-22:
GUI.  When alias (and vice versa minus) is marked, a number is added to the register address. Discrete registers + 10001, input + 30001, holding + 40001 - became correct. Prior to version 3.4.6, there were 10000, 30000 and 40000. But coils does not change, although it should be + 1 with alias.

GermanBluefox, 2021-06-22:
Fixed in 3.4.7

closed 2021-06-23 by GermanBluefox