ioBroker/ioBroker.modbus #138: Reading Holding registers without alias

jmades, 2021-06-22:
As a user I want to read a holding register 0x4500 (16Bit) of a "native" modbus device. I configured the iobroker.modbus adapter **not to use aliases** and added a holding register in the following way.
[image]

The adapter generates:
[image]

Expected output/request:
01 03 45 00 00 02 Crc1 Crc2

GermanBluefox, 2021-06-22:
Which version is that?
I cannot reproduce the problem with disabled aliases 
[image]

jmades, 2021-06-22:
Yes, I installed 3.4.6.
The only difference I recognize is, that I'm using serial communication:
[image]

GermanBluefox, 2021-06-23:
What do you see here?
[image]

jmades, 2021-06-23:
Where do I find this view/file?

jmades, 2021-06-23:
Found it, it shows...
[image]

jmades, 2021-06-24:
I "solved" this issue. The problem belongs to the parity settingsof the serial driver.
Changing the parity from "even" to "none" the device recieves the correct request.

Should I open a new issue for that.

Apollon77, 2021-06-24:
> Should I open a new issue for that?

What should the adapter do in such a case? SO what would be the issue? In fact you need to make sure to provide correct settings for the connected device ...

jmades, 2021-06-24:
You got we wrong, or I was not precise enough...again:
Having both, the adapter and the device configured with parity "even" the device receives a wrong request.
Changing both to parity "none" it works. Can you please check if the adpater configures the serial port correctly in case of parity "even".

Apollon77, 2021-06-24:
Hm ... interesting ... code wise it should be fine. Can you please change config to even and store the settings and then check the "systerm.adapter.modbus.X object JOSN and there in native section what is set there as parity?
YOu can check that in Admin when you enable expert mode and turn off all filters

jmades, 2021-06-24:
It is even...strange...
[image]

Apollon77, 2021-06-24:
Then it should be also used @GermanBluefox  any idea?

jmades, 2021-06-24:
I did some further testing:
Changed parity in the adapter to "even" and "odd", kept the device parity to "none" --> in both cases I recieve a correct request.
Looks like the serial port is not changing parity.
Adapter runs on a raspberry with an RS485 shield.

Is there a way to check/read back the setting of the serial port

GermanBluefox, 2021-06-24:
"even" will be delivered to SerialPort.
[image]

May be it is issue of "serialport" package?

Apollon77, 2021-06-24:
I yesterday released an update for modbux and also updated serialport ... maybe try again the newest version?

jmades, 2021-06-24:
Just tested 3.4.8 same behaviour....sorry.
(I'm offline now until Monday - Thanks for your suppport until now, hoping we find the problem.)

jmades, 2021-06-28:
Back from my trip, I tested the following this morning:
- checked permissions of serial port, but they are ok - set them for test also to 777, without success.
- Changed baudrate instead of parity - this works fine. So there is NO general problem of port settings.

Summary: I still can't change the parity to "even".

GermanBluefox, 2021-06-28:
We just set the parity in Serialport:
https://github.com/ioBroker/ioBroker.modbus/blob/master/lib/jsmodbus/transports/modbus-client-serial.js#L35

You can verify it yourself by modifying the debug output before new SerialPort to:
`this.log.debug(`connect to serial ${serial.portName} with ${serial.baudRate} and with parity ${serial.parity}`);

And enabling the debug level.

If you see "even" there, so it is issue of Serial port https://github.com/serialport/node-serialport/issues
If you not see "even", it is issue of ioBroker, but it could be fixed very fast and easy.

jmades, 2021-06-28:
Can you give me a hint how to run (compile) the adapter with the changes mentioned above.
...or is it sufficient to modify the sources here: 
/opt/iobroker/node_modules/iobroker.modbus/lib/jsmodbus/transports
and iobroker compiles the sources before installation of the adpater?

GermanBluefox, 2021-06-28:
> Can you give me a hint how to run (compile) the adapter with the changes mentioned above.
> ...or is it sufficient to modify the sources here:
> /opt/iobroker/node_modules/iobroker.modbus/lib/jsmodbus/transports
> and iobroker compiles the sources before installation of the adpater?

Just change the code and restart the instance. Nothing must b ecompiled

jmades, 2021-06-28:
Ok, got it work: ...and I see "even" in the log!
...so it is an issue of Serial port (https://github.com/serialport/node-serialport/issues).
I will open an issue.

jmades, 2021-06-30:
The problem could be solved by changing the UART (miniUART does not support parity even).
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=198999

closed 2021-06-30 by jmades