ioBroker/ioBroker.modbus #139: unknown function code: 0x40, 0xf6 and problems to reconnect

5G7K, 2021-06-25:
Modbus works fine for several hours

```
2021-06-25 00:45:20.632  - [34mdebug[39m: modbus.0 (24001) Poll device 2
2021-06-25 00:45:20.633  - [34mdebug[39m: modbus.0 (24001) Poll holdingRegs DevID(2) address 1 - 20 bytes
2021-06-25 00:45:20.738  - [34mdebug[39m: modbus.0 (24001) Poll holdingRegs DevID(2) address 256 - 30 bytes
2021-06-25 00:45:20.845  - [34mdebug[39m: modbus.0 (24001) Poll holdingRegs DevID(2) address 336 - 16 bytes
2021-06-25 00:45:20.954  - [34mdebug[39m: modbus.0 (24001) Poll holdingRegs DevID(2) address 384 - 9 bytes
2021-06-25 00:45:36.181  - [34mdebug[39m: modbus.0 (24001) Poll device 2
2021-06-25 00:45:36.183  - [34mdebug[39m: modbus.0 (24001) Poll holdingRegs DevID(2) address 1 - 20 bytes
2021-06-25 00:45:36.289  - [34mdebug[39m: modbus.0 (24001) Poll holdingRegs DevID(2) address 256 - 30 bytes
2021-06-25 00:45:36.395  - [34mdebug[39m: modbus.0 (24001) Poll holdingRegs DevID(2) address 336 - 16 bytes
2021-06-25 00:45:36.500  - [34mdebug[39m: modbus.0 (24001) Poll holdingRegs DevID(2) address 384 - 9 bytes
...
```

and then following error occurs - I don't know why. 
```
2021-06-25 11:44:29.286  - [31merror[39m: modbus.0 (10376) unknown function code: 0x40, 0xf6
2021-06-25 11:44:29.749  - [33mwarn[39m: modbus.0 (10376) Poll error count: 1 code: "App Timeout"
2021-06-25 11:44:29.758  - [33mwarn[39m: modbus.0 (10376) Error: undefined
2021-06-25 11:44:29.759  - [31merror[39m: modbus.0 (10376) Request timed out.
```

after this, connection is broken and the adapter tries to reconnect? But this fails several times and the modbus adapter cannot be used.

```
modbus.0 | 2021-06-25 11:56:47.906 | error | (10376) Reconnect did not help, restart adapter
-- | -- | -- | --
modbus.0 | 2021-06-25 11:56:47.905 | warn | (10376) Poll error count: 13 code: {"err":"timeout","timeout":500}
modbus.0 | 2021-06-25 11:56:47.904 | error | (10376) Request timed out.
modbus.0 | 2021-06-25 11:56:47.902 | warn | (10376) Error: undefined
modbus.0 | 2021-06-25 11:55:46.393 | warn | (10376) Poll error count: 12 code: {"err":"timeout","timeout":500}
modbus.0 | 2021-06-25 11:55:46.392 | error | (10376) Request timed out.
modbus.0 | 2021-06-25 11:55:46.391 | warn | (10376) Error: undefined
modbus.0 | 2021-06-25 11:54:44.880 | warn | (10376) Poll error count: 11 code: {"err":"timeout","timeout":500}
modbus.0 | 2021-06-25 11:54:44.880 | error | (10376) Request timed out.
modbus.0 | 2021-06-25 11:54:44.879 | warn | (10376) Error: undefined
modbus.0 | 2021-06-25 11:53:43.370 | warn | (10376) Poll error count: 10 code: {"err":"timeout","timeout":500}
modbus.0 | 2021-06-25 11:53:43.369 | error | (10376) Request timed out.
modbus.0 | 2021-06-25 11:53:43.368 | warn | (10376) Error: undefined
```

Only after several minutes the adapter restarts itself and this helps to be reconnected again.

Why does this error occurs? Adapter problem, modbus or Modbus client problem?
How can this handling be approved, so that the connection can be repaired within seconds and not after more than 10 minutes?

[image]

**Versions:**  
 - Adapter version: 3.4.8 (same problem with older versions)
 - JS-Controller version: 3.3.14
 - Node version: v12.22.1
 - Operating system: raspberry

Thank you very much!

5G7K, 2021-06-27:
The error is not always the same:

```
2021-06-27 14:00:00.546 - error: modbus.0 (8414) unknown function code: 0x40, 0x4
2021-06-27 14:00:01.017 - warn: modbus.0 (8414) Error: undefined
2021-06-27 14:00:01.018 - error: modbus.0 (8414) Request timed out.
2021-06-27 14:00:01.019 - warn: modbus.0 (8414) Poll error count: 1 code: {"err":"timeout","timeout":500}
```

```
2021-06-27 21:43:11.142 - error: modbus.0 (24207) unknown function code: 0x40, 0x11
2021-06-27 21:43:11.615 - warn: modbus.0 (24207) Error: undefined
2021-06-27 21:43:11.616 - error: modbus.0 (24207) Request timed out.
2021-06-27 21:43:11.618 - warn: modbus.0 (24207) Poll error count: 1 code: {"err":"timeout","timeout":500}
```

```
2021-06-27 09:44:25.445 - error: modbus.0 (10883) unknown function code: 0x40, 0x42
2021-06-27 09:44:25.810 - warn: modbus.0 (10883) Poll error count: 1 code: "App Timeout"
2021-06-27 09:44:25.920 - warn: modbus.0 (10883) Error: undefined
2021-06-27 09:44:25.920 - error: modbus.0 (10883) Request timed out.
```

Here the error occurs while trying to write a value.
```
2021-06-27 21:00:00.936 - warn: modbus.0 (27257) Error: undefined
2021-06-27 21:00:00.940 - error: modbus.0 (27257) Request timed out.
2021-06-27 21:00:00.942 - error: modbus.0 (27257) Cannot write multiple registers [257]: {"err":"timeout","timeout":500}
2021-06-27 21:00:01.099 - warn: modbus.0 (27257) Poll error count: 1 code: "App Timeout"
```

GermanBluefox, 2021-06-28:
One user has an issue with "even" settings. And I am surprised, that by you it is working OK.
Can you change the parity to "none" and try again?
Because the error looks like the data stream is corrupted.

5G7K, 2021-06-28:
With NONE it does not work from beginning...

```
modbus.0 | 2021-06-28 16:40:19.875 | debug | (32036) Cleaning up request fifo.
-- | -- | -- | --
modbus.0 | 2021-06-28 16:40:19.874 | debug | (32036) Clearing timeout of the current request.
modbus.0 | 2021-06-28 16:40:19.865 | info | (32036) Disconnected from slave
modbus.0 | 2021-06-28 16:40:18.865 | warn | (32036) Poll error count: 1 code: {"err":"timeout","timeout":500}
modbus.0 | 2021-06-28 16:40:18.863 | error | (32036) Request timed out.
modbus.0 | 2021-06-28 16:40:18.862 | warn | (32036) Error: undefined
modbus.0 | 2021-06-28 16:40:18.355 | debug | (32036) Poll holdingRegs DevID(2) address 1 - 20 bytes
modbus.0 | 2021-06-28 16:40:18.355 | debug | (32036) Initialization of scale factors done!
modbus.0 | 2021-06-28 16:40:18.353 | debug | (32036) Poll device 2
modbus.0 | 2021-06-28 16:40:18.352 | info | (32036) Connected to slave
modbus.0 | 2021-06-28 16:40:18.339 | debug | (32036) connect to serial /dev/ttyUSB0 with 115200
modbus.0 | 2021-06-28 16:40:17.386 | debug | (32036) Add holdingRegisters.40393_Störungscode_N4: {"_id":"holdingRegisters.40393_Störungscode_N4","type":"state","common":{"name":"Störungscode N-4","role":"state","type":"number","read":true,"write
modbus.0 | 2021-06-28 16:40:17.385 | debug | (32036) Add holdingRegisters.40385_Code_Störung: {"_id":"holdingRegisters.40385_Code_Störung","type":"state","common":{"name":"Code der aktuellen Störung","role":"state","type":"number","read":true,"w
modbus.0 | 2021-06-28 16:40:17.385 | debug | (32036) Add holdingRegisters.40352_Innenlufttemperatur: {"_id":"holdingRegisters.40352_Innenlufttemperatur","type":"state","common":{"name":"Innenlufttemperatur","role":"state","type":"number","read":
modbus.0 | 2021-06-28 16:40:17.385 | debug | (32036) Add holdingRegisters.40351_Frischlufttemperatur: {"_id":"holdingRegisters.40351_Frischlufttemperatur","type":"state","common":{"name":"Frischlufttemperatur","role":"state","type":"number","rea
modbus.0 | 2021-06-28 16:40:17.385 | debug | (32036) Add holdingRegisters.40350_Bypass_Strom: {"_id":"holdingRegisters.40350_Bypass_Strom","type":"state","common":{"name":"Vom Bypass aufgenommener Strom","role":"state","type":"number","read":tru
modbus.0 | 2021-06-28 16:40:17.384 | debug | (32036) Add holdingRegisters.40349_Bypass_Status: {"_id":"holdingRegisters.40349_Bypass_Status","type":"state","common":{"name":"Bypass Status","role":"state","type":"number","read":true,"write":true,
modbus.0 | 2021-06-28 16:40:17.384 | debug | (32036) Add holdingRegisters.40348_Filterstatus: {"_id":"holdingRegisters.40348_Filterstatus","type":"state","common":{"name":"Filterstatus Std","role":"state","type":"number","read":true,"write":true
modbus.0 | 2021-06-28 16:40:17.384 | debug | (32036) Add holdingRegisters.40347_Filterstatus: {"_id":"holdingRegisters.40347_Filterstatus","type":"state","common":{"name":"Filterstatus","role":"state","type":"number","read":true,"write":true,"de
modbus.0 | 2021-06-28 16:40:17.383 | debug | (32036) Add holdingRegisters.40340_Schalterstatus: {"_id":"holdingRegisters.40340_Schalterstatus","type":"state","common":{"name":"Schalterstatus","role":"state","type":"number","read":true,"write":tr
modbus.0 | 2021-06-28 16:40:17.383 | debug | (32036) Add holdingRegisters.40339_Eingang_10V: {"_id":"holdingRegisters.40339_Eingang_10V","type":"state","common":{"name":"Eingang 0-10V","role":"state","type":"number","read":true,"write":true,"def
modbus.0 | 2021-06-28 16:40:17.383 | debug | (32036) Add holdingRegisters.40338_Versorgungsspannung: {"_id":"holdingRegisters.40338_Versorgungsspannung","type":"state","common":{"name":"Versorgungsspannung","role":"state","type":"number","read":
modbus.0 | 2021-06-28 16:40:17.382 | debug | (32036) Add holdingRegisters.40337_Status_Eingänge: {"_id":"holdingRegisters.40337_Status_Eingänge","type":"state","common":{"name":"Status der Eingänge","role":"state","type":"number","read":true,"wr
modbus.0 | 2021-06-28 16:40:17.382 | debug | (32036) Add holdingRegisters.40286_Parameter_U2: {"_id":"holdingRegisters.40286_Parameter_U2","type":"state","common":{"name":"Parameter 0-10V U2","role":"state","type":"number","read":true,"write":tr
modbus.0 | 2021-06-28 16:40:17.381 | debug | (32036) Add holdingRegisters.40285_Parameter_U1: {"_id":"holdingRegisters.40285_Parameter_U1","type":"state","common":{"name":"Parameter 0-10V U1","role":"state","type":"number","read":true,"write":tr
modbus.0 | 2021-06-28 16:40:17.381 | debug | (32036) Add holdingRegisters.40284_Unknown: {"_id":"holdingRegisters.40284_Unknown","type":"state","common":{"name":"Unknown","role":"state","type":"number","read":true,"write":true,"def":0,"unit":""}
modbus.0 | 2021-06-28 16:40:17.381 | debug | (32036) Add holdingRegisters.40283_Sollwert_Komforttemperatur: {"_id":"holdingRegisters.40283_Sollwert_Komforttemperatur","type":"state","common":{"name":"Sollwert Komforttemperatur Sommer","role":"st
modbus.0 | 2021-06-28 16:40:17.380 | debug | (32036) Add holdingRegisters.40279_Balance: {"_id":"holdingRegisters.40279_Balance","type":"state","common":{"name":"Balance Zuluft-/Abluft-Fördermenge","role":"state","type":"number","read":true,"wri
modbus.0 | 2021-06-28 16:40:17.380 | debug | (32036) Add holdingRegisters.40278_Sollwert_Zuluft_Geschwindigkeit: {"_id":"holdingRegisters.40278_Sollwert_Zuluft_Geschwindigkeit","type":"state","common":{"name":"Sollwert Zuluft-Geschwindigkeit","r
[... cut ...]
modbus.0 | 2021-06-28 16:40:17.369 | debug | (32036) Initialize Objects for holdingRegs: [{"_address":"40002","name":"ID_Code","description":"ID-Code Anlage","unit":"","type":"uint32be","len":2,"factor":1,"offset":0,"formula":"","role":"state","
modbus.0 | 2021-06-28 16:40:17.368 | debug | (32036) Initialize Objects for inputRegs: []
modbus.0 | 2021-06-28 16:40:17.368 | debug | (32036) Initialize Objects for coils: []
modbus.0 | 2021-06-28 16:40:17.367 | debug | (32036) Initialize Objects for disInputs: []
modbus.0 | 2021-06-28 16:40:17.110 | info | (32036) starting. Version 3.4.8 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.22.1, js-controller: 3.3.14
```

GermanBluefox, 2021-06-28:
As i said, the error looks like as invalid characters. May be cable is too long?

5G7K, 2021-06-28:
Thank you for this hint. It's a 7m twisted pair. But I will try a better shield one the next few days and currently it's in a cable duct with some power cables...

closed 2021-06-28 by 5G7K