ioBroker/ioBroker.modbus #143: modbus.0 did not show already configurated and working holdingregisters. all other 3 instances do...

smartboart, 2021-07-05:
modbus.0 did not show already configurated and working holdingregisters in web gui. all other 3 instances do...

I am also missing the import and export function... 
i found this problem as i wanted to copy  / export the holding registers for another forth instance..

Actual system repository beta release..

see pictures and system infos below.

[image]
[image]
[image]

ioBroker-Master

Platform
    linux
Betriebssystem
    linux
Architektur
    arm
CPUs
    4
Geschwindigkeit
    1500 MHz
Modell
    ARMv7 Processor rev 3 (v7l)
RAM
    7.71 GB
System Betriebszeit
    02:27:14
Node.js
    v12.22.2 (Es gibt eine neuere Version: v12.22.3)
NPM
    6.14.13
Festplatte Größe
    117.39 GB
Festplatte frei
    90.87 GB
Anzahl der Adapter
    421
Betriebszeit
    02:27:09
Aktive Instanzen
    28
location
    /opt/iobroker/
Hostname
    ioBroker-Master

Raspi-Pool

Platform
    linux
Betriebssystem
    linux
Architektur
    arm
CPUs
    4
Geschwindigkeit
    1500 MHz
Modell
    ARMv7 Processor rev 3 (v7l)
RAM
    3.33 GB
System Betriebszeit
    01:27:19
Node.js
    v12.22.2 (Es gibt eine neuere Version: v12.22.3)
NPM
    6.14.13
Festplatte Größe
    117.37 GB
Festplatte frei
    111.82 GB
Anzahl der Adapter
    421
Betriebszeit
    07:36:47
Aktive Instanzen
    3
location
    /opt/iobroker/
Hostname
    Raspi-Pool

Raspi-Wall

Platform
    linux
Betriebssystem
    linux
Architektur
    arm
CPUs
    4
Geschwindigkeit
    1500 MHz
Modell
    ARMv7 Processor rev 3 (v7l)
RAM
    7.54 GB
System Betriebszeit
    01:27:07
Node.js
    v12.22.2 (Es gibt eine neuere Version: v12.22.3)
NPM
    6.14.13
Festplatte Größe
    117.37 GB
Festplatte frei
    112.41 GB
Anzahl der Adapter
    421
Betriebszeit
    06:16:40
Aktive Instanzen
    2
location
    /opt/iobroker/
Hostname
    Raspi-Wall

Apollon77, 2021-07-06:
Are there any infos in the browsr javascript console?

smartboart, 2021-07-06:
I will check this next time I am sitting in front of my notebook... Just checked with my mobile and Firefox app... Same issue.. .

smartboart, 2021-07-08:
hello, just updated the modbus adapter. The last update 3.4.9 fixed this issue. Thanks a lot ...

closed 2021-07-08 by smartboart

kaybau, 2021-07-14:
with latest version the adapter is changing several holding registers  each second. this change is not triggered by a script, the script is stopped.
Each second the values are flapping 0C and 10 C