ioBroker/ioBroker.modbus #145: wirenboard module returns "null"

Anzic23, 2021-07-13:
**Describe the bug**  
When connecting the wb-mr 6c v2 module via a usb adapter and specifying the coil registers, the control occurs normally (the relays turn on and off), but the state in the objects is returned null. If poll the same module via modbus poll, there is no such problem.
**To Reproduce**  

**Expected behavior**  
the channel state corresponds to the actual one.

**Screenshots & Logfiles**  
[image]

[image]

[image]

[image]

[image]

[image]

[image]

**Versions:**  
 - Adapter version: 3.4.9
 - JS-Controller version: 3.3.14
 - Node version: v12.22.1
 - NPM: 6.14.12
 - Operating system: Debian

**Additional context**  
https://wirenboard.com/wiki/Relay_Module_Modbus_Management/en

GermanBluefox, 2021-07-30:
Включи "Опрос" для значений, которые хочешь видеть

Anzic23, 2021-07-30:
Вроде включал. В понедельник ещё раз проверю.

Anzic23, 2021-08-02:
да, если поставить галочку на "опрос", все работает как надо! Спасибо за подсказку!

closed 2021-08-02 by Anzic23