ioBroker/ioBroker.modbus #146: version 3.2.6. : "control object" does not set the value

kaybau, 2021-07-14:
**Describe the bug**  
Hello,
I am using the blockly block "control object" to set a Modbus parameter of my heating device. After adapter update the Value is not changed, but the timestamp of the object has changed to the time when the change happened.

**To Reproduce**  
Blockly code:
control object "TEMP_WW_Soll" with value of object "_userdata.0.Heizung_-_manuelle_Settings.Warmwassertemperatur_default_-_manuell" 
JS code:
<field name="OID">modbus.1.holdingRegisters.105_Temp__WW Soll</field>
  <field name="WITH_DELAY">FALSE</field>
  <value name="VALUE">
    <block type="get_value" id="Up;#|ear%p9PA.REH8{[">
      <field name="ATTR">val</field>
      <field name="OID">0_userdata.0.Heizung_-_manuelle_Settings.Warmwassertemperatur_default_-_manuell</field>
    </block>
  </value>
</block>

changing the value in iobroker/objects will affect the modbus device. changing the value with VIS (+/-) will also affect the modbus device. Only blocky "control block" is no longer affecting the device.

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
control object should set value on modbus device and show new value in iobroker/objects

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: 3.2.6
 - JS-Controller version: 3.2.16
 - Node version: 12.22.1
 - Operating system: Raspbian GNU/Linux 10 (buster)

**Additional context**  
Add any other context about the problem here.

Apollon77, 2021-07-14:
Please test latest version if issue is still there ... we are at 3.4.9 or such in the meantime

kaybau, 2021-07-14:
Hi Ingo,
have seen 3.4.9 in beta, but stable was still 3.2.6.
Will test and feedback. Thanks

kaybau, 2021-07-14:
with latest version the adapter is changing several holding registers each second. this change is not triggered by a script, the script is stopped.
Each second the values are flapping 0C and 10 C

kaybau, 2021-07-14:
[image]
[image]

kaybau, 2021-07-14:
I did a downgrade, back to 3.2.6 - the flapping issue is now gone. Also "control object/steuere Objekt" is working again

closed 2021-07-14 by kaybau

Apollon77, 2021-07-14:
Whyever it should work again after the downgrade ... not understandable. What you mean with "changing several holding registers each second"? Can ypu please provide a debug log? Because other users do not reported this issue

kaybau, 2021-07-14:
see the screenshot above. it war changing 2 of the holding registers each second. values of "0" and "10" were chosen, both values are not defined in any script or parameter. it just happened by it's own.
to provide debug log i have to upgrade again and what to I have to do then?

Apollon77, 2021-07-14:
>  it just happened by it's own.

nothing happens "by it's own", the code reacts on external triggers. so only a debug log could show us what the trigger was. But yes this would mean you need to upgrade again, but it would help to understand it

kaybau, 2021-07-14:
with 3.4.9 flapping issue came back. 
debug is acitvated.

kaybau, 2021-07-14:
within 3.2.6. the debug looks like:
<html><body>

modbus.1 | 2021-07-14 12:42:01.299 | debug | (18055) Closing client on purpose.
-- | -- | -- | --
modbus.1 | 2021-07-14 12:42:01.298 | info | (18055) Got terminate signal TERMINATE_YOURSELF
modbus.1 | 2021-07-14 12:42:00.733 | debug | (18055) Poll holdingRegs DevID(1) address 133 - 1 bytes
modbus.1 | 2021-07-14 12:42:00.729 | debug | (18055) Poll holdingRegs DevID(1) address 100 - 6 bytes
modbus.1 | 2021-07-14 12:42:00.725 | debug | (18055) Poll inputRegs DevID(1) address 68 - 1 bytes
modbus.1 | 2021-07-14 12:42:00.721 | debug | (18055) Poll inputRegs DevID(1) address 42 - 3 bytes
modbus.1 | 2021-07-14 12:42:00.717 | debug | (18055) Poll inputRegs DevID(1) address 26 - 16 bytes
modbus.1 | 2021-07-14 12:42:00.713 | debug | (18055) Poll inputRegs DevID(1) address 10 - 16 bytes
modbus.1 | 2021-07-14 12:42:00.712 | debug | (18055) Poll device 1
modbus.1 | 2021-07-14 12:42:00.029 | debug | (18055) Write successfully [105]: 460
modbus.1 | 2021-07-14 12:41:59.657 | debug | (18055) Poll holdingRegs DevID(1) address 133 - 1 bytes
modbus.1 | 2021-07-14 12:41:59.654 | debug | (18055) Poll holdingRegs DevID(1) address 100 - 6 bytes
modbus.1 | 2021-07-14 12:41:59.639 | debug | (18055) Poll inputRegs DevID(1) address 68 - 1 bytes
modbus.1 | 2021-07-14 12:41:59.635 | debug | (18055) Poll inputRegs DevID(1) address 42 - 3 bytes
modbus.1 | 2021-07-14 12:41:59.630 | debug | (18055) Poll inputRegs DevID(1) address 26 - 16 bytes
modbus.1 | 2021-07-14 12:41:59.626 | debug | (18055) Poll inputRegs DevID(1) address 10 - 16 bytes
modbus.1 | 2021-07-14 12:41:59.625 | debug | (18055) Poll device 1
modbus.1 | 2021-07-14 12:41:58.571 | debug | (18055) Poll holdingRegs DevID(1) address 133 - 1 bytes
modbus.1 | 2021-07-14 12:41:58.563 | debug | (18055) Poll holdingRegs DevID(1) address 100 - 6 bytes
modbus.1 | 2021-07-14 12:41:58.561 | debug | (18055) Poll inputRegs DevID(1) address 68 - 1 bytes
modbus.1 | 2021-07-14 12:41:58.558 | debug | (18055) Poll inputRegs DevID(1) address 42 - 3 bytes
modbus.1 | 2021-07-14 12:41:58.556 | debug | (18055) Poll inputRegs DevID(1) address 26 - 16 bytes
modbus.1 | 2021-07-14 12:41:58.553 | debug | (18055) Poll inputRegs DevID(1) address 10 - 16 bytes
modbus.1 | 2021-07-14 12:41:58.552 | debug | (18055) Poll device 1
modbus.1 | 2021-07-14 12:41:57.500 | debug | (18055) Poll holdingRegs DevID(1) address 133 - 1 bytes
modbus.1 | 2021-07-14 12:41:57.497 | debug | (18055) Poll holdingRegs DevID(1) address 100 - 6 bytes
modbus.1 | 2021-07-14 12:41:57.495 | debug | (18055) Poll inputRegs DevID(1) address 68 - 1 bytes
modbus.1 | 2021-07-14 12:41:57.493 | debug | (18055) Poll inputRegs DevID(1) address 42 - 3 bytes
modbus.1 | 2021-07-14 12:41:57.491 | debug | (18055) Poll inputRegs DevID(1) address 26 - 16 bytes
modbus.1 | 2021-07-14 12:41:57.488 | debug | (18055) Poll inputRegs DevID(1) address 10 - 16 bytes
modbus.1 | 2021-07-14 12:41:57.487 | debug | (18055) Poll device 1
modbus.1 | 2021-07-14 12:41:56.435 | debug | (18055) Poll holdingRegs DevID(1) address 133 - 1 bytes
modbus.1 | 2021-07-14 12:41:56.432 | debug | (18055) Poll holdingRegs DevID(1) address 100 - 6 bytes
modbus.1 | 2021-07-14 12:41:56.430 | debug | (18055) Poll inputRegs DevID(1) address 68 - 1 bytes
modbus.1 | 2021-07-14 12:41:56.428 | debug | (18055) Poll inputRegs DevID(1) address 42 - 3 bytes
modbus.1 | 2021-07-14 12:41:56.425 | debug | (18055) Poll inputRegs DevID(1) address 26 - 16 bytes
modbus.1 | 2021-07-14 12:41:56.423 | debug | (18055) Poll inputRegs DevID(1) address 10 - 16 bytes
modbus.1 | 2021-07-14 12:41:56.422 | debug | (18055) Poll device 1
modbus.1 | 2021-07-14 12:41:55.369 | debug | (18055) Poll holdingRegs DevID(1) address 133 - 1 bytes
modbus.1 | 2021-07-14 12:41:55.366 | debug | (18055) Poll holdingRegs DevID(1) address 100 - 6 bytes
modbus.1 | 2021-07-14 12:41:55.363 | debug | (18055) Poll inputRegs DevID(1) address 68 - 1 bytes
modbus.1 | 2021-07-14 12:41:55.360 | debug | (18055) Poll inputRegs DevID(1) address 42 - 3 bytes
modbus.1 | 2021-07-14 12:41:55.357 | debug | (18055) Poll inputRegs DevID(1) address 26 - 16 bytes
modbus.1 | 2021-07-14 12:41:55.354 | debug | (18055) Poll inputRegs DevID(1) address 10 - 16 bytes
modbus.1 | 2021-07-14 12:41:55.353 | debug | (18055) Poll device 1
modbus.1 | 2021-07-14 12:41:54.297 | debug | (18055) Poll holdingRegs DevID(1) address 133 - 1 bytes
modbus.1 | 2021-07-14 12:41:54.294 | debug | (18055) Poll holdingRegs DevID(1) address 100 - 6 bytes
modbus.1 | 2021-07-14 12:41:54.291 | debug | (18055) Poll inputRegs DevID(1) address 68 - 1 bytes
modbus.1 | 2021-07-14 12:41:54.288 | debug | (18055) Poll inputRegs DevID(1) address 42 - 3 bytes
modbus.1 | 2021-07-14 12:41:54.283 | debug | (18055) Poll inputRegs DevID(1) address 26 - 16 bytes
modbus.1 | 2021-07-14 12:41:54.280 | debug | (18055) Poll inputRegs DevID(1) address 10 - 16 bytes
modbus.1 | 2021-07-14 12:41:54.279 | debug | (18055) Poll device 1
modbus.1 | 2021-07-14 12:41:53.225 | debug | (18055) Poll holdingRegs DevID(1) address 133 - 1 bytes
modbus.1 | 2021-07-14 12:41:53.221 | debug | (18055) Poll holdingRegs DevID(1) address 100 - 6 bytes
modbus.1 | 2021-07-14 12:41:53.212 | debug | (18055) Poll inputRegs DevID(1) address 68 - 1 bytes
modbus.1 | 2021-07-14 12:41:53.207 | debug | (18055) Poll inputRegs DevID(1) address 42 - 3 bytes
modbus.1 | 2021-07-14 12:41:53.203 | debug | (18055) Poll inputRegs DevID(1) address 26 - 16 bytes
[... cut ...]
modbus.1 | 2021-07-14 12:41:11.819 | debug | (18055) Poll inputRegs DevID(1) address 68 - 1 bytes
modbus.1 | 2021-07-14 12:41:11.809 | debug | (18055) Poll inputRegs DevID(1) address 42 - 3 bytes
modbus.1 | 2021-07-14 12:41:11.798 | debug | (18055) Poll inputRegs DevID(1) address 26 - 16 bytes
modbus.1 | 2021-07-14 12:41:11.780 | debug | (18055) Poll inputRegs DevID(1) address 10 - 16 bytes
modbus.1 | 2021-07-14 12:41:11.779 | debug | (18055) Initialization of scale factors done!
modbus.1 | 2021-07-14 12:41:11.778 | debug | (18055) Poll device 1

</body>
</html>

Apollon77, 2021-07-14:
In fact I only see that a polling interval of 1s is set and he is getting data ... so seems like device returns it that way ?! How the log from the old version ooks like in comparism?

kaybau, 2021-07-14:
my latest comment is of old version (3.2.6)- only polling requests are there.
the debug of latest version is the previous comment which starts with initilizing the objects.

Apollon77, 2021-07-14:
Then honestly I do not know. There are some new settings in 3.4.9 in what to use to read in certain cases, maybe play around with this a bit