ioBroker/ioBroker.modbus #147: Cannot read two holding register

HJS72, 2021-07-20:
I want to read several holding register from a Maico WS75, but when I try to read more than one register I get following error:
"Poll error count: 1 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}".

Reading the registers with ModbusClient for Mac just works fine.

Logfile:

`2021-07-20 19:33:46.582  - [32minfo[39m: host.tinkerboard "system.adapter.modbus.2" enabled 2021-07-20 19:33:46.806  - [32minfo[39m: host.tinkerboard instance system.adapter.modbus.2 started with pid 23772 2021-07-20 19:33:48.015  - [34mdebug[39m: modbus.2 (23772) Redis Objects: Use Redis connection: 127.0.0.1:9001 2021-07-20 19:33:48.080  - [34mdebug[39m: modbus.2 (23772) Objects client ready ... initialize now 2021-07-20 19:33:48.098  - [34mdebug[39m: modbus.2 (23772) Objects create System PubSub Client 2021-07-20 19:33:48.101  - [34mdebug[39m: modbus.2 (23772) Objects create User PubSub Client 2021-07-20 19:33:48.102  - [34mdebug[39m: modbus.2 (23772) Objects client initialize lua scripts 2021-07-20 19:33:48.128  - [34mdebug[39m: modbus.2 (23772) Objects connected to redis: 127.0.0.1:9001 2021-07-20 19:33:48.133  - [35msilly[39m: modbus.2 (23772) redis psubscribe cfg.o.system.user.* 2021-07-20 19:33:48.142  - [34mdebug[39m: modbus.2 (23772) objectDB connected 2021-07-20 19:33:48.146  - [34mdebug[39m: modbus.2 (23772) Redis States: Use Redis connection: 127.0.0.1:9000 2021-07-20 19:33:48.160  - [34mdebug[39m: modbus.2 (23772) States create System PubSub Client 2021-07-20 19:33:48.162  - [34mdebug[39m: modbus.2 (23772) States create User PubSub Client 2021-07-20 19:33:48.212  - [34mdebug[39m: modbus.2 (23772) States connected to redis: 127.0.0.1:9000 2021-07-20 19:33:48.213  - [34mdebug[39m: modbus.2 (23772) statesDB connected 2021-07-20 19:33:48.642  - [34mdebug[39m: modbus.2 (23772) Plugin sentry Initialize Plugin (enabled=true) 2021-07-20 19:33:48.865  - [32minfo[39m: modbus.2 (23772) starting. Version 3.4.9 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.22.2, js-controller: 3.2.16 2021-07-20 19:33:48.964  - [35msilly[39m: modbus.2 (23772) States system redis pmessage system.adapter.modbus.2.logLevel/system.adapter.modbus.2.logLevel:{"val":"silly","ack":true,"ts":1626809628945,"q":0,"from":"system.adapter.modbus.2","lc":1626809628945} 2021-07-20 19:33:49.068  - [34mdebug[39m: modbus.2 (23772) Initialize Objects for disInputs: [] 2021-07-20 19:33:49.069  - [34mdebug[39m: modbus.2 (23772) Initialize Objects for coils: [] 2021-07-20 19:33:49.070  - [34mdebug[39m: modbus.2 (23772) Initialize Objects for inputRegs: [] 2021-07-20 19:33:49.070  - [34mdebug[39m: modbus.2 (23772) Initialize Objects for holdingRegs: [{"_address":"550","name":"Betriebsart","description":"","unit":"","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":"","cw":false,"isScale":"","address":550,"deviceId":1,"id":"holdingRegisters.550_Betriebsart"},{"_address":"554","name":"Lueftungsstufe","description":"","unit":"","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":"","cw":false,"isScale":"","address":554,"deviceId":1,"id":"holdingRegisters.554_Lueftungsstufe"}] 2021-07-20 19:33:49.071  - [34mdebug[39m: modbus.2 (23772) Add holdingRegisters.550_Betriebsart: {"_id":"holdingRegisters.550_Betriebsart","type":"state","common":{"name":"","role":"value","type":"number","read":true,"write":true,"def":0,"unit":""},"native":{"regType":"holdingRegs","address":550,"deviceId":1,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}} 2021-07-20 19:33:49.072  - [34mdebug[39m: modbus.2 (23772) Add holdingRegisters.554_Lueftungsstufe: {"_id":"holdingRegisters.554_Lueftungsstufe","type":"state","common":{"name":"","role":"value","type":"number","read":true,"write":true,"def":0,"unit":""},"native":{"regType":"holdingRegs","address":554,"deviceId":1,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}} 2021-07-20 19:33:49.428  - [35msilly[39m: modbus.2 (23772) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":false,"ack":true,"ts":1626809629425,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1626809585871} 2021-07-20 19:33:49.431  - [32minfo[39m: modbus.2 (23772) Connected to slave 10.13.10.131 2021-07-20 19:33:49.432  - [34mdebug[39m: modbus.2 (23772) Poll device 1 2021-07-20 19:33:49.434  - [34mdebug[39m: modbus.2 (23772) Initialization of scale factors done! 2021-07-20 19:33:49.435  - [34mdebug[39m: modbus.2 (23772) Poll holdingRegs DevID(1) address 550 - 5 bytes 2021-07-20 19:33:49.494  - [35msilly[39m: modbus.2 (23772) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":true,"ack":true,"ts":1626809629491,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1626809629491} 2021-07-20 19:33:49.564  - [34mdebug[39m: modbus.2 (23772) Received pdu describes an error. 2021-07-20 19:33:49.567  - [33mwarn[39m: modbus.2 (23772) Poll error count: 1 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"} 2021-07-20 19:33:49.579  - [35msilly[39m: modbus.2 (23772) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":false,"ack":true,"ts":1626809629574,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1626809629574} 2021-07-20 19:33:51.992  - [32minfo[39m: host.tinkerboard "system.adapter.modbus.2" disabled 2021-07-20 19:33:51.994  - [32minfo[39m: host.tinkerboard stopInstance system.adapter.modbus.2 (force=false, process=true) 2021-07-20 19:33:52.006  - [35msilly[39m: modbus.2 (23772) States system redis pmessage system.adapter.modbus.2.sigKill/system.adapter.modbus.2.sigKill:{"val":-1,"ack":false,"ts":1626809632000,"q":0,"from":"system.host.tinkerboard","lc":1626809632000} 2021-07-20 19:33:52.007  - [32minfo[39m: modbus.2 (23772) Got terminate signal TERMINATE_YOURSELF 2021-07-20 19:33:52.009  - [34mdebug[39m: modbus.2 (23772) Closing client on purpose. 2021-07-20 19:33:52.011  - [32minfo[39m: modbus.2 (23772) terminating 2021-07-20 19:33:52.012  - [34mdebug[39m: modbus.2 (23772) Plugin sentry destroyed 2021-07-20 19:33:52.013  - [32minfo[39m: modbus.2 (23772) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason 2021-07-20 19:33:52.042  - [32minfo[39m: host.tinkerboard stopInstance system.adapter.modbus.2 send kill signal 2021-07-20 19:33:52.054  - [35msilly[39m: modbus.2 (23772) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":false,"ack":true,"ts":1626809632050,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1626809629574} 2021-07-20 19:33:52.057  - [34mdebug[39m: modbus.2 (23772) Socket closed with error 2021-07-20 19:33:52.058  - [34mdebug[39m: modbus.2 (23772) Cleaning up request fifo. 2021-07-20 19:33:52.748  - [32minfo[39m: host.tinkerboard instance system.adapter.modbus.2 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
--`

GermanBluefox, 2021-07-30:
Set "Max read request length" to 1

HJS72, 2021-08-24:
Hello GermanBlueFox,
I managed to get the Adapter Running with the Maico WS75. But after 3-5 days the adapter running fine I get following error:

[image]

`2021-08-24 00:18:02.834  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 550 - 1 bytes
2021-08-24 00:18:03.378  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 553 - 1 bytes
2021-08-24 00:18:03.979  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 554 - 1 bytes
2021-08-24 00:18:04.580  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 650 - 1 bytes
2021-08-24 00:18:05.179  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 655 - 1 bytes
2021-08-24 00:19:06.331  - [34mdebug[39m: modbus.2 (13929) Poll device 1
2021-08-24 00:19:06.331  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 550 - 1 bytes
2021-08-24 00:19:06.877  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 553 - 1 bytes
2021-08-24 00:19:07.476  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 554 - 1 bytes
2021-08-24 00:19:08.078  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 650 - 1 bytes
2021-08-24 00:19:08.688  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 655 - 1 bytes
2021-08-24 00:19:11.533  - [33mwarn[39m: modbus.2 (13929) Poll error count: 1 code: "App Timeout"
2021-08-24 00:19:11.545  - [35msilly[39m: modbus.2 (13929) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":false,"ack":true,"ts":1629764351541,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1629764351541}
2021-08-24 00:19:13.690  - [33mwarn[39m: modbus.2 (13929) Error: undefined
2021-08-24 00:19:13.690  - [31merror[39m: modbus.2 (13929) Request timed out.
2021-08-24 00:19:13.691  - [31merror[39m: modbus.2 (13929) Client in error state.
2021-08-24 00:19:13.692  - [34mdebug[39m: modbus.2 (13929) Socket closed with error
2021-08-24 00:19:13.693  - [34mdebug[39m: modbus.2 (13929) Clearing timeout of the current request.
2021-08-24 00:19:13.693  - [34mdebug[39m: modbus.2 (13929) Cleaning up request fifo.
2021-08-24 00:19:14.690  - [34mdebug[39m: modbus.2 (13929) Closing client on purpose.
2021-08-24 00:19:14.691  - [32minfo[39m: modbus.2 (13929) Disconnected from slave 10.13.10.131
2021-08-24 00:19:14.701  - [35msilly[39m: modbus.2 (13929) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":false,"ack":true,"ts":1629764354696,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1629764351541}
2021-08-24 00:20:14.697  - [32minfo[39m: modbus.2 (13929) Connected to slave 10.13.10.131
2021-08-24 00:20:14.698  - [34mdebug[39m: modbus.2 (13929) Poll device 1
2021-08-24 00:20:14.698  - [34mdebug[39m: modbus.2 (13929) Poll holdingRegs DevID(1) address 550 - 1 bytes
2021-08-24 00:20:14.709  - [35msilly[39m: modbus.2 (13929) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":true,"ack":true,"ts":1629764414705,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1629764414705}
2021-08-24 00:20:19.700  - [33mwarn[39m: modbus.2 (13929) Error: undefined
2021-08-24 00:20:19.700  - [31merror[39m: modbus.2 (13929) Request timed out.
2021-08-24 00:20:19.701  - [31merror[39m: modbus.2 (13929) Client in error state.
2021-08-24 00:20:19.701  - [33mwarn[39m: modbus.2 (13929) Poll error count: 2 code: {"err":"timeout","timeout":5000}
2021-08-24 00:20:19.704  - [34mdebug[39m: modbus.2 (13929) Socket closed with error
2021-08-24 00:20:19.706  - [34mdebug[39m: modbus.2 (13929) Clearing timeout of the current request.
2021-08-24 00:20:19.707  - [34mdebug[39m: modbus.2 (13929) Cleaning up request fifo.
2021-08-24 00:20:19.764  - [35msilly[39m: modbus.2 (13929) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":false,"ack":true,"ts":1629764419759,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1629764419759}
2021-08-24 00:20:20.700  - [34mdebug[39m: modbus.2 (13929) Closing client on purpose.
2021-08-24 00:20:20.701  - [32minfo[39m: modbus.2 (13929) Disconnected from slave 10.13.10.131`