ioBroker/ioBroker.modbus #149: Modbus poll time - to long

MaaG-dA, 2021-07-25:
After update modbus adapter to 3.4.9 pollTime was changed from (about) 50ms to.. 1700ms.
I try on varous modbus version adapter. Last "work" version are 3.4.4 (with short/fast pollTime).
See on iobroker administrator panel - Objects/modbus.X/info/pollTime

GermanBluefox, 2021-07-30:
There are many settings that could affect the total poll time:
- Wait time,
- Poll delay
- Read interval (new)
- Write interval (new)

Try to set these values to very small values.

MaaG-dA, 2021-07-30:
I try to change one of value as you describe, but every change only update poll time to longer (up to .. 5400ms).

MaaG-dA, 2021-08-03:
3.4.11 still over 1700 ms vs 40ms on v3.4.4

GermanBluefox, 2021-08-31:
I still cannot see, where the tim can be lost.
Please try 3.4.14 and if the issue still exists, provide the log in debug mode (only short peace between two "Poll starts") like:
[image]

MaaG-dA, 2021-09-01:
I fixed my problem.. but i don't know when and how. (I reinstall various version of modbus instance, restore older version iobroker, remove and add again all modbus and internal variable....
Now i use modbus 3.4.11 with 27ms polltime...