ioBroker/ioBroker.modbus #150: Poll timeout after multiple successful pollings

josefzahner, 2021-07-27:
I'm using v3.4.9 with serial connection:
[image]

The holding register polling works multiple times until it stops with a timeout where only a (manual/auto) restart of the adapter solves the issue. Any comments how to fix the issue?

`Disconnected from slave
Poll error count: 2 code: {"err":"timeout","timeout":5000}
Request timed out.
Error: undefined
Poll holdingRegs DevID(20) address 16384 - 2 bytes
Poll device 20
Poll holdingRegs DevID(20) address 16384 - 2 bytes
Poll device 20
...`

josefzahner, 2021-07-27:
Additionally after the disconnect & reconnect, the polling works only once and I'm getting again a timeout and restart. After a manual restart it works again for multiple pollings.

Apollon77, 2021-07-27:
Please enable debug logging for the instance and provide a  more complete log in debug mode when the error appears (and show some time before and some time after the exact starting timepoint)

josefzahner, 2021-07-27:
Log as attachment. I've started the adapter and after about 2minutes it produces a timeout. 60s later the adapter gets automatically restartet, however polling works only once and I'm getting again a timeout. This continues infinite
[debug.log](https://github.com/ioBroker/ioBroker.modbus/files/6888136/debug.log)

Apollon77, 2021-07-27:
Schonmal versucht dire Polling Zeit bissl hochzusetzen? Also es ist wirklich so das der adapter keine antwort mehr vom Gerät bekommt. Also entweder das Gerät hat ein Problem oder es liegt am Seriellen kram im Linux bereich. Ist in /var/log/syslog irgendwas drin? Oder sagt "dmesg" was? weil wenn es nach einem neustart wieder tut dann hängt es irgendwie am Prozess

josefzahner, 2021-07-27:
dmesg und auch /var/log/syslog haben rein gar nichts im Zusammenhang mit Modbus drin. 
Wenn ich die Polling Zeit auf 10s hochschraube kommt ein anderer Fehler (unknown function code: 0x14, 0x3). Siehe Attachment:
[poll_time_10s.log](https://github.com/ioBroker/ioBroker.modbus/files/6888541/poll_time_10s.log)

josefzahner, 2021-07-27:
Sorry gerade gemerkt das ich in der Zwischenzeit auch noch das "Read Interval" auf 48ms statt auf 0ms stand. Daher wohl die andere Fehlermeldung.

Apollon77, 2021-07-27:
Nochwas: auf was steht den  "Reconnect Delay"? der versuch 3ms nach dem disconnect schon einen reconnect. das kann nicht klappen. kann sein das da eine 0 steht?

Apollon77, 2021-07-27:
Please install GitHub version, do a manual restart, try again ad provide log

josefzahner, 2021-07-28:
Ok, I've removed the official version and installed the one from github. In the "General" tab I've only removed the checkmark for "Use aliases", the rest is default.
[modbus_githubversino.log](https://github.com/ioBroker/ioBroker.modbus/files/6896277/modbus_githubversino.log)

Apollon77, 2021-07-30:
Ok, now he is waiting the 1s before recinnection ... STill: What you set as reconnect value in your slave config? Mybe try to increatse to 10s to see if it brings something?

is there any outout with shell command "dmesg" when the issue occurs?

josefzahner, 2021-07-30:
dmesg shows nothing about modbus.
I'm guessing you are referring to "Data Polling Interval", I've changed it from 1s to 10s. Same behaviour as with 1s and in the logs above.

Apollon77, 2021-07-30:
dmesg was more avout serial communication

josefzahner, 2021-08-11:
I switched over to "mbusd" https://github.com/3cky/mbusd, at least for the modbus serial connection part. I'm now using a localhost TCP connection which will be translated to the serial port on mbusd. The registers are still configured on iobroker and it works as it should. So the issue is for sure on the modbus adapter for the serial connection.

HansBorjes, 2021-08-22:
Thanks @josefzahner for pointing out the positive experience with mbusd! Works like a charm.

I was afraid my modbus network could have issues, but fortunately this is not the case.

Anzic23, 2021-09-06:
I have a similar problem, with this device working normally on another computer #145, I switched it to "prodaction", and got a timeout error. I don't even know where to look?
```
2021-09-06 08:38:18.778 - info: modbus.0 (28557) starting. Version 3.4.14 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.22.5, js-controller: 3.3.15
--
2021-09-06 08:38:19.499 - debug: modbus.0 (28557) Initialize Objects for disInputs: [{"_address":0,"name":"","description":"","formula":"","role":"switch","room":"","cw":"","isScale":"","address":0,"deviceId":152,"id":"discreteInputs.0","len":1},{"_address":1,"name":"","description":"","formula":"","role":"switch","room":"","cw":"","isScale":"","address":1,"deviceId":152,"id":"discreteInputs.1","len":1},{"_address":2,"name":"","description":"","formula":"","role":"switch","room":"","cw":"","isScale":"","address":2,"deviceId":152,"id":"discreteInputs.2","len":1},{"_address":3,"name":"","description":"","formula":"","role":"switch","room":"","cw":"","isScale":"","address":3,"deviceId":152,"id":"discreteInputs.3","len":1},{"_address":4,"name":"","description":"","formula":"","role":"switch","room":"","cw":"","isScale":"","address":4,"deviceId":152,"id":"discreteInputs.4","len":1},{"_address":5,"name":"","description":"","formula":"","role":"switch","room":"","cw":"","isScale":"","address":5,"deviceId":152,"id":"discreteInputs.5","len":1},{"_address":6,"name":"","description":"","formula":"","role":"switch","room":"","cw":"","isScale":"","address":6,"deviceId":152,"id":"discreteInputs.6","len":1}]
2021-09-06 08:38:19.499 - debug: modbus.0 (28557) Add discreteInputs.0: {"_id":"discreteInputs.0","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":false,"def":false},"native":{"regType":"disInputs","address":0,"deviceId":152}}
2021-09-06 08:38:19.500 - debug: modbus.0 (28557) Add discreteInputs.1: {"_id":"discreteInputs.1","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":false,"def":false},"native":{"regType":"disInputs","address":1,"deviceId":152}}
2021-09-06 08:38:19.500 - debug: modbus.0 (28557) Add discreteInputs.2: {"_id":"discreteInputs.2","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":false,"def":false},"native":{"regType":"disInputs","address":2,"deviceId":152}}
2021-09-06 08:38:19.501 - debug: modbus.0 (28557) Add discreteInputs.3: {"_id":"discreteInputs.3","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":false,"def":false},"native":{"regType":"disInputs","address":3,"deviceId":152}}
2021-09-06 08:38:19.501 - debug: modbus.0 (28557) Add discreteInputs.4: {"_id":"discreteInputs.4","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":false,"def":false},"native":{"regType":"disInputs","address":4,"deviceId":152}}
2021-09-06 08:38:19.501 - debug: modbus.0 (28557) Add discreteInputs.5: {"_id":"discreteInputs.5","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":false,"def":false},"native":{"regType":"disInputs","address":5,"deviceId":152}}
2021-09-06 08:38:19.502 - debug: modbus.0 (28557) Add discreteInputs.6: {"_id":"discreteInputs.6","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":false,"def":false},"native":{"regType":"disInputs","address":6,"deviceId":152}}
2021-09-06 08:38:19.502 - debug: modbus.0 (28557) Initialize Objects for coils: [{"_address":0,"name":"","description":"","formula":"","role":"switch","room":"","poll":true,"wp":"","cw":"","isScale":"","address":0,"deviceId":152,"id":"coils.0","len":1},{"_address":1,"name":"","description":"","formula":"","role":"switch","room":"","poll":true,"wp":"","cw":"","isScale":"","address":1,"deviceId":152,"id":"coils.1","len":1},{"_address":2,"name":"","description":"","formula":"","role":"switch","room":"","poll":true,"wp":"","cw":"","isScale":"","address":2,"deviceId":152,"id":"coils.2","len":1},{"_address":3,"name":"","description":"","formula":"","role":"switch","room":"","poll":true,"wp":"","cw":"","isScale":"","address":3,"deviceId":152,"id":"coils.3","len":1},{"_address":4,"name":"","description":"","formula":"","role":"switch","room":"","poll":true,"wp":"","cw":"","isScale":"","address":4,"deviceId":152,"id":"coils.4","len":1},{"_address":5,"name":"","description":"","formula":"","role":"switch","room":"","poll":true,"wp":"","cw":"","isScale":"","address":5,"deviceId":152,"id":"coils.5","len":1},{"_address":6,"name":"","description":"","formula":"","role":"switch","room":"","poll":true,"wp":"","cw":"","isScale":"","address":6,"deviceId":152,"id":"coils.6","len":1}]
2021-09-06 08:38:19.503 - debug: modbus.0 (28557) Add coils.0: {"_id":"coils.0","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":true,"def":false},"native":{"regType":"coils","address":0,"deviceId":152,"poll":true,"wp":""}}
2021-09-06 08:38:19.503 - debug: modbus.0 (28557) Add coils.1: {"_id":"coils.1","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":true,"def":false},"native":{"regType":"coils","address":1,"deviceId":152,"poll":true,"wp":""}}
2021-09-06 08:38:19.503 - debug: modbus.0 (28557) Add coils.2: {"_id":"coils.2","type":"state","common":{"name":"","role":"switch","type":"boolean","read":true,"write":true,"def":false},"native":{"regType":"coils","address":2,"deviceId":152,"poll":true,"wp":""}}
[... cut ...]
2021-09-06 08:39:10.823 - debug: modbus.0 (28620) [DevID_152] Poll start ---------------------
2021-09-06 08:39:10.824 - debug: modbus.0 (28620) [DevID_152/disInputs] Poll address 0 - 7 bits
2021-09-06 08:39:16.218 - warn: modbus.0 (28620) Error: undefined
2021-09-06 08:39:16.219 - error: modbus.0 (28620) Request timed out.
2021-09-06 08:39:16.220 - warn: modbus.0 (28620) Poll error count: 1 code: {"err":"timeout","timeout":5000}
2021-09-06 08:39:17.220 - info: modbus.0 (28620) Disconnected from slave

```