ioBroker/ioBroker.modbus #153: Last holding register not visible

dklinger, 2021-07-30:
**Describe the bug**  
Add as many holding registeres as necessary until the screen is completely filled.
Then, add another holding register. Try to scroll to the last item. It's not possible as the last item is hidden by the Save-Div.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to 'Holding-Register'
2. Click on '+' as many times as needed to fill the whole screen with Entries.
3. Scroll down to 'the last item'
4. See error behavior

**Expected behavior**  
When I scroll down to the botom, the last Entry should be visible and not hidden by the Save-Div.

**Screenshots & Logfiles**  
There is another entry after 375, which is not visible. Scrolling is not possible as the Div Height is not aligned to each other.
[image]

**Versions:**  
 - Adapter version: 3.4.9
 - JS-Controller version: 3.2.16
 - Node version: 12.22.2
 - Operating system: buanet docker-container

Apollon77, 2021-07-30:
Which Admin version? 4 or 5?

dklinger, 2021-07-30:
Admin version is 4.2.2

Apollon77, 2021-07-30:
Can you please try with Admin 5 (in latest repo)? Because if it is fixed there it will be a "not to be changed". Admin5 will come into stable in next week

dklinger, 2021-07-30:
Thank, tried with v5 and works well.
Reverted back to v4 and will wait until v5 is stable.

Apollon77, 2021-07-30:
Thank you for the check! Then I close here :-)

closed 2021-07-30 by Apollon77