ioBroker/ioBroker.modbus #154: Proplem importing registers

stilicrafter, 2021-07-30:
Hi,
I am running IO Broker on a RaspberryPi 4b+.
I wanted to import Input and holdingregisters but i always get an error.
Also after adding a line manually and then edit it in TabView gave me an error Message.
Any ideas?

I copyed as a workaround the information from an old version of Modbus from the 
/opt/iobroker/iobroker-data/objects.json .  That had worked for me.

Apollon77, 2021-07-30:
>  Any ideas?

Maybe ... if you tell or show us the error messages you got

stilicrafter, 2021-08-02:
No errormessage if i try to edit en existing one.
To reproduce, install new Modbus in iobroker. Click on edit and go to holding register. Add one line and then go to the tabed edit. Alter the name and click save. It is disapered

stilicrafter, 2021-08-02:
I deteced the Problem whyle try to import this from a website: 
[1612277985677-eingangsregister_stand2021_02_02.txt](https://github.com/ioBroker/ioBroker.modbus/files/6918786/1612277985677-eingangsregister_stand2021_02_02.txt)
[1612383966263-holding-register2021_02_04.txt](https://github.com/ioBroker/ioBroker.modbus/files/6918787/1612383966263-holding-register2021_02_04.txt)

GermanBluefox, 2021-08-31:
> No errormessage if i try to edit en existing one.
> To reproduce, install new Modbus in iobroker. Click on edit and go to holding register. Add one line and then go to the tabed edit. Alter the name and click save. It is disapered

Are you sure, that you did not delete the TAB chars? I cannot reproduce the problem

[image]

stilicrafter, 2021-09-01:
I am sure i didnt delete anything. Unless my Browser useing Windows does a automatic conversion.
To reproduce, i click on "Edit as TSV", and then Save. This is for me enought, to clear all existing values. 
After my holiday i will record a Video to show it

GermanBluefox, 2021-09-02:
> I am sure i didnt delete anything. Unless my Browser useing Windows does a automatic conversion.
> To reproduce, i click on "Edit as TSV", and then Save. This is for me enought, to clear all existing values.
> After my holiday i will record a Video to show it

This problem was fixed in previous version.

stilicrafter, 2021-09-04:
@GermanBluefox How many days bevore, if it was a knowen issue? If you alted this behaviour or fixed there Something?  On Monday my holiday ends and i can try to reproduce.