ioBroker/ioBroker.modbus #155: After update iobroker modbus adapter to 3.4.9 input and selection IP address: 0.0.0.0 with slave no longer possible

GOETSCHHOFER, 2021-07-31:
After update iobroker modbus adapter to 3.4.9 input and selection IP address: 0.0.0.0 with slave no longer possible and adapter remains red, before update it worked.

The input partner IP 0.0.0.0 is only possible if master is selected. But I need the cominbation slave and 0.0.0.0. 

Thank you for the help.

GermanBluefox, 2021-07-31:
0.0.0.0 is "Listen on all IPs"
[image]

Why do you want to select the IP address, that is not present on your system?

GOETSCHHOFER, 2021-07-31:
Hallo Bluefox,

thank you very much for the quick response.

In version 3.2.6 I chose Slave and entered IP address: 0.0.0.0. Now I can only choose lists on all IP's. Do I understand this correctly, that with lists on all IP's the IP address 0.0.0.0 is automatically addressed. In my case, the master has the IP address 192.168.22 and port 502. Do I now have to select all IP's as IP address lists for slave in my case?
![Error message modbus adapter slave lists on all IP's
](https://user-images.githubusercontent.com/74736696/127744539-4c9a4186-07e9-4c93-a58d-0966e8620f44.PNG)

What could be wrong ?

Many Thanks in advance for your assistance.

LG

Karl

Apollon77, 2021-07-31:
Please run "Iobroker fix" ... it seems that your permissions are not the correct ones

GOETSCHHOFER, 2021-08-01:
Many Thanks, this was the solution !

LG

Karl

closed 2021-08-01 by GOETSCHHOFER