ioBroker/ioBroker.modbus #157: Vers. 3.4.11: Scale factor 0.10 multiplies the value by factor 10

GOETSCHHOFER, 2021-08-01:
In the modbus adapter.1 I get the temperature value e.g. 197 (=19.7 degrees) as a slave and want to multiply this by the factor 0.10. The result, however, is 1970. That is, the factor 0.10 obviously multiplies by 10 instead of 0.10.

https://forum.iobroker.net/topic/46630/modbus-adapter-faktor?_=1627809764854

https://github.com/ioBroker/ioBroker.modbus
**Factor**
This factor is used **to multiply** the read value from Bus for static scaling. So the calculation looks like following val = x * Factor + Offset

Thank you for the info on how to solve the problem.

LG

Karl

Apollon77, 2021-08-01:
Please attach Debug log. Enable debug log via Admin - Instances - Expert mode - Loglevel to Debug on relevant modbix instance. Then show log from start and such a value to be received

ManuelK83, 2021-08-02:
Hallo, 
habe das selbe Problem. 
Habe mal die Log kopiert. Bei der Temperatur und bei dem Strom hätte ich es gerne geändert.

Danke

[image]

`
<html>
<body>

modbus.0 | 2021-08-02 11:08:32.453 | debug | (27106) Calculation result = 6950, type = number
-- | -- | -- | --
modbus.0 | 2021-08-02 11:08:32.452 | debug | (27106) Formula = x < 0 ? 0 : x
modbus.0 | 2021-08-02 11:08:32.452 | debug | (27106) Input Value = 6950
modbus.0 | 2021-08-02 11:08:32.451 | debug | (27106) Calculation result = 6969, type = number
modbus.0 | 2021-08-02 11:08:32.450 | debug | (27106) Formula = x < 0 ? 0 : x
modbus.0 | 2021-08-02 11:08:32.449 | debug | (27106) Input Value = 6969
modbus.0 | 2021-08-02 11:08:32.449 | debug | (27106) Calculation result = 6945, type = number
modbus.0 | 2021-08-02 11:08:32.448 | debug | (27106) Formula = x < 0 ? 0 : x
modbus.0 | 2021-08-02 11:08:32.447 | debug | (27106) Input Value = 6945
modbus.0 | 2021-08-02 11:08:32.299 | debug | (27106) Poll inputRegs DevID(3) address 30977 - 6 bytes
modbus.0 | 2021-08-02 11:08:32.297 | debug | (27106) Calculation result = 432, type = number
modbus.0 | 2021-08-02 11:08:32.297 | debug | (27106) Formula = x < 0 ? 0 : x
modbus.0 | 2021-08-02 11:08:32.296 | debug | (27106) Input Value = 432
modbus.0 | 2021-08-02 11:08:32.147 | debug | (27106) Poll inputRegs DevID(3) address 30953 - 2 bytes
javascript.0 | 2021-08-02 11:08:32.020 | error | (1538) script.js.Strom.PV_Berechnungen: Trigger modbus
modbus.0 | 2021-08-02 11:08:32.013 | debug | (27106) Poll inputRegs DevID(3) address 30865 - 2 bytes
modbus.0 | 2021-08-02 11:08:32.007 | debug | (27106) Calculation result = 1632, type = number
modbus.0 | 2021-08-02 11:08:32.007 | debug | (27106) Formula = x < 0 ? 0 : x
modbus.0 | 2021-08-02 11:08:32.006 | debug | (27106) Input Value = 1632
modbus.0 | 2021-08-02 11:08:32.006 | debug | (27106) Calculation result = 1637, type = number
modbus.0 | 2021-08-02 11:08:32.005 | debug | (27106) Formula = x < 0 ? 0 : x
modbus.0 | 2021-08-02 11:08:32.005 | debug | (27106) Input Value = 1637
modbus.0 | 2021-08-02 11:08:32.005 | debug | (27106) Calculation result = 1634, type = number
modbus.0 | 2021-08-02 11:08:32.004 | debug | (27106) Formula = x < 0 ? 0 : x
modbus.0 | 2021-08-02 11:08:32.004 | debug | (27106) Input Value = 1634
modbus.0 | 2021-08-02 11:08:32.003 | debug | (27106) Calculation result = 4903, type = number
modbus.0 | 2021-08-02 11:08:32.002 | debug | (27106) Formula = x < 0 ? 0 : x
modbus.0 | 2021-08-02 11:08:32.002 | debug | (27106) Input Value = 4903
modbus.0 | 2021-08-02 11:08:31.858 | debug | (27106) Poll inputRegs DevID(3) address 30775 - 8 bytes
modbus.0 | 2021-08-02 11:08:31.705 | debug | (27106) Poll inputRegs DevID(3) address 30531 - 8 bytes
modbus.0 | 2021-08-02 11:08:31.647 | debug | (27106) Poll inputRegs DevID(3) address 30201 - 2 bytes
modbus.0 | 2021-08-02 11:08:31.647 | debug | (27106) Initialization of scale factors done!
modbus.0 | 2021-08-02 11:08:31.645 | debug | (27106) Poll device 3
modbus.0 | 2021-08-02 11:08:31.643 | info | (27106) Connected to slave 192.168.0.16
modbus.0 | 2021-08-02 11:08:31.240 | debug | (27106) Initialize Objects for holdingRegs: []
modbus.0 | 2021-08-02 11:08:31.239 | debug | (27106) Add inputRegisters.30981_Strom_Außenleiter Phase 3: {"_id":"inputRegisters.30981_Strom_Außenleiter Phase 3","type":"state","common":{"name":"Strom Außenleiter Phase 3","role":"value","type":"n
modbus.0 | 2021-08-02 11:08:31.238 | debug | (27106) Add inputRegisters.30979_Strom_Außenleiter Phase 2: {"_id":"inputRegisters.30979_Strom_Außenleiter Phase 2","type":"state","common":{"name":"Strom Außenleiter Phase 2","role":"value","type":"n
modbus.0 | 2021-08-02 11:08:31.238 | debug | (27106) Add inputRegisters.30977_Strom_Außenleiter Phase 1: {"_id":"inputRegisters.30977_Strom_Außenleiter Phase 1","type":"state","common":{"name":"Strom Außenleiter Phase 1","role":"value","type":"n
modbus.0 | 2021-08-02 11:08:31.238 | debug | (27106) Add inputRegisters.30953_Temperatur: {"_id":"inputRegisters.30953_Temperatur","type":"state","common":{"name":"Temperatur","role":"value","type":"number","read":true,"write":false,"def":0,"uni
modbus.0 | 2021-08-02 11:08:31.237 | debug | (27106) Add inputRegisters.30865_Leistung_vom Netz: {"_id":"inputRegisters.30865_Leistung_vom Netz","type":"state","common":{"name":"Leistung vom Netz","role":"value","type":"number","read":true,"writ
modbus.0 | 2021-08-02 11:08:31.236 | debug | (27106) Add inputRegisters.30781_Wirkleistung_Phase 3: {"_id":"inputRegisters.30781_Wirkleistung_Phase 3","type":"state","common":{"name":"Wirkleistung Phase 3","role":"value","type":"number","read":t
modbus.0 | 2021-08-02 11:08:31.236 | debug | (27106) Add inputRegisters.30779_Wirkleistung_Phase 2: {"_id":"inputRegisters.30779_Wirkleistung_Phase 2","type":"state","common":{"name":"Wirkleistung Phase 2","role":"value","type":"number","read":t
modbus.0 | 2021-08-02 11:08:31.235 | debug | (27106) Add inputRegisters.30777_Wirkleistung_Phase 1: {"_id":"inputRegisters.30777_Wirkleistung_Phase 1","type":"state","common":{"name":"Wirkleistung Phase 1","role":"value","type":"number","read":t
modbus.0 | 2021-08-02 11:08:31.235 | debug | (27106) Add inputRegisters.30775_Wirkleistung_gesamt (W): {"_id":"inputRegisters.30775_Wirkleistung_gesamt (W)","type":"state","common":{"name":"Wirkleistung gesamt (W)","role":"value","type":"number"
[... cut ...] 2be","len":2,"factor":1,"offset":0,"formula":
modbus.0 | 2021-08-02 11:08:31.232 | debug | (27106) Initialize Objects for coils: []
modbus.0 | 2021-08-02 11:08:31.231 | debug | (27106) Initialize Objects for disInputs: []
modbus.0 | 2021-08-02 11:08:30.453 | info | (27106) starting. Version 3.4.9 in /opt/iobroker/node_modules/iobroker.modbus, node: v12.22.1, js-controller: 3.2.16
modbus.0 | 2021-08-02 11:08:30.252 | debug | (27106) Plugin sentry Initialize Plugin (enabled=true)

</body>
</html>`

Apollon77, 2021-08-02:
Ok, wenn man eine Fomel angibt dann wird die Eigene berechnung des Adapter ausgeschaltet weil er denkt man berechnet es eh selbst. Also passe die Formel an und ist erledigt. Ist auch unter https://github.com/ioBroker/ioBroker.modbus#formula klar so dokumentiert.

@GermanBluefox Wir könnten die Felder hiden bzw ausgrauen oder so sobald eine Formel angegeben ist=? Dann ist es inn der UI ersichtlich das Faktor und so nicht greift wenn eine Formel angegeben ist

ManuelK83, 2021-08-02:
Ah, 
OK das mit dem Ausgrauen wäre gut. Habe ich wohl in der Anleitung  übersehen.  
Allerdings bin ich nicht der experte mit diesen Formeln. Wie würde diese in meinem Fall lauten wenn ich den Wert um 0,01 ändern möchte. 

 Danke

Apollon77, 2021-08-02:
x < 0 ? 0 : x*0.01

?

ManuelK83, 2021-08-02:
Danke. 
Großes Lob an den Support. Weiter so !!

GOETSCHHOFER, 2021-08-02:
Wie kann man die Spalte Formel einblenden ?
Die Formelspalte habe ich nicht.
Danke

Apollon77, 2021-08-02:
Hm ... neee müssen es anders machen ...
Die regel "Formel geht vor Faktor/Offset" gilt nur für Werte die gelesen werden. Wenn man schreibt wird immer Faktor und Offset beachtet. Also Felder ausblenden geht nicht so wirklich gut weil sonst die Faktor/Offset Konvertierung beim Schreiben nicht mehr greift.

@GOETSCHHOFER Expertenmodus aktivieren

Apollon77, 2021-08-03:
Die Formel ist abe rnur für die leserichtung relevant. Schreibrichtung dann doch wieder faktor/Offset :-( Aktiviere mal den Expertenmodus

GOETSCHHOFER, 2021-08-03:
Meinst du wieder den Expertenmodus im iobroker u. anschließend debug ?
Kannst du mir bitte sagen, wie ich die Spalte Formel einblende ?
Danke
[image]

Apollon77, 2021-08-03:
Klick auf dem Kopf oben im Admin, das ist der Expertenmodus. Dann sollte die Spalte da sein

GOETSCHHOFER, 2021-08-04:
Ich habe den Expertenmodus im admin.0 bereits auf dauerhaft on.
Habe ebenfalls im modbus.1 auf Expertenmodus geschaltet, aber Spalte Formel wird im Holdingregister nicht sichtbar ?

Danke

[image]

[image]

[image]

GOETSCHHOFER, 2021-08-08:
@Apollon77  Klick auf dem Kopf oben im Admin, das ist der Expertenmodus. Dann sollte die Spalte da sein

Hallo Ingo,

links oben kann ich auch aus- und einblenden, aber die Spalte Formel kommt nicht ?

Danke

[image]

[image]

Apollon77, 2021-08-08:
@GermanBluefox ?

GOETSCHHOFER, 2021-08-09:
@GermanBluefox 

Könntest du mir bitte bei meinem Problem mit dem iobroker modbus adapter-3.4.6 weiterhelfen, ich kann dieses Problem einfach nicht lösen. Habe schon alles versucht.
Ich habe ein Problem mit dem Faktor und die Spalte Formel wird beim Typ Slave nicht eingeblendet.

Danke im Voraus

Karl

GermanBluefox, 2021-08-31:
> Jetzt habe ich den Faktor mal auf 0.10 (Punkt) anstatt 0,1(Komma) geändert und dann bekomme ich nur eine gerade Zahl und der Wert hinter dem Komma geht verloren.
> Resümee: Faktor 0,10(Komma) multipliziert mit 10, Faktor 0.10(Punkt) multipliziert, aber rundet auch gleichzeitig auf eine ganze Zahl auf.
> 
> Habe auch auf value.temperature umgestellt, aber keine Änderung.
> Könnte dies mit der Spalte Länge zusammenhängen ?
> 
> [image]
> [image]
> [image]

Was ist hier eingestellt?
[image]

GOETSCHHOFER, 2021-08-31:
Ja, dies war das Problem, ich hatte keine Anzahl Kommastellen eingegeben.

Entschuldige bitte vielmals, dass ich dir hier Arbeit bereitet habe.

Sollte ich hier auch die Hacken setzen wie du ? 
Was haben diese für Bedeutung ?

Danke im Voraus.

[image]

[image]

[image]

GermanBluefox, 2021-08-31:
Wenn alles funktioniert, muss man nicht alles wie bei mir machen.

Copy&paste from the adapter documentation:

### Do not align addresses to 16 bits (word)
Normally the coils and the discrete inputs addresses are aligned to 16 bit. Like addresses from 3 to 20 will be aligned to 0 up 32.
If this option is active the addresses will not be aligned.

### Use only multiple write registers
If slave only supports "write multiple registers" command, you can activate so the registers will be written always with FC15/FC16 command.

closed 2021-08-31 by GermanBluefox

GOETSCHHOFER, 2021-08-31:
Danke !

GOETSCHHOFER, 2021-11-13:
Hallo German Bluefox,

dürfte ich dich bitte wieder zu einem Problem zum Modbus Adapter fragen ?

Ich habe das Problem, dass ich bei einem Objekt den korrekten Wert z. B. 281,4 Kelvin nur einige Sekunden angezeigt bekomme und anschließend wird wieder der Wert 10 Kelvin angezeigt, Obwohl beim Fremdgerät im Modbus TCP-Ausgang der Wert permanent mit ca. 281 korrekt geliefert wird.

Im Mobus TCP-Adapter multipliziere ich diesen Wert anschließend mit dem Faktor 0,10.

[image]
[image]

[image]

[image]

Meines Erachtens hat es aber mal funktioniert und ich kann mir nicht vorstellen, dass dies mit dem Update von 3.4.12 auf 3.4.14 zusammenhängt.

Ich verstehe auch nicht die Fehlermeldung in den Protokollen, da ich denke, dass die Parameter korrekt mit Number eingestellt sind. ?
Warum verlangt er hier ein boolean ? Expected Boolean, received number ?

Danke im Voraus für deine Hilfe.

LG

Karl